NVIDIA/open-gpu-kernel-modules #391: Broken build when `CONFIG_X86_PAT` is not defined

staticfloat, 2022-10-15:
### NVIDIA Open GPU Kernel Modules Version

7c345b838b8a63126ed425781400c1ae8a7a4a1d

### Operating System and Version

Ubuntu 22.04.1 LTS

### Kernel Release

v5.16.0, custom configured

### Build Command

make SYSSRC="/path/to/custom/kernel"

### Terminal output/Build Log

```
$ make SYSSRC="/path/to/custom/kernel"
...
/cache/build/default-deepsea1-6/julia-computing-1/xtrx-julia/software/nvidia-driver/kernel-open/nvidia/nv-pat.c: In function ‘nv_disable_caches’:
/cache/build/default-deepsea1-6/julia-computing-1/xtrx-julia/software/nvidia-driver/kernel-open/nvidia/nv-pat.c:66:5: error: implicit declaration of function ‘__flush_tlb’; did you mean ‘__flush_tlb_all’? [-Werror=implicit-function-declaration]
   66 |     __flush_tlb();
      |     ^~~~~~~~~~~
      |     __flush_tlb_all
```

### More Info

When compiling against a linux kernel without `CONFIG_X86_PAT`, [this conditional](https://github.com/NVIDIA/open-gpu-kernel-modules/blob/d8f3bcff924776518f1e63286537c3cf365289ac/kernel-open/nvidia/nv-pat.c#L46) enables code that [uses `__flush_tlb()`](https://github.com/NVIDIA/open-gpu-kernel-modules/blob/d8f3bcff924776518f1e63286537c3cf365289ac/kernel-open/nvidia/nv-pat.c#L73).  This function was, if I am reading the kernel source correctly, [removed in Linux 5.8](https://github.com/torvalds/linux/commit/2faf153bb7346b7dfc895f916edf93a86297ec0a).  This codepath should be updated to no longer use the outdated function on newer Linux kernel versions.

niv, 2022-10-27:
Thank you, tracking internally: 3848010

Saperxxx, 2022-10-30:
> ### NVIDIA Open GPU Kernel Modules Version
>
> 7c345b838b8a63126ed425781400c1ae8a7a4a1d
>
> ### Operating System and Version
>
> Ubuntu 22.04.1 LTS
>
> ### Kernel Release
>
> v5.16.0, custom configured
>
> ### Build [...]