NVIDIA/open-gpu-kernel-modules #392: Dynamic Boost doesn't work on AMD CPUs

cyberphantom52, 2022-10-17:
### NVIDIA Open GPU Kernel Modules Version

7c345b838b8a63126ed425781400c1ae8a7a4a1d

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Fedora 37 Workstation

### Kernel Release

6.0.2-300.fc37.x86_64

### Hardware: GPU

NVIDIA GeForce RTX 3060 Laptop GPU

### Describe the bug

nvidia-powerd was added to the linux driver to enable dynamic boost but currently its completely broken on all AMD-NVIDIA based laptops because powerd service is unable to read the cpu powerdata. I have attched the output of service log below.

```
× nvidia-powerd.service - nvidia-powerd service
     Loaded: loaded (/usr/lib/systemd/system/nvidia-powerd.service; disabled; preset: enabled)
     Active: failed (Result: exit-code) since Mon 2022-10-17 13:18:00 IST; 13s ago
    Process: 10710 ExecStart=/usr/bin/nvidia-powerd (code=exited, status=1/FAILURE)
   Main PID: 10710 (code=exited, status=1/FAILURE)
        CPU: 1.148s

Oct 17 13:17:58 rog systemd[1]: Starting nvidia-powerd.service - nvidia-powerd service...
Oct 17 13:17:58 rog /usr/bin/nvidia-powerd[10710]: nvidia-powerd version:1.0(build 1)
Oct 17 13:18:00 rog /usr/bin/nvidia-powerd[10710]: Failed to read the data for calculating CPU power
Oct 17 13:18:00 rog /usr/bin/nvidia-powerd[10710]: Failed to initialize GPU Boost controller
Oct 17 13:18:00 rog systemd[1]: nvidia-powerd.service: Main process exited, code=exited, status=1/FAILURE
Oct 17 13:18:00 rog systemd[1]: nvidia-powerd.service: Failed with result 'exit-code'.
Oct 17 13:18:00 rog systemd[1]: Failed to start nvidia-powerd.service - nvidia-powerd service.
Oct 17 13:18:00 rog systemd[1]: nvidia-powerd.service: Consumed 1.148s CPU time.
```

### To Reproduce

1. Check power consumption in GPU heavy programs.
2. Service logs indicate nvidia-powerd service cannot start.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9798139/nvidia-bug-report.log.gz)

aritger, 2022-10-20:
Thanks, we're aware of the limitation in nvidia-powerd; that should be addressed in an upcoming release.

DualityKyle, 2022-10-22:
@aritger Will the fix address all AMD CPU's or just 5000 series? I ask as we experience this issue on 4000 series Ryzen CPU's as well

cyberphantom52, 2022-11-10:
@aritger Hey, the recent 525.53 beta added support for Dynamic Boost on AMD but its limited to Rembrandt and newer AMD chips. Are there plans to support older 4000 and 5000 series chips. It doesn't make any sense to not support as most people don't have the latest AMD chips.

aritger, 2022-11-10:
Sorry, I don't know off hand why the support is limited to those chipsets, but I'll try to find out.

cyberphantom52, 2022-11-10:
> Sorry, I don't know off hand why the support is limited to those chipsets, but I'll try to find out.

We'd really appreciate that, AMD users have not been able to use their nvidia hardware at full potential for so long, i hope it really gets resolved now.

Grimish-ng, 2022-11-13:
Don't worry. Its not exactly like the newer chips with NVIDIA are working very well... or at all on laptops.

Kleidis, 2023-01-02:
> Sorry, I don't know off hand why the support is limited to those chipsets, but I'll try to find out.

Did the bug ever get fixed on 5000 series AMd CPU's
Because i am still having issues as i reported here

https://github.com/NVIDIA/open-gpu-kernel-modules/issues/435

sinanmohd, 2023-02-28:
@aritger 

hi, any updates on AMD 5th Gen cezanne support?, is there any technical limitation that's stopping you guys from supporting it?

aritger, 2023-05-30:
Could I trouble you to retest with 535.43.02?  The release highlights call out:
```
    * Extended Dynamic Boost support on notebooks to include older Renoir
      and Cezanne chipsets, in addition to Rembrandt and newer AMD chipsets.
```
https://www.nvidia.com/Download/driverResults.aspx/205039/en-us/

kleidiss, 2023-06-02:
I forgot to mention but i am indeed using the 535 driver

amrit1711, 2023-06-05:
Dear All,
We have already filed a bug 4142071 for this issue and it has been also root caused.
We will integrate the fix in future release drivers and shall update accordingly.

kleidiss, 2023-06-14:
Hey is this bug fixed in the new production 535 driver?

shseo2023, 2024-02-03:
I keep getting the bug even with 550 version open driver and private driver,my gpu is 3050 laptop and cpu is 5800h.

checkinindza, 2024-02-11:
Seems like the issue is still not resolved. 

HP Omen 15 with Ryzen 7 5800H and RTX 3060.
EndeavourOS with KDE 
Linux Kernel 6.7.4-arch1-1
nvidia-dkms 545.29.06

GPU is stuck at 80W, Dynamic Boost should add 20W. 

nvidia-smi -pl xxx gives an error

`Changing power management limit is not supported in current scope for GPU: 00000000:01:00.0.`

SzczepanRy, 2024-07-07:
Dam guess im going back to js .....