NVIDIA/open-gpu-kernel-modules #393: XPS9520 3050Ti Mobile failed to load driver

himekifee, 2022-10-19:
### NVIDIA Open GPU Kernel Modules Version

520.56.06

### Does this happen with the proprietary driver (of the same version) as well?

I cannot test this

### Operating System and Version

Arch Linux

### Kernel Release

6.0.2-arch1-1

### Hardware: GPU

GeForce RTX 3050 Ti Laptop

### Describe the bug

Might be a simple check mismatch.
```
[  +0.000421] nvidia: probe of 0000:01:00.0 failed with error -1
[  +0.000012] NVRM: The NVIDIA probe routine failed for 1 device(s).
[  +0.000001] NVRM: None of the NVIDIA devices were initialized.
[  +0.000086] nvidia-nvlink: Unregistered Nvlink Core, major device number 508
[  +0.200771] nvidia-nvlink: Nvlink Core is being initialized, major device number 508
[  +0.000005] NVRM: The NVIDIA GPU 0000:01:00.0 (PCI ID: 10de:25a0)
              NVRM: installed in this system is not supported by the
              NVRM: NVIDIA 520.56.06 driver release.
              NVRM: Please see 'Appendix A - Supported NVIDIA GPU Products'
              NVRM: in this release's README, available on the operating system
              NVRM: specific graphics driver download page at www.nvidia.com.
```

### To Reproduce

Install via the official arch repository, reboot, bang.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

Contains sensitive info. If really needed, please ask. Thanks.

aritger, 2022-10-19:
It would help to at least see the complete NVRM messages in dmesg, even if you can't provide the full nvidia-bug-report.log.gz.

Are you sure you have the NVreg_OpenRmEnableUnsupportedGpus kernel module set?  Or does the Arch package set that (or patch the source to remove the need for it) automatically?

Testing with proprietary 520.56.06 release would help isolate the issue.  Thanks.

himekifee, 2022-10-20:
Setting `NVreg_OpenRmEnableUnsupportedGpus` indeed worked. That's kinda weird as my old 2070 laptop also worked without the bit set but whatever, works now.

closed 2022-10-20 by himekifee