NVIDIA/open-gpu-kernel-modules #394: My pc hang if choose nvidia (open kernel) metapackage from nvidia-515-open (propietery,tested)

fhazal, 2022-10-20:
### NVIDIA Open GPU Kernel Modules Version

nvidia (open kernel) metapackage from nvidia-515-open (propietery,tested)

### Does this happen with the proprietary driver (of the same version) as well?

No

### Operating System and Version

Ubuntu 22.04.1 LTS

### Kernel Release

Linux ROG-Ubuntu 5.15.0-52-generic #58-Ubuntu SMP Thu Oct 13 08:03:55 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

### Hardware: GPU

Failed to initialize NVML: Driver/library version mismatch (RTX 2080Ti)

### Describe the bug

my pc will hang when i choose nvidia (open kernel) metapackage from nvidia-515-open (propietery,tested) , i can't do anything 

### To Reproduce

sudo ubuntu-drivers autoinstall
[sudo] password for myusername: 
Traceback (most recent call last):
  File "/usr/bin/ubuntu-drivers", line 513, in <module>
    greet()
  File "/usr/lib/python3/dist-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 84, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/usr/bin/ubuntu-drivers", line 432, in autoinstall
    command_install(config)
  File "/usr/bin/ubuntu-drivers", line 187, in command_install
    UbuntuDrivers.detect.nvidia_desktop_pre_installation_hook(to_install)
  File "/usr/lib/python3/dist-packages/UbuntuDrivers/detect.py", line 839, in nvidia_desktop_pre_installation_hook
    with_nvidia_kms = version >= 470
UnboundLocalError: local variable 'version' referenced before assignment

1. This error come out when i use sudo ubuntu-drivers autoinstall
2. If i choose nvidia (open kernel) metapackage from nvidia-515-open (propietery,tested) in addtional driver my pc will hang and can't click or do anything.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9827105/nvidia-bug-report.log.gz)

vars1ty, 2022-10-20:
Don't expect much help from here, they seem to be merging absolutely nothing and just ignore most issues.

aaronp24, 2022-10-20:
The Python traceback sounds like a downstream problem with Ubuntu's packaging rather than a problem that can be solved by this repository. Looking at your bug report log, I see a variety of errors from API mismatches to errors caused by `NVreg_OpenRmEnableUnsupportedGpus` not being enabled.

Can you please try installing the latest release tag (currently [520.56.06](https://github.com/NVIDIA/open-gpu-kernel-modules/commit/7c345b838b8a63126ed425781400c1ae8a7a4a1d) and the corresponding [user-mode driver components](https://download.nvidia.com/XFree86/Linux-x86_64/520.56.06/)?

awilkins, 2022-10-21:
Confirm that this also happens on 20.04 ; 

When you boot, you just get to a state where there is a low-res text cursor flashing in the top left of a black screen, which blanks out and comes back from time to time.

Is it anything to do with this stuff in the log?

```
Oct 21 10:20:15 asuna kernel: [   23.542777] NVRM: Open nvidia.ko is only ready for use on Data Center GPUs.
Oct 21 10:20:15 asuna kernel: [   23.542780] NVRM: To force use of Open nvidia.ko on other GPUs, see the
Oct 21 10:20:15 asuna kernel: [   23.542781] NVRM: 'OpenRmEnableUnsupportedGpus' kernel module parameter described
Oct 21 10:20:15 asuna kernel: [   23.542782] NVRM: in the README.
<snip>
Oct 21 10:20:16 asuna kernel: [   24.207222] NVRM: GPU 0000:09:00.0: RmInitAdapter failed! (0x63:0x0:1888)
Oct 21 10:20:16 asuna kernel: [   24.208460] NVRM: GPU 0000:09:00.0: rm_init_adapter failed, device minor number 0
```

Dumb question (that should probably be addressed to the Ubuntu maintainers) ; if these drivers only work (without intervention) on datacentre GPUs, why are they marked as the "tested" driver for a desktop operating system?

TLDR : should probably use the proprietary driver package for desktop for a while longer.

fhazal, 2022-10-21:
> The Python traceback sounds like a downstream problem with Ubuntu's packaging rather than a problem that can be solved by this repository. Looking at your bug report log, I see a variety of errors [...]

install both the suggest file u link to? after that what should i do? is it going to add driver in my additional driver and choose from there? and i forgot to mention a multiple error come out GRUB screen such as it not detect all the amd driver or something not authorize at the boot screen.

blackknight36, 2022-10-24:
> Confirm that this also happens on 20.04 ;
>
> When you boot, you just get to a state where there is a low-res text cursor flashing in the top left of a black screen, which blanks out and comes back [...]

Same problem here on a system with a Quadro RTX 5000 card.  I was able to get Xorg working again by downgrading to driver version 510 and removing all traces of 520.

These packages work.

```
ii  libnvidia-cfg1-510:amd64         510.85.02-0ubuntu0.20.04.1 amd64        NVIDIA binary OpenGL/GLX configuration library
ii  libnvidia-common-510             510.85.02-0ubuntu0.20.04.1 all          Shared files used by the NVIDIA libraries
ii  libnvidia-compute-510:amd64      510.85.02-0ubuntu0.20.04.1 amd64        NVIDIA libcompute package
ii  libnvidia-compute-510:i386       510.85.02-0ubuntu0.20.04.1 i386         NVIDIA libcompute package
ii  libnvidia-decode-510:amd64       510.85.02-0ubuntu0.20.04.1 amd64        NVIDIA Video Decoding runtime libraries
ii  libnvidia-decode-510:i386        510.85.02-0ubuntu0.20.04.1 i386         NVIDIA Video Decoding runtime libraries
ii  libnvidia-encode-510:amd64       510.85.02-0ubuntu0.20.04.1 amd64        NVENC Video Encoding runtime library
ii  libnvidia-encode-510:i386        510.85.02-0ubuntu0.20.04.1 i386         NVENC Video Encoding runtime library
ii  libnvidia-extra-510:amd64        510.85.02-0ubuntu0.20.04.1 amd64        Extra libraries for the NVIDIA driver
ii  libnvidia-fbc1-510:amd64         510.85.02-0ubuntu0.20.04.1 amd64        NVIDIA OpenGL-based Framebuffer Capture runtime library
ii  libnvidia-fbc1-510:i386          510.85.02-0ubuntu0.20.04.1 i386         NVIDIA OpenGL-based Framebuffer Capture runtime library
ii  libnvidia-gl-510:amd64           510.85.02-0ubuntu0.20.04.1 amd64        NVIDIA OpenGL/GLX/EGL/GLES GLVND libraries and Vulkan ICD
ii  libnvidia-gl-510:i386            510.85.02-0ubuntu0.20.04.1 i386         NVIDIA OpenGL/GLX/EGL/GLES GLVND libraries and Vulkan ICD
ii  nvidia-compute-utils-510         510.85.02-0ubuntu0.20.04.1 amd64        NVIDIA compute utilities
ii  nvidia-dkms-510                  510.85.02-0ubuntu0.20.04.1 amd64        NVIDIA DKMS package
hi  nvidia-driver-510                510.85.02-0ubuntu0.20.04.1 amd64        NVIDIA driver metapackage
ii  nvidia-kernel-common-510         510.85.02-0ubuntu0.20.04.1 amd64        Shared files used with the kernel module
ii  nvidia-kernel-source-510         510.85.02-0ubuntu0.20.04.1 amd64        NVIDIA kernel source package
ii  nvidia-prime                     0.8.16~0.20.04.2           all          Tools to enable NVIDIA's Prime
ii  nvidia-settings                  470.57.01-0ubuntu0.20.04.3 amd64        Tool for configuring the NVIDIA graphics driver
ii  nvidia-utils-510                 510.85.02-0ubuntu0.20.04.1 amd64        NVIDIA driver support binaries
hi  xserver-xorg-video-nvidia-510    510.85.02-0ubuntu0.20.04.1 amd64        NVIDIA binary Xorg driver
```