NVIDIA/open-gpu-kernel-modules #397: Display power off not functioning

atol71, 2022-10-22:
### NVIDIA Open GPU Kernel Modules Version

Ubuntu latest from "Software & Updates" nvidia-driver-515-open

### Does this happen with the proprietary driver (of the same version) as well?

No

### Operating System and Version

Ubuntu 22.04.1 LTS

### Kernel Release

5.15.0-52-generic

### Hardware: GPU

NVIDIA GeForce RTX 3060 (Asus)

### Describe the bug

The screen is not going into power off mode after given period from settings app.

### To Reproduce

- install driver from Additional Drivers (Software & Updates)
- Reboot
- Wait for Display to power off

### Bug Incidence

Always

### nvidia-bug-report.log.gz

This is bit hard on this issue. Display does not go to power save and pub is in inter-operation between Gnome and driver.

aaronp24, 2022-10-26:
Can you please attach a bug report log? It's not clear just from this report whether GNOME is actually requesting that the monitor power off. Does the monitor turn off if you trigger it manually with `sleep 5; xset dpms force off`?

It's also possible that your mouse is jittering and resetting the screen timer. You can use `xinput` to list the available input devices and then `xinput test '<device name>'` to watch for events from your mouse.

atol71, 2025-03-22:
The current driver only occasionally bugs this way 1-2 times a month.

closed 2025-03-22 by atol71