NVIDIA/open-gpu-kernel-modules #401: Application Window Artifacting in KDE Plasma

ThisNekoGuy, 2022-11-10:
### NVIDIA Open GPU Kernel Modules Version

520.56.06

### Does this happen with the proprietary driver (of the same version) as well?

I cannot test this

### Operating System and Version

Arch Linux

### Kernel Release

(User-compiled / User-built) [Linux ARCH 6.0.1-270-tkg-pds #1 TKG SMP PREEMPT_DYNAMIC Thu, 13 Oct 2022 19:23:40 +0000 x86_64 GNU/Linux](https://github.com/frogging-family/linux-tkg)

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 2080 Ti (UUID: GPU-6e80fac2-88da-6bda-dd46-6b2803c67aea)

### Describe the bug

There's an issue that can occur that causes multiple open application windows to rapidly "flip" into view, even artifacting through previous frames to partially render themselves. Additionally, it also inverted some windows, and their window decorations, so that they were displayed upside-down (more easily noticed with the browser window running Discord) while this occurs.

I recognize that this can be difficult to understand through words, so IO managed to get a screen recording of what it looks like when this happens:

NOTE:
The artifacting is pretty aggressive, so don't watch the video if you're medically prone to flashing images.

[screen-recording cloud link](https://mega.nz/file/gvp3gBCQ#dqSMiBBZ8QKtcsbTT0LR8qudxnu3P5l2YWj0Nrv-Nyo)
(The file was too big to upload directly, so I used Mega cloud storage to record it)

-----

**Desktop Environment info:**
_Plasma version:_ 5.26.3
_KDE Frameworks version:_ 5.99.0
_Qt Version:_ 5.15.7
_Window Manager:_ Kwin

### To Reproduce

I'm not entirely sure how to replicate this in detail; it's happened to me several times with this version but there's one thing I do know that triggers it:
it happens every time I have the system running for a few days, I play some games (namely Persona 5 Royal via Proton), and exiting the game happens to trigger this odd issue.

That aside, I have this script run on KDE Plasma's startup in `~/.config/plasma-workspace/env/nvidia-zink.sh`:
```bash
#! /usr/bin/bash

if [ -f /usr/lib/libGLX_mesa.so ] && [ -x "$(command -v nvidia-smi > /dev/null 2>&1)" ]; then
    export __GLX_VENDOR_LIBRARY_NAME="mesa"
    export MESA_LOADER_DRIVER_OVERRIDE="zink"
    export GALLIUM_DRIVER="zink"
fi
```
Although I must note this might not even be required; I recall attempting to use zink as a solution to this issue just to find that it apparently made no difference. Though, I also can't say for certain that Plasma is even respecting this script and doing as I instructed, thus this might not even be because of zink at all which is why I suspect this to be a driver issue.

So, in bullet-point format, that would be:

- Have KDE Plasma installed
- Place the above script at `~/.config/plasma-workspace/env/`
- Restart the system
- Log in
- Use the system for a few days, play games regularly (in my case, Persona 5 Royal on Steam)
- Exit a game and have this issue start suddenly occurring

### Bug Incidence

Sometimes

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9979166/nvidia-bug-report.log.gz)

unhappy-ending, 2022-11-13:
The default configuration of open kernel is to use the proprietary OpenGL driver.  You're running Mesa's Zink OpenGL implementation on top of the proprietary Nvidia Vulkan driver. Do the artifacts occur without using Zink? 

AFAIK, ZInk isn't supposed to be ready for something like a fully composited Plasma desktop. If the artifacts don't occur when not using Zink, then it's a Zink issue. Could be specifically Zink on Nvidia, Zink on kernel-open driver issue, or otherwise. That would probably need reported in the Mesa Zink bugtracker.

ThisNekoGuy, 2022-11-13:
I'll turn it off again and see if it happens without it; I can't exactly remember if this was the particular reason why I started doing it, looking for a solution for it or something else to stop

I'll be back to check in
In the meantime, I'll try to mention this on the Zink side and see if someone might know something about this already or not
Aside from that, I don't typically have issues with Zink running otherwise though but I haven't extensively checked it out

unhappy-ending, 2022-11-13:
This is 7 months old but doesn't seem to have been updated.

https://gitlab.freedesktop.org/mesa/mesa/-/issues/6150

Mike, the coder being Zink, wrote:

"Oh, you're running the compositor with zink? Yeah, that's not going to work ever on any hardware/distro right now."

FWIW the proprietary OpenGL stack worked fine on the open kernel module when I tried it out with Plasma. I never got Zink to work with it, IMO your problem is Zink not the Nvidia driver.

ThisNekoGuy, 2022-11-13:
They were using SteamOS 3, and that only until **_very_** recently on their beta builds, used ancient packages - and 7 months ago, they might have been even older

But I don't doubt that what you're saying might be true, but I've not seen any behavior otherwise look even remotely that bad

unhappy-ending, 2022-11-17:
@ThisNekoGuy did you test this without Zink?

ThisNekoGuy, 2022-11-17:
So far since then, I haven't experienced it without Zink, no

ThisNekoGuy, 2022-11-17:
Nevermind, scratch that, I just had it happen again

unhappy-ending, 2022-11-17:
I used your script in the folder you placed it in and it made no difference to the Plasma session so we can rule out Zink and I found it odd that you'd even be able to run a session on Zink. KDE Info Center details under Graphics > Window Manager that Kwin is still using Nvidia's implementation of OpenGL/EGL.

I'm on a similar setup to you as far as KDE Frameworks, KDE Plasma, and Qt versions are concerned. Have you tried a default kernel to test against? Do you have the option to install a binary kernel?

ThisNekoGuy, 2022-11-17:
I can use a repo kernel, I just tend not to both for gaming performance reasons and because it's kind of a hobby of mine

unhappy-ending, 2022-11-17:
I'd try to test your custom kernel vs a default repo kernel, and then the proprietary module vs the open module. You want to have a baseline to test off, so being able to load up a default system config is necessary. Get as default as you can, and then try to find where something custom you did changes behavior.

ThisNekoGuy, 2022-11-17:
Honestly, I didn't really deviate that much from a default build; by that, I mean I didn't actually touch kernel graphics options at all, I mainly only removed industrial hardware support, Intel CPU support, and touchscreen support

I can try what you suggested but everything related is already default settings, specifically because I already know that Nvidia acts weird if you touch those

That aside... I did try the proprietary module for some time to test that already but I actually found that my game performance was unexpectedly _better_ with the Open-GPU module... so I'm not really compelled (as a user) to settle for sub-par frames...

ThisNekoGuy, 2022-12-16:
More information on this, it continues in 525.60.11, and I found that disabling the KDE Plasma compositor/Kwin with `SHIFT+ALT+F12` stops the strange behavior but re-enabling it causes the issue to appear again.
If it helps to know: Kwin is being run with OpenGL 2

ThisNekoGuy, 2023-08-16:
Plasma can be run via Zink with driver 535.98 now, but Kwin crashes whenever composition is enabled.
Also had to put these in `/etc/environment`:
```bash
export KWIN_OPENGL_INTERFACE=glx
export __GLX_VENDOR_LIBRARY_NAME=mesa
export MESA_LOADER_DRIVER_OVERRIDE=zink
```
Additionally, Steam is very angry about Nvidia+Zink:

<details><summary>Screenshots:</summary>

[image]

Steam:
[image]
</details>