NVIDIA/open-gpu-kernel-modules #402: nvlogAllocBuffer wrap failed

eugeneRover, 2022-11-11:
### NVIDIA Open GPU Kernel Modules Version

510.68.02

### Does this happen with the proprietary driver (of the same version) as well?

I cannot test this

### Operating System and Version

Debian GNU/Linux 11 (bullseye)

### Kernel Release

Linux kanazawa 5.10.0-18-amd64 #1 SMP Debian 5.10.140-1 (2022-09-02) x86_64 GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3050 Ti Laptop GPU (UUID: GPU-8cee1145-fc68-ffcd-427e-91de4ee6e8a9)

### Describe the bug

System repeatedly freezes for a short period of time (less than a second).
In that moment, `/var/log/messages` shows pair of `nvlogAllocBuffer nowrap/wrap failed` lines:

    Nov 11 08:47:55 kanazawa kernel: [ 6476.967242] nvlogAllocBuffer nowrap failed
    Nov 11 08:47:55 kanazawa kernel: [ 6476.967243] nvlogAllocBuffer wrap failed
    Nov 11 08:50:00 kanazawa kernel: [ 6602.538093] nvlogAllocBuffer nowrap failed
    Nov 11 08:50:00 kanazawa kernel: [ 6602.538094] nvlogAllocBuffer wrap failed
    Nov 11 08:51:27 kanazawa kernel: [ 6688.797467] nvlogAllocBuffer nowrap failed
    Nov 11 08:51:27 kanazawa kernel: [ 6688.797469] nvlogAllocBuffer wrap failed
    Nov 11 08:51:48 kanazawa kernel: [ 6709.800914] nvlogAllocBuffer nowrap failed
    Nov 11 08:51:48 kanazawa kernel: [ 6709.800915] nvlogAllocBuffer wrap failed
    Nov 11 08:54:44 kanazawa kernel: [ 6886.185008] nvlogAllocBuffer nowrap failed
    Nov 11 08:54:44 kanazawa kernel: [ 6886.185010] nvlogAllocBuffer wrap failed

There is no constant interval between such events.

### To Reproduce

No special reproduce steps found

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9986657/nvidia-bug-report.log.gz)

### More Info

I know, 510.68.02 is rather old release, but it possible that the problem persists with new versions.
I found that mentioned error messages come from [liblogdecode.c](https://github.com/NVIDIA/open-gpu-kernel-modules/blob/main/src/common/uproc/os/libos-v3.1.0/lib/liblogdecode.c).
After some browsing, I think disabling `LIBOS_LOG_TO_NVLOG` or even `LIBOS_LOG_DECODE_ENABLE` flags will help to get rid of the errors.
Would you please describe what functionality the flags are responsible for? Is it sensible to disable them ?
Thank you!

ngolovliov-nv, 2022-11-11:
```
[    3.311755] NVRM: loading NVIDIA UNIX x86_64 Kernel Module  510.68.02  Wed Apr 20 21:10:34 UTC 2022
```

You are running the proprietary driver, so you won't be able to recompile liblogdecode.c.

Does the issue reproduce with the latest driver version?

VirMaker, 2022-11-11:
@ngolovliov-nv I can reproduce this issue on 
`[    3.164947] NVRM: loading NVIDIA UNIX x86_64 Kernel Module  515.65.01  Wed Jul 20 14:00:58 UTC 2022`
I'm on ubuntu 22.10

VirMaker, 2022-11-12:
Quick update. After disabling armory crate in BIOS on my Asus TUF A15 I no longer see these messages.

eugeneRover, 2022-11-13:
@ngolovliov-nv After update to 515.76 (downloaded from https://www.nvidia.ru/Download/driverResults.aspx/193095/en-us) the error persists.

ngolovliov-nv, 2022-11-15:
Can you try with version 525.53, both proprietary and open source?

In case it reproduces with the open source driver, can you try applying the following patch? This will help us diagnose the error.
```
diff --git a/src/common/uproc/os/libos-v3.1.0/lib/liblogdecode.c b/src/common/uproc/os/libos-v3.1.0/lib/liblogdecode.c
index e421339..788b40c 100644
--- a/src/common/uproc/os/libos-v3.1.0/lib/liblogdecode.c
+++ b/src/common/uproc/os/libos-v3.1.0/lib/liblogdecode.c
@@ -28,6 +28,7 @@
 
 #    include <core/core.h>
 #    include <stddef.h> // size_t
+#include "os/os.h"
 
 #    define printf(fmt, ...) portDbgExPrintfLevel(LEVEL_ERROR, fmt, ##__VA_ARGS__)
 #    define snprintf         nvDbgSnprintf
@@ -1378,7 +1379,8 @@ void libosLogAddLogEx(LIBOS_LOG_DECODE *logDecode, void *buffer, NvU64 bufferSiz
         }
         else
         {
-            printf("nvlogAllocBuffer nowrap failed\n");
+            printf("nvlogAllocBuffer nowrap failed, status = 0x%x\n", status);
+            osAssertFailed();
         }
     }
     else
@@ -1416,7 +1418,8 @@ void libosLogAddLogEx(LIBOS_LOG_DECODE *logDecode, void *buffer, NvU64 bufferSiz
     }
     else
     {
-        printf("nvlogAllocBuffer wrap failed\n");
+        printf("nvlogAllocBuffer wrap failed, status = 0x%x\n", status);
+        osAssertFailed();
     }
 #endif // LIBOS_LOG_TO_NVLOG
 }
```

`LIBOS_LOG_TO_NVLOG` is responsible for collecting firmware diagnostic logs. Disabling this flag is not recommended.

eugeneRover, 2022-11-17:
Installed 525.53 version, proprietary kernel ([download link](https://us.download.nvidia.com/XFree86/Linux-x86_64/525.53/NVIDIA-Linux-x86_64-525.53.run))
Bug is not reproduced.

Tomorrow will try open-source kernel with your patch

eugeneRover, 2022-11-20:
Applied your patch, built and installed open-kernel 525.53.
```
$ cat /proc/driver/nvidia/version
NVRM version: NVIDIA UNIX Open Kernel Module for x86_64  525.53  Release Build  (user@kanazawa)  Вс 20 ноя 2022 09:42:24 +04
GCC version:  gcc version 10.2.1 20210110 (Debian 10.2.1-6) 
```

No `nvlogAllocBuffer failed` messages now, and freezes are gone.
I think I will stay on the version. 

@ngolovliov-nv  Thanks for the assistance, issue seems to be resolved.

closed 2022-11-21 by ngolovliov-nv

eugeneRover, 2022-11-24:
Here is what appeared in log recently:

Nov 24 22:57:55 kanazawa kernel: [142500.501028] acer_wmi: Unknown function number - 4 - 0
Nov 24 22:57:55 kanazawa kernel: [142500.501070] NVRM nvAssertFailedNoLog: Assertion failed: 0 @ osapi.c:1688
Nov 24 22:57:56 kanazawa kernel: [142501.800546] acer_wmi: Unknown function number - 4 - 0
Nov 24 22:57:56 kanazawa kernel: [142501.800573] NVRM nvAssertFailedNoLog: Assertion failed: 0 @ osapi.c:1688
Nov 24 22:57:56 kanazawa kernel: [142501.992629] acer_wmi: Unknown function number - 4 - 0
Nov 24 22:57:56 kanazawa kernel: [142501.992669] NVRM nvAssertFailedNoLog: Assertion failed: 0 @ osapi.c:1688
Nov 24 22:57:57 kanazawa kernel: [142502.200619] acer_wmi: Unknown function number - 4 - 0
Nov 24 22:57:57 kanazawa kernel: [142502.200647] NVRM nvAssertFailedNoLog: Assertion failed: 0 @ osapi.c:1688

I nevertheless see no freezes.
Hope it helps.