NVIDIA/open-gpu-kernel-modules #405: nvidia 520.56.06 dkms fails to build for kernel compiled with clang/lto

ltsdw, 2022-11-13:
### NVIDIA Open GPU Kernel Modules Version

520.56.06

### Operating System and Version

Arch Linux x86_64

### Kernel Release

Linux archlinux 6.0.8-2-ltsdw #2 SMP PREEMPT_DYNAMIC Sun, 13 Nov 2022 02:59:59 +0000 x86_64 GNU/Linux (custom kernel)

### Build Command

pacman -S nvidia-open-dkms

### Terminal output/Build Log

[make.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9997454/make.log)

I think the interesting part is:

```
objcopy: /var/lib/dkms/nvidia/520.56.06/build/kernel-open/nvidia-modeset.ko: file format not recognized
btf_encoder__write_elf: failed to add .BTF section to '/var/lib/dkms/nvidia/520.56.06/build/kernel-open/nvidia-modeset.ko': 2!
Failed to encode BTF
make[4]: *** [scripts/Makefile.modfinal:60: /var/lib/dkms/nvidia/520.56.06/build/kernel-open/nvidia-modeset.ko] Error 1
make[4]: *** Deleting file '/var/lib/dkms/nvidia/520.56.06/build/kernel-open/nvidia-modeset.ko'
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [scripts/Makefile.modpost:134: __modpost] Error 2
make[2]: *** [Makefile:1766: modules] Error 2
make[2]: Leaving directory '/usr/lib/modules/6.0.8-2-ltsdw/build'
make[1]: *** [Makefile:82: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/nvidia/520.56.06/build/kernel-open'
make: *** [Makefile:59: modules] Error 2
```

### More Info

The modules compiles normally if I use gcc.

Also I don't think it's related to #214 and #377.

ltsdw, 2022-11-14:
Thank you, so, I saw that too and I thought it wasn't related to my case because apparently the objcopy was linked to another place, right?

I think mine looks right, isn't it?

```
pacman -Q binutils
binutils 2.39-3

type -a objcopy
/usr/bin/objcopy

ls -lha /usr/bin/objcopy 
160K -rwxr-xr-x 1 root root 159K ago 19 21:18 /usr/bin/objcopy
```

So I tried building the modules manually from git, but I think I'm doing something wrong.

So, what I did:

```
$ make modules -j$(nproc) CC=clang CXX=clang++ LD="/usr/bin/ld.lld"
# make modules_install -j$(nproc) CC=clang CXX=clang++ LD="/usr/bin/ld.lld"
```

The logs doesn't seem to show any error, even though, I think no modules are being installed (I can't do `modprobe nvidia` or `prime-run glxinfo`).

[make_modules.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10007474/make_modules.log)
[make_modules_install.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10007475/make_modules_install.log)

aritger, 2022-11-14:
What does `/usr/bin/objcopy --version` report?

I'm not positive whether kbuild will use `$(OBJCOPY)` versus just `objcopy`, so I don't know if the experiment of building with `OBJCOPY="/usr/bin/llvm-objcopy"` is sufficient to rule out a problem due to using the wrong objcopy.

ltsdw, 2022-11-14:
```
$ objcopy --version
GNU objcopy (GNU Binutils) 2.39.0
Copyright (C) 2022 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.
```

aritger, 2022-11-14:
Does the build fare any better if `objcopy` is llvm-objcopy?  E.g., temporary uninstall binutils, or update your PATH so that an objcopy symlink points to llvm-objcopy, or similar?

ltsdw, 2022-11-14:
```
$ ls -lha /usr/bin/objcopy 
0 lrwxrwxrwx 1 root root 21 nov 14 19:44 /usr/bin/objcopy -> /usr/bin/llvm-objcopy
[ dev_user @ shadow open-gpu-kernel-modules ]$ 

$ objcopy --version
llvm-objcopy, compatible with GNU objcopy
LLVM (http://llvm.org/):
  LLVM version 14.0.6
  Optimized build.
  Default target: x86_64-pc-linux-gnu
  Host CPU: tigerlake

$ make modules -j$(nproc) TARGET_ARCH=x86_64 CC=clang CXX=clang++ LD="/usr/bin/ld.lld" AR="/usr/bin/llvm-ar" OBJCOPY="/usr/bin/objcopy"

# make modules_install -j$(nproc) TARGET_ARCH=x86_64 CC=clang CXX=clang++ LD="/usr/bin/ld.lld" AR="/usr/bin/llvm-ar" OBJCOPY="/usr/bin/objcopy"
```

No difference. Logs the same, and no modules installed (at least I think there's no module installed).

`find -type f -name "*nvidia*"` doesn't show nothing at `/usr/lib/modules/6.0.8-2-ltsdw`

ltsdw, 2022-11-14:
Like, I'm saying the logs are the same, but I mean that they are the same for building manually, not the same log reported with dkms, I don't know if I made myself clear there 😄.

At least building manually I don't see any `btf_encoder__write_elf: failed to add .BTF section to ...
Failed to encode BTF`

aritger, 2022-11-14:
Can you please provide a log of building manually, where objcopy is llvm's, not binutils?

ltsdw, 2022-11-15:
[make_modules.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10007941/make_modules.log)
[make_modules_install.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10007942/make_modules_install.log)

It seems to me the same as the binutils.

aritger, 2022-11-15:
Oh, I think the error is this:
```
[ nvidia-modeset    ]  LD           _out/Linux_x86_64/maxwell_shaders.xz.o
[...]
ld.lld: error: target emulation unknown: -m or at least one .o file required
make[1]: *** [Makefile:189: _out/Linux_x86_64/maxwell_shaders.xz.o] Error 1
```
Can you test if this change fixes the problem?
```
$ git diff
diff --git a/utils.mk b/utils.mk
index 17421ba9f840..749d09da8ec1 100644
--- a/utils.mk
+++ b/utils.mk
@@ -569,7 +569,7 @@ define READ_ONLY_OBJECT_FROM_FILE_RULE
   $$(OUTPUTDIR)/$$(notdir $(1)).o: $(1)
        $(at_if_quiet)$$(MKDIR) $$(OUTPUTDIR)
        $(at_if_quiet)cd $$(dir $(1)); \
-       $$(call quiet_cmd_no_at,LD) -r -z noexecstack --format=binary \
+       $$(call quiet_cmd_no_at,LD) -r -z noexecstack -m elf_x86_64 --format=binary \
            $$(notdir $(1)) -o $$(OUTPUTDIR_ABSOLUTE)/$$(notdir $$@)
        $$(call quiet_cmd,OBJCOPY) \
            --rename-section .data=.rodata,contents,alloc,load,data,readonly \
```
(update the READ_ONLY_OBJECT_FROM_FILE_RULE macro in utils.mk to pass `-m elf_x86_64` to LD?)

ltsdw, 2022-11-15:
Ok, now the build progresses a little bit more, but still fails.

[make_modules.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10008091/make_modules.log)

Now I think I hit the #214

ptr1337, 2022-11-15:
Which pahole version you are using?

What is happening if you use the default arch linux kernel? Does it work then?

ltsdw, 2022-11-15:
```
$ pahole --version
v1.24
```

Hi there, with the default arch linux kernel it compiles normally, because the default arch linux kernel was compiled with gcc, if I compile the custom kernel with gcc then the modules compile too, as I explained in one of the posts above.

ptr1337, 2022-11-15:
Oh, my bad. It would require to compile the arch/zen kernel with llvm.
I think the closed drivers are working correctly with it? I use them, and all kernels (gcc,gcc lto, clang thinlto,clang fulllto) are working fine.

I think it is now a equal issue to 214. He has tested there also a patch which I posted, this change the thinlto.cache patch, which fixes the permission issue. But then he did also run into a equal issue as you have.

I hope that will be fixed soon. Same as the llvm 15 issue, which is still present at 525.

ltsdw, 2022-11-15:
yeah the closed drivers are working fine, yup.

I hope that too.

unhappy-ending, 2022-11-17:
Hello, would like to chime in that passing `LLVM=1` on the command line prior to building your linux kernel will make sure the kernel build uses the entire LLVM toolchain, i.e. clang, llvm-ar, llvm-nm, llvm-objcopy, lld and so on so you don't have to pass CC and the likes. That way you can know for sure whether or not llvm-objcopy is being used for the kernel.

Nvidia drivers should also have the same build environment as much as possible:

CC=clang    \
LD=ld.lld     \
AR=llvm-ar     \
CXX=clang++   \
OBJCOPY=llvm-objcopy

If you missed any of those, it could cause a mismatch. I did see in your build log a CC mismatch error, right after CXX run that it didn't match the CC used in the kernel even though both reported clang. I have a clang built kernel and system, when I have a chance I'll try building the open kernel as well and see what I come up with.

If possible, it would be nice if Nvidia could add the environment variable `LLVM=1` to use the LLVM toolchain exactly as the Linux kernel does.

ltsdw, 2022-11-17:
@unhappy-ending hi there, so, I'm using LLVM=1
in short I do something like this to compile my kernel
```
make LLVM=1 LLVM_IAS=1 -j$(nproc) olddefconfig
./scripts/config -e LTO_CLANG_THIN
make LLVM=1 LLVM_IAS=1 -j$(nproc) nconfig
```

unhappy-ending, 2022-11-17:
You don't need to pass LLVM_IAS=1 anymore, it's already on by default.

https://www.kernel.org/doc/html/latest/kbuild/llvm.html?highlight=llvm

`"The integrated assembler is enabled by default. You can pass LLVM_IAS=0 to disable it."`

Your kernel command line looks good. You're passing all the required environment variables to make when building the nvidia drivers? Not just CC and LD? Have you tried NV_VERBOSE?

ltsdw, 2022-11-17:
> You don't need to pass LLVM_IAS=1 anymore, it's already on by default.
> 
> https://www.kernel.org/doc/html/latest/kbuild/llvm.html?highlight=llvm
> 
> `"The integrated assembler is enabled by default. You can pass LLVM_IAS=0 to disable it."`

Thank you, didn't know about that.

> Your kernel command line looks good. You're passing all the required environment variables to make when building the nvidia drivers? Not just CC and LD? Have you tried NV_VERBOSE?

Yes, I mentioned it in one of the posts above.

```
make modules -j$(nproc) TARGET_ARCH=x86_64 CC=clang CXX=clang++ LD="/usr/bin/ld.lld" AR="/usr/bin/llvm-ar" OBJCOPY="/usr/bin/llvm-objcopy" NV_VERBOSE=1
```

The same error with or without all the envars, sadly.

ltsdw, 2022-11-19:
I don't know, maybe it's a Arch Linux thingy then: [[1/1] Kbuild, clang: add option for choosing a ThinLTO cache directory](https://patchwork.kernel.org/project/linux-kbuild/patch/b45b2430-3670-b310-b6ad-2d6db50c2d18@mailbox.org/#24308209)

Also tried specifying the LDFLAGS="--thinlto-cache-dir=some/path/here", as said by the man page of ld.lld:

```
[...]
--thinlto-cache-dir=value
             Path to ThinLTO cached object file directory.
[...]
```

But I think that is supposed to tell the linker where are the thinlto cache files and not where they should be put.

unhappy-ending, 2022-11-20:
My kernel is built with full LTO, I can try thinLTO later. FWIW I've never had to specify a cache for LTO, but I will report back after trying thin.

Have you tried using a full LTO kernel?

ltsdw, 2022-11-20:
Nope, haven't tried yet, I'll recompile later and test it out, thank you.

ltsdw, 2022-11-20:
@unhappy-ending you're right, with full lto the modules compile normally. It does still fails for dkms though.

ltsdw, 2022-11-20:
I think I'll be closing this issue, there's already the #214 to keep track of the problem with thin lto.

And my initial problem was with dkms, and it seems not to be an issue with the nvidia open kernel modules, but with dkms, because if I symlink /usr/bin/llvm-objcopy to /usr/bin/objcopy, then dkms can compile and install the modules too.

closed 2022-11-20 by ltsdw