NVIDIA/open-gpu-kernel-modules #409: Backlight problem

GrayYip, 2022-11-22:
### NVIDIA Open GPU Kernel Modules Version

525.53

### Does this happen with the proprietary driver (of the same version) as well?

I cannot test this

### Operating System and Version

ArchLinux

### Kernel Release

5.10.155-1-lts510

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3060 Laptop GPU (UUID: GPU-405b1dbf-3008-c44f-9f11-0513aaf6746d)

### Describe the bug

I tested the latest open source driver and found that the backlight works.

echo 10 > brightness ，the screen is at its brightest

cat max_brightness ,it is 100

### To Reproduce

none

### Bug Incidence

Always

### nvidia-bug-report.log.gz

none

nvaert1986, 2022-12-02:
I don't think this is related to the open-gpu-kernel-modules, as I'm experiencing the same  exact same issue with a Lenovo ThinkPad P52 and I'm not using the open-gpu-kernel-modules, as it's not supported on Pascal based hardware: 01:00.0 VGA compatible controller: NVIDIA Corporation GP107GLM [Quadro P1000 Mobile] (rev a1). I have the exact same issue. It doesn't work with 525.xx, but works fine with 515.xx. I made a report on the forums too: https://forums.developer.nvidia.com/t/brightness-on-a-lenovo-thinkpad-p52-no-longer-controllable-after-updating-nvidia-drivers-to-525-xx-series/236008/2

LRitzdorf, 2022-12-09:
I also seem to be experiencing a similar issue, and have posted further details at https://bbs.archlinux.org/viewtopic.php?pid=2072664.

In short, brightness levels from 0 to 10 are scaled (in a non-linear fashion) to produce actual brightnesses equivalent to the 0 to 100 range, and then levels from 11 to 100 work as expected.

I observed this issue appearing with 525.60.11, while the previous Arch release (520.56.06) was unaffected.

dsrafsvlho, 2022-12-09:
I have the same problem. 
P15 with nvidia-520.56.06-12-x86_64 and NVIDIA T1000 with Archlinux.
Tried both linux and linux-lts have backlight problem on both. 
Currently roll-back to: 
linux-lts-5.15.80-1-x86_64 (lts that I usually use) and linux-6.0.10.arch2-1-x86_64
and
nvidia-lts-1:520.56.06-10-x86_64 and nvidia-520.56.06-12-x86_64

LRitzdorf, 2022-12-09:
I can confirm that (my manifestation of) the issue is independent of kernel version as well. It was resolved by rolling back to driver 520.56.06, with no changes to my kernel.

LRitzdorf, 2023-01-13:
BTW, this is still a problem with driver version 525.78.01, which was recently released to the Arch repos...

smazur, 2023-01-19:
Same problem. Fixed after switching to integrated amd gpu.

LRitzdorf, 2023-02-28:
@ngolovliov-nv Apologies for the ping here, but this issue is fairly significant and has received minimal attention. Even if you're not the right person to handle this, would you mind bringing it up internally, or tagging the appropriate assignee here? We'd be happy to provide further details to aid in troubleshooting.

For context: I was, until recently, able to avoid this issue by using 520.56 (via `nvidia-dkms` on Arch), but today's upgrade to kernel 6.2 breaks the old DKMS module. In order to keep my kernel up to date, then, I'm forced to use a version of this driver which manifests the backlight issue.

dsrafsvlho, 2023-03-01:
The same problem: upgrade to kernel 6.2 breaks the old DKMS module. In order to keep my kernel up to date, then, I'm forced to use a version of this driver which manifests the backlight issue.

parmi93, 2023-05-12:
I have same issue on system:
Dell-G16 
CPU: i9-13900HX (iGPU disabled from BIOS)
GPU: RTX 4070, using NVIDIA proprietary driver 530.41.03 (installed from Software & Updated tools)
OS: Ubuntu 22.04.2 LTS (kernel: 5.19.0-41-generic)

albertnetymk, 2023-06-16:
Backlight (Brightness) keys are completely useless on fresh start-up but appear to kind of function (as described in #428) after resuming from sleep. Debian 12 + nvidia-driver 525.105.17 on Thinkpad P53.

LRitzdorf, 2023-06-17:
Still an issue with driver 535.54.03 and kernel 6.3.8. Nvidia, can we please get some kind of response to this? Even if you don't have enough information to work with at the moment, we as a community would love to help you fix that!

LRitzdorf, 2023-06-22:
Actually, we seem to be fixed now! Not sure exactly what changed, but I'm on driver 535.54.03 and the backlight control range is correct again.Note that I'm on Arch, so for anyone experiencing this problem on another distro, it may take a minute for this driver to arrive.
...thanks, Nvidia, I guess?

Note that I'm on Arch, so for anyone experiencing this problem on another distro, it may take a minute for this driver to reach your package repos.

joseluisq, 2023-06-23:
For my HP Zbook G8 (`NVIDIA RTX A2000`) with Arch Linux (Kernel `6.3.9`) still not working using the latest `535.54.03`.

```sh
$ uname -a
# Linux 6.3.9-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 21 Jun 2023 20:46:20 +0000 x86_64 GNU/Linux

$ lspci -k | grep -A 2 -E "(VGA|3D)"
# 01:00.0 VGA compatible controller: NVIDIA Corporation GA107GLM [RTX A2000 Mobile] (rev a1)
# 	Subsystem: Hewlett-Packard Company GA107GLM [RTX A2000 Mobile]
# 	Kernel driver in use: nvidia

$ glxinfo | grep NVIDIA
# server glx vendor string: NVIDIA Corporation
# client glx vendor string: NVIDIA Corporation
# OpenGL vendor string: NVIDIA Corporation
# OpenGL renderer string: NVIDIA RTX A2000 Laptop GPU/PCIe/SSE2
# OpenGL core profile version string: 4.6.0 NVIDIA 535.54.03
# OpenGL core profile shading language version string: 4.60 NVIDIA
# OpenGL version string: 4.6.0 NVIDIA 535.54.03
# OpenGL shading language version string: 4.60 NVIDIA
# OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 535.54.03

$ dmesg | grep xhci
# [    8.576205] xhci_hcd 0000:00:0d.0: xHCI Host Controller
# [    8.576209] xhci_hcd 0000:00:0d.0: new USB bus registered, assigned bus number 1
# [    8.577270] xhci_hcd 0000:00:0d.0: hcc params 0x20007fc1 hci version 0x120 quirks 0x0000000200009810
# [    8.577496] xhci_hcd 0000:00:0d.0: xHCI Host Controller
# [    8.577498] xhci_hcd 0000:00:0d.0: new USB bus registered, assigned bus number 2
# [    8.577499] xhci_hcd 0000:00:0d.0: Host supports USB 3.1 Enhanced SuperSpeed
# [    8.577537] usb usb1: Manufacturer: Linux 6.3.9-arch1-1 xhci-hcd
# [    8.577994] usb usb2: Manufacturer: Linux 6.3.9-arch1-1 xhci-hcd
# [    8.579040] xhci_hcd 0000:00:14.0: xHCI Host Controller
# [    8.579045] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 3
# [    8.580218] xhci_hcd 0000:00:14.0: hcc params 0x20007fc1 hci version 0x120 quirks 0x0000000200009810
# [    8.580574] xhci_hcd 0000:00:14.0: xHCI Host Controller
# [    8.580576] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 4
# [    8.580578] xhci_hcd 0000:00:14.0: Host supports USB 3.2 Enhanced SuperSpeed
# [    8.580618] usb usb3: Manufacturer: Linux 6.3.9-arch1-1 xhci-hcd
# [    8.583425] usb usb4: Manufacturer: Linux 6.3.9-arch1-1 xhci-hcd
# [    9.583594] usb 3-7: new high-speed USB device number 3 using xhci_hcd
# [   10.043598] usb 3-8: new full-speed USB device number 4 using xhci_hcd
# [   10.310115] usb 3-14: new full-speed USB device number 5 using xhci_hcd
```

tukusejssirs, 2023-06-25:
On my system, it seems to be fixed too (albeit the screen started blinking a lot recently … again). I use Arch Linux with:

- Lenovo ThinkPad P15 Gen 1;
- NVIDIA Quadro T1000 Mobile (TU117GLM);
- `linux@6.3.9-arch1-1`;
- `nvidia-open-dkms@535.54.03-3`;
- `nvidia-utils@535.54.03-1`;
- `sway-git@r7153.7fbd9fbf-1` (`sway version 1.9-dev-7fbd9fbf (Jun 24 2023, branch 'master')`);
- `wlroots-git@0.17.0.r6436.71b57b8d-1`.

mamaraddio, 2023-09-25:
Same problem on  a Lenovo Ideapad L340 with NVIDIA GTX 1650 Max-Q running ArchLinux and Hyprland.
The issue is present with a bunch of kernel version and with both open and proprietary, dkms and not dkms drivers. Always the same issue...
With brightness from 1 to 10 it behave like brightness was multiplied 10.
With brightness from 11 to 100 it behaves normal. So I decided to write a simple bash script to avoid to go down level 11 to not kill my eyes passing from 11 (very dark) to 10 corresponding to max brightness.
If anyone could be interested take a look at my "[nvidi-backlight](https://github.com/mamaraddio/nvidia-backlight)" script

dsrafsvlho, 2023-11-08:
Seems after the last update it comes back, backlight on P15 with Intel Core i7-10750H CPU @ 2.60GHz x 12 and NVIDIA T1000 with Archlinux doesn't work anymore (it was fixed in Jun 22)

joseluisq, 2023-11-28:
I just created a small tutorial to set the brightness and gamma colors for folks using Debian 12.
https://gist.github.com/joseluisq/05b593896f76a2cfdddead5d627c7c6b

Jesus82, 2024-12-05:
Having the same issue (lightness controls appear, but no effect when changing). Also tried with light and brightess utility, but no luck.

I have a Dell G15 with a Nvidia GeForce RTX 3060 Mobile, with drivers `NVIDIA-SMI 560.35.03`  running on ubuntu 24.10

`ls -l /sys/class/backlight`

```
total 0
lrwxrwxrwx 1 root root 0 Dec  4 21:34 nvidia_wmi_ec_backlight -> ../../devices/pci0000:00/PNP0C14:00/wmi_bus/wmi_bus-PNP0C14:00/603E9613-EF25-4338-A3D0-C46177516DB7/backlight/nvidia_wmi_ec_backlight
```

`sudo lspci -k | grep -EA3 'VGA|3D|Display'`

```
pcilib: Error reading /sys/bus/pci/devices/0000:00:08.3/label: Operation not permitted
01:00.0 VGA compatible controller: NVIDIA Corporation GA106M [GeForce RTX 3060 Mobile / Max-Q] (rev a1)
	Subsystem: Dell Device 0b5e
	Kernel driver in use: nvidia
	Kernel modules: nvidiafb, nouveau, nvidia_drm, nvidia
--
34:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Rembrandt [Radeon 680M] (rev c8)
	Subsystem: Dell Device 0b5e
	Kernel driver in use: amdgpu
	Kernel modules: amdgpu
```

Things I've tried:
- `GRUB_CMDLINE_LINUX_DEFAULT="quiet splash acpi_backlight=video"`
- `GRUB_CMDLINE_LINUX_DEFAULT="quiet splash acpi_backlight=native"`
- `GRUB_CMDLINE_LINUX_DEFAULT="quiet splash  acpi_backlight=nvidia_wmi_ec"`
- `GRUB_CMDLINE_LINUX_DEFAULT="quiet splash nvidia.NVreg_RegistryDwords=EnableBrightnessControl=1"`

Anyone has found any other workaround for this?