NVIDIA/open-gpu-kernel-modules #412: error: implicit declaration of function ‘sys_close’; did you mean ‘ksys_close’?

Vaderpucong, 2022-11-25:
### NVIDIA Open GPU Kernel Modules Version

520

### Operating System and Version

 Ubuntu 20.04.1 LTS

### Kernel Release

Linux localhost 5.10.35-00023-gcf7d90a076c #1 SMP PREEMPT Wed Sep 21 11:48:07 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux

### Build Command

```
make modules -j `nproc`
```

### Terminal output/Build Log

  CC [M]  /home/open-gpu-kernel-modules/kernel-open/nvidia-uvm/uvm_hopper.o
/home/open-gpu-kernel-modules/kernel-open/nvidia/nv-caps.c: In function ‘nv_cap_close_fd’:
/home/open-gpu-kernel-modules/kernel-open/nvidia/nv-**caps.c:567:5: error: implicit declaration of function ‘sys_close’; did you mean ‘ksys_close’? [-Werror=implicit-function-declaration]
  567 |     sys_close(fd);
      |     ^~~~~~~~~
      |     ksys_close**
  CC [M]  /home/open-gpu-kernel-modules/kernel-open/nvidia-uvm/uvm_hopper_fault_buffer.o
  CC [M]  /home/open-gpu-kernel-modules/kernel-open/nvidia-uvm/uvm_hopper_ce.o
  CC [M]  /home/open-gpu-kernel-modules/kernel-open/nvidia-uvm/uvm_hopper_host.o
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:279: /home/open-gpu-kernel-modules/kernel-open/nvidia/nv-caps.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:1821: /home/open-gpu-kernel-modules/kernel-open] Error 2
make[2]: Leaving directory '/home/linux'
make[1]: *** [Makefile:82: modules] Error 2
make[1]: Leaving directory '/home/open-gpu-kernel-modules/kernel-open'
make: *** [Makefile:59: modules] Error 2

Vaderpucong, 2022-11-28:
I use NXP LX2160A board and disable symbal table check feature

mtijanic, 2022-11-28:
Hey there, looks like the conftest that checks which of these functions to call didn't work on your system:
https://github.com/NVIDIA/open-gpu-kernel-modules/blob/main/kernel-open/nvidia/nv-caps.c#L583-L599

Which is odd since 5.10.35 _does_ export `__close_fd()`: https://elixir.bootlin.com/linux/v5.10.35/source/fs/file.c#L673

Could you run this bit and post the output?
```sh
open-gpu-kernel-modules$ grep -r close_fd ./kernel-open/conftest/
```
e.g., on my 5.15 kernel I get:
```
open-gpu-kernel-modules$ grep -r close_fd ./kernel-open/conftest/
./kernel-open/conftest/compile-tests/is_export_symbol_present_close_fd.h:#define NV_IS_EXPORT_SYMBOL_PRESENT_close_fd 1
./kernel-open/conftest/compile-tests/is_export_symbol_present___close_fd.h:#define NV_IS_EXPORT_SYMBOL_PRESENT___close_fd 0
./kernel-open/conftest/symbols.h:#define NV_IS_EXPORT_SYMBOL_PRESENT___close_fd 0
./kernel-open/conftest/symbols.h:#define NV_IS_EXPORT_SYMBOL_PRESENT_close_fd 1
```
which makes the above code call `close_fd()`. I'd _expect_ for 5.10.35 to be inverse.

Vaderpucong, 2022-12-01:
```

root@localhost:/home/open-gpu-kernel-modules# grep -r close_fd ./kernel-open/conftest/
./kernel-open/conftest/symbols.h:#define NV_IS_EXPORT_SYMBOL_PRESENT___close_fd 0
./kernel-open/conftest/symbols.h:#define NV_IS_EXPORT_SYMBOL_PRESENT_close_fd 0
./kernel-open/conftest/compile-tests/is_export_symbol_present_close_fd.h:#define NV_IS_EXPORT_SYMBOL_PRESENT_close_fd 0
./kernel-open/conftest/compile-tests/is_export_symbol_present___close_fd.h:#define NV_IS_EXPORT_SYMBOL_PRESENT___close_fd 0
```

Vaderpucong, 2022-12-01:
**I compile linux kernel in local**, and recompile NVIDIA Open GPU Kernel Modules. There seems to be no more problem here.

```
root@localhost:/home/open-gpu-kernel-modules# grep -r close_fd ./kernel-open/conftest/
./kernel-open/conftest/symbols.h:#define NV_IS_EXPORT_SYMBOL_PRESENT___close_fd 1
./kernel-open/conftest/symbols.h:#define NV_IS_EXPORT_SYMBOL_PRESENT_close_fd 0
./kernel-open/conftest/compile-tests/is_export_symbol_present_close_fd.h:#define NV_IS_EXPORT_SYMBOL_PRESENT_close_fd 0
./kernel-open/conftest/compile-tests/is_export_symbol_present___close_fd.h:#define NV_IS_EXPORT_SYMBOL_PRESENT___close_fd 1
```

closed 2022-12-05 by Vaderpucong