NVIDIA/open-gpu-kernel-modules #413: Fan speeds between 1 and 24 are allowed for desktop Turing GPUs, they must *not* be

birdie-github, 2022-11-28:
### NVIDIA Open GPU Kernel Modules Version

525.60.11

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Fedora release 36 (Thirty Six)

### Kernel Release

6.0.10 x86_64

### Hardware: GPU

NVIDIA GeForce GTX 1660 Ti

### Describe the bug

There are **two issues** with drivers all drivers between 520.56.06 and 525.60.11 for the **desktop** version of GTX 1660 Ti (possibly others).

1. It's **possible** to set the fan speed to values between 1 and 24% which has a huge chance of **frying** the fan motor **fast** because it switches to constant start/stop mode and at low fan speeds it may not have enough power to rotate the blades thus it will simply overheat.

2. For every nvidia-settings invocation regarding fan control, e.g.
```
$ nvidia-settings -a "[gpu:0]/GPUFanControlState=1"

ERROR: The current user does not have permission for operation

  Attribute 'GPUFanControlState' (zen:0[gpu:0]) assigned value 1.
```
there's a **bogus error** message which shouldn't be there. The operation succeeds anyways.

### To Reproduce

See/read above.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

Unnecessary.

### More Info

This is a continuation of #395 but NVIDIA developers/QA personnel have stopped replying to it, so I'm creating a new one.

Internally this is **bug 3408200**.

amrit1711, 2022-12-14:
@birdie-github 
Can you please share bug report from repro state.
Also regarding the ERROR, you are using non root account to set GPUFanControlState so NVML returned "NVML_ERROR_NO_PERMISSION".
To avoid it, please run nvidia-settings with sudo permissions.

birdie-github, 2022-12-14:
Bug report why?

I have a plain desktop GPU, namely MSI GeForce 1660 Ti Gaming X and **I can set fan speed from 1 to 24 which makes it alternate between starting and stopping all the time** which **can easily lead to the fan motor failing**.

> Also regarding the ERROR, you are using non root account to set GPUFanControlState so NVML returned "NVML_ERROR_NO_PERMISSION".
> To avoid it, please run nvidia-settings with sudo permissions.

This was **not** necessary previously and the command still works which means the error message is **misleading** and **redundant**.

birdie-github, 2022-12-21:
Bug report: https://github.com/NVIDIA/open-gpu-kernel-modules/issues/395#issuecomment-1357314196

Extra info: https://github.com/NVIDIA/open-gpu-kernel-modules/issues/395#issuecomment-1357316015

birdie-github, 2023-01-24:
Driver 525.85.05 still allows to set fan speed from 1 to 24 which could lead to a premature fan motor death.

amrit1711, 2023-02-07:
We are seeing similar behavior on Windows platform as well where setting fan speed between1 to 24 makes it alternate switching between start and stop all the time.
AIC-MSI uses an adjustable item of the fan settings of VBIOS to implement their own "FANSTOP" function.
We suggest you to report this problem to MSI directly and they will be able to clarify about the characteristics of their fan.

birdie-github, 2023-02-13:
> We are seeing similar behavior on Windows platform as well where setting fan speed between1 to 24 makes it alternate switching between start and stop all the time. AIC-MSI uses an adjustable item of [...]

MSI Afterburnder does **not** allow to set fan speeds below 25. What app did you use under Windows to configure fan speeds?

amrit1711, 2023-02-23:
You can use EVGA precision X1 https://www.evga.com/precisionx1/ to limit fan speed.

birdie-github, 2023-03-19:
I'm away from my PC for an indefinite amount of time, so let's close this bug report for a while because I cannot cooperate any longer.

closed 2023-03-19 by birdie-github