NVIDIA/open-gpu-kernel-modules #415: NvEncoder : m_nvenc.nvEncOpenEncodeSessionEx(&encodeSessionExParams, &hEncoder) returned error 10

ichergui, 2022-11-29:
### NVIDIA Open GPU Kernel Modules Version

98553501593ef05bddcc438689ed1136f732d40a

### Does this happen with the proprietary driver (of the same version) as well?

I cannot test this

### Operating System and Version

Description:	Ubuntu 20.04.4 LTS

### Kernel Release

Linux ubuntu 5.10.65-tegra #1 SMP PREEMPT Mon Jun 13 16:27:40 PDT 2022 aarch64 aarch64 aarch64 GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA RTX A6000 (UUID: GPU-ddee49af-0c3d-35de-e476-a1a390bd60ed)

### Describe the bug

Hey everyone,

Running the `AppEncCuda` sample application of the `Video_Codec_SDK_11.0.10` failed 
I'm not sure what is missing.
Could you please help ?

Ilies

 

### To Reproduce

```
nvidia@ubuntu:/media/m2/video-codec-sdk/Video_Codec_SDK_11.0.10/Samples/build/AppEncode/AppEncCuda$ sudo ./AppEncCuda -h
Options:
-i               Input file path
-o               Output file path
-s               Input resolution in this form: WxH
-if              Input format: iyuv nv12 yuv444 p010 yuv444p16 bgra bgra10 ayuv abgr abgr10
-gpu             Ordinal of GPU to use
-outputInVidMem  Set this to 1 to enable output in Video Memory
-cuStreamType    Use CU stream for pre and post processing when outputInVidMem is set to 1
                 CRC of encoded frames will be computed and dumped to file with suffix '_crc.txt' added
                 to file specified by -o option 
                 0 : both pre and post processing are on NULL CUDA stream
                 1 : both pre and post processing are on SAME CUDA stream
                 2 : both pre and post processing are on DIFFERENT CUDA stream
-codec       Codec: h264 hevc
-preset      Preset: p1 p2 p3 p4 p5 p6 p7
-profile     H264: baseline main high high444; HEVC: main main10 frext
-tuninginfo  TuningInfo: hq lowlatency ultralowlatency lossless
-multipass   Multipass: disabled qres fullres
-444         (Only for RGB input) YUV444 encode
-fps         Frame rate
-bf          Number of consecutive B-frames
-rc          Rate control mode: constqp vbr cbr
-gop         Length of GOP (Group of Pictures)
-bitrate     Average bit rate, can be in unit of 1, K, M
-maxbitrate  Max bit rate, can be in unit of 1, K, M
-vbvbufsize  VBV buffer size in bits, can be in unit of 1, K, M
-vbvinit     VBV initial delay in bits, can be in unit of 1, K, M
-aq          Enable spatial AQ and set its stength (range 1-15, 0-auto)
-temporalaq  (No value) Enable temporal AQ
-cq          Target constant quality level for VBR mode (range 1-51, 0-auto)
-lookahead   Maximum depth of lookahead (range 0-(31 - number of B frames))
-qmin        Min QP value
-qmax        Max QP value
-initqp      Initial QP value
-constqp     QP value for constqp rate control mode
Note: QP value can be in the form of qp_of_P_B_I or qp_P,qp_B,qp_I (no space)

Encoder Capability

NvEncoder : m_nvenc.nvEncOpenEncodeSessionEx(&encodeSessionExParams, &hEncoder) returned error 10 at /media/m2/video-codec-sdk/Video_Codec_SDK_11.0.10/Samples/NvCodec/NvEncoder/NvEncoder.cpp:52
nvidia@ubuntu:/media/m2/video-codec-sdk/Video_Codec_SDK_11.0.10/Samples/build/AppEncode/AppEncCuda$
```

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10112559/nvidia-bug-report.log.gz)

ich777, 2023-01-05:
Please see also #378 
It's already a known issue and it's already marked as `NV-Triaged` but nothing changed since then.

ichergui, 2023-01-05:
Thanks @ich777 
I created this issue because video encoding is not working and I didn't see any fix for that in the version I mentioned here.

ich777, 2023-01-05:
@ichergui the issue will persist in every upcoming driver version until Nvidia pushes a fix.

ichergui, 2023-01-05:
@ich777 this version `525.60.13` of the OpenRM driver have the fix already :wink: 
I tried it and it is working fine.

* OpenRM driver version

```
# cat /proc/driver/nvidia/version 
NVRM version: NVIDIA UNIX Open Kernel Module for aarch64  525.60.13  Release Build  (@TS-CS-002)  Sat Dec 17 21:57:24 UTC 2022
GCC version:  gcc version 11.3.0 (GCC) 
#
```

* AppEncCuda sample application

```
# ./AppEncCuda -i ./HeavyHand_1080p.yuv -o test.h264 -s 1920x1080
GPU in use: NVIDIA RTX A6000
[INFO ][10:23:38] Encoding Parameters:
        codec        : h264
        preset       : p3
        tuningInfo   : hq
        profile      : (default)
        chroma       : yuv420
        bitdepth     : 8
        rc           : vbr
        fps          : 30/1
        gop          : 250
        bf           : 1
        multipass    : 0
        size         : 1920x1080
        bitrate      : 0
        maxbitrate   : 0
        vbvbufsize   : 0
        vbvinit      : 0
        aq           : disabled
        temporalaq   : disabled
        lookahead    : disabled
        cq           : 0
        qmin         : P,B,I=0,0,0
        qmax         : P,B,I=0,0,0
        initqp       : P,B,I=0,0,0
Total frames encoded: 76
Bitstream saved in file test.h264
#
```

Let me know if I can help

ich777, 2023-01-05:
> @ich777 this version `525.60.13` of the OpenRM driver have the fix already wink

Oh really, have to compile this version and try it.
Strange that nobody from Nvidia closed the issue or wrote in that thread.

ich777, 2023-01-10:
@ichergui thank you for the hint, you are completely correct that it now works until driver version `525.60.13`

ichergui, 2023-01-10:
@ich777 You're welcome

ngolovliov-nv, 2023-01-23:
Sorry, looks like a changelog update didn't make it. The fix is now present in 525.60.x and newer.

closed 2023-01-23 by ngolovliov-nv

ichergui, 2023-01-23:
@ngolovliov-nv 

Are you planing to back port these changes on `515.65.01` ?

ngolovliov-nv, 2023-03-31:
@ichergui , the fix is now available in [515.105.01](https://github.com/NVIDIA/open-gpu-kernel-modules/commit/21c357a47156af85a2488428d4a604535ab34c7c). Sorry about the radio silence.

ichergui, 2023-03-31:
Thanks @ngolovliov-nv 
Much appreciated