NVIDIA/open-gpu-kernel-modules #416: Proton game Forza Horizon 5 crashes with Xid NVRM 31

wael444, 2022-11-29:
### NVIDIA Open GPU Kernel Modules Version

525.60.11

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

KISS Linux

### Kernel Release

6.1.0-rc7

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 2060 (UUID: GPU-e04204f3-e13f-0796-244f-349cf99271d4)

### Describe the bug

This has been happening since kernel version 5.18 and nvidia 515, essentially when this game came out working for Proton.

The game will crash with this error on completely random times and with no apparent reason.

This issue is is re-created by many many other uses across kernel versions and linux distributions.

### To Reproduce

1. Use a nvidia GPU
2. Use the NVIDIA drivers (open or proprietary)
3. Play this game for more than 20 seconds

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10113283/nvidia-bug-report.log.gz)

### More Info

https://user-images.githubusercontent.com/47404953/204537485-beeec7af-db98-4e04-9e3c-3210c388dfd1.mp4

here it is not even getting past the Main menu.

TURING-V2, 2022-11-30:
Not sure related to ur issue ,on CP2077 it loads and freezes screen  Im on latest dxvk vkd3d , downgrading to 515 works fine.  
1650Ti
[image]

snorkellingcactus, 2023-11-30:
Same here, Proton 8-25, kernel 5.15.96, nvidia proprietary 535.104.05, nvidia 1660ti mobile, GPU-ba900b20-4691-c7da-881c-6ad46ccceb80

SteavenGamerYT, 2024-02-05:
same here with 550