NVIDIA/open-gpu-kernel-modules #417: Frame presentation issue. All driver versions

kndgs, 2022-12-02:
### NVIDIA Open GPU Kernel Modules Version

525.60.11

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

6.0.10-zen2-1-zen

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3060 Ti (UUID: GPU-ba73bc75-4c91-6012-1365-c8e673737f6b)

### Describe the bug

[issue example at 25% speed](https://media.discordapp.net/attachments/822138635155013712/1048011628886638692/stutter.gif) from 60fps game, instead of showing proper frame it jumps back to a frame which is 3 frames old and then continues with the most recent one.

This happens with and without compositor, with and without composition/fullcomposition pipeline in any app, browser, etc.
This also happens on both Xorg and Wayland (wlroots gl and vulkan).
This also happens on any driver version, both propriatary and open.

Only time this doesn't happen is when there is no compositor or compositionpipeline and app/game is in its native fullscreen mode and has vsync turned on and runs on vulkan.

nvidia-settings:
+Sync to VBlank
+Allow Flipping (without it there is tearing)

### To Reproduce

Good way to see it is to open any YT video like [this](https://youtu.be/gmHaa5pvpVc) and enjoy the stutters or also multiple tearing lines if no compositor or composition pipeline is present.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10142966/nvidia-bug-report.log.gz)

### More Info

This issue exists as long as I remember myself using Nvidia GPUs across different Linux systems.

ValZapod, 2022-12-03:
This remind of 

Does it happen on 60/1.001 too?

theyareonit, 2023-07-12:
Experiencing the same behavior on all tested kernel versions & nvidia driver versions, currently on 535.54.03 with kernel 6.4.3-tkg-bore on Arch Linux. I've only tested kernel & nvidia driver versions from the past 7 months or so.

HW: Zotac RTX 3060 Twin Edge OC, Asus TUF Z690 D4, i5 12500

Here's my own recording of the issue if it helps at all: https://www.youtube.com/watch?v=fNqf-Uw9FG4
It's very obvious in TestUFO or any other instance of sidescrolling motion, but it occurs in all programs.

Using a higher framerate or the "Prefer Maximum Performance" power option mitigates the issue, but does not completely remove it, and it remains extremely distracting even at 240hz with 240FPS.

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/12033652/nvidia-bug-report.log.gz)