NVIDIA/open-gpu-kernel-modules #421: Dmesg NVRM messages on Dell Precision 7760

w5pny, 2022-12-05:
### NVIDIA Open GPU Kernel Modules Version

525.60.11

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Ubuntu 20.04.5 LTS

### Kernel Release

Linux sirius 5.14.0-1054-oem #61-Ubuntu SMP Fri Oct 14 13:05:50 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA RTX A3000 Laptop GPU (UUID: GPU-417ead46-fcb2-4593-f174-2f1cda4562be)

### Describe the bug

I have a Dell Precision 7760 Laptop with an Nvidia GPU.
It is running Ubumtu 20.04 LTS and was using the
Nvidie 520 driver (non-open).  When I did the upgrade
to 525 I went to the open kernel module version and
I had to add

options nvidia NVreg_OpenRmEnableUnsupportedGpus=1

to the /etc/modeprobe.d directory in order for
the nvidia utilities to work.

Things seem to work fine except that when booting 
I get the following in the demsg output:

[    1.792940] NVRM: loading NVIDIA UNIX Open Kernel Module for x86_64  525.60.11  Release Build  (dvs-builder@U16-T01-9-2)  Wed Nov 23 23:04:11 UTC 2022
[    1.924873] NVRM objClInitPcieChipset: *** Chipset Setup Function Error!

Since things seem to be working ok, I don't know of this is a bug or not...

nvidia-smi -q yields:

==============NVSMI LOG==============

Timestamp                                 : Sun Dec  4 17:42:13 2022
Driver Version                            : 525.60.11
CUDA Version                              : 12.0

Attached GPUs                             : 1
GPU 00000000:01:00.0
    Product Name                          : NVIDIA RTX A3000 Laptop GPU
    Product Brand                         : NVIDIA RTX
    Product Architecture                  : Ampere
    Display Mode                          : Disabled
    Display Active                        : Disabled
    Persistence Mode                      : Disabled
    MIG Mode
        Current                           : N/A
        Pending                           : N/A
    Accounting Mode                       : Disabled
    Accounting Mode Buffer Size           : 4000
    Driver Model
        Current                           : N/A
        Pending                           : N/A
    Serial Number                         : N/A
    GPU UUID                              : GPU-417ead46-fcb2-4593-f174-2f1cda4562be
    Minor Number                          : 0
    VBIOS Version                         : 94.04.42.40.02
    MultiGPU Board                        : No
[... 172 lines cut ...]
            Used GPU Memory               : 3 MiB
        GPU instance ID                   : N/A
        Compute instance ID               : N/A
        Process ID                        : 41698
            Type                          : G
            Name                          : /usr/bin/akonadi_sendlater_agent
            Used GPU Memory               : 3 MiB
        GPU instance ID                   : N/A
        Compute instance ID               : N/A
        Process ID                        : 41699
            Type                          : G
            Name                          : /usr/bin/akonadi_unifiedmailbox_agent
            Used GPU Memory               : 75 MiB

### To Reproduce

Same dmesg NVRM messages appear with ever reboot. 

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10150176/nvidia-bug-report.log.gz)

w5pny, 2023-01-19:
Ubuntu just released 525.78.01 and this problem still exists with that version.

mtijanic, 2023-01-23:
Hey there, thanks for the report!

    [ 1.792940] NVRM: loading NVIDIA UNIX Open Kernel Module for x86_64 525.60.11 Release Build (dvs-builder@U16-T01-9-2) Wed Nov 23 23:04:11 UTC 2022

This one is expected to be printed during normal operation, no worries there.

    [ 1.924873] NVRM objClInitPcieChipset: *** Chipset Setup Function Error!

This one is very unfortunately written. It comes up when we fail to apply some per-chipset specific hacks and workarounds. This can happen for two reasons:
1. The function that applies these hacks fails for some reason.
2. The chipset is not known to the driver

In your case, it's (2). You can see a list of known chipsets here:
https://github.com/NVIDIA/open-gpu-kernel-modules/blob/main/src/nvidia/arch/nvalloc/common/inc/nvcst.h#L101-L189
Tiger Lake is missing from that list. I'm not sure why, probably an oversight. I'll check.

If there are no special hacks to be applied to a given chipset, then this error message is harmless spam. AFAIK this is the case here.

However, it's good that you opened this issue as we really need to clean up this print. It shouldn't spam by default if the chipset is not found, and the error message should be clearer.

Since you report that everything is working, you shouldn't worry. And sorry for spamming your logs.

w5pny, 2023-01-23:
Thanks for the update on this!  That explains the reason for NVRM message.  I have another message to nividia in the linux nvidia forum about the proprietary kernel
version on the 7760 though, since ever since 520, the proprietary kernel version
freezes on boot -- only the open kernel version works and I was thinking maybe the NVRM message I was getting was a reason for that as well.  I would think the proprietary kernel version should work as well....  So "evertyhing is working" is ONLY true for the open-kernel version!

w5pny, 2023-01-23:
I mis-spoke above, the proprietary kernel version with the 7760 freezes on
boot ever since version 515, not 520.  510, was the last working proprietary
version.  The open-kernel version has always worked, except I needed to non-recognized gpu modprobe workaround for the nvidia tools to work with the open-kernel version, but that is now no longer required either with 525-78.

w5pny, 2023-01-23:
See:

https://forums.developer.nvidia.com/t/525-60-11-proprietary-kernel-version-wont-boot-with-dell-precision-t7760-laptop/238763

w5pny, 2023-03-02:
With 525.85.05 this issue seems to be resolved!

closed 2023-03-02 by w5pny