NVIDIA/open-gpu-kernel-modules #422: ASUS TUF Gaming A15 can't wake after sleep

TanawatJukmongkol, 2022-12-05:
### NVIDIA Open GPU Kernel Modules Version

525.60.11

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

6.0.11-zen1-1-zen

### Hardware: GPU

NVIDIA GeForce RTX 3050 Laptop GPU

### Describe the bug

Cannot wake the laptop up after running `systemctl suspend`, or press the sleep button.

`
> cat /sys/power/mem_sleep
$ [s2idle]
> cat /sys/power/state
$ freeze mem disk
`

### To Reproduce

N/A

### Bug Incidence

Once

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10151068/nvidia-bug-report.log.gz)

amrit1711, 2022-12-20:
Thanks for reporting issue to us, there is a bug 3908600 internally filed for tracking purpose and team is working on it.

amrit1711, 2023-03-27:
Please test with driver 530.41.03 and share test results.

TanawatJukmongkol, 2023-03-29:
Works wonderfully! Tried with both the sleep button, and `systemctl suspend`. Works both on Wayland and Xorg KDE-Plasma. Have not tested on Tiling WMs, but if there's any other issues related to this, or other issues arises I'll report right away! Closing this for now :)

closed 2023-03-29 by TanawatJukmongkol