NVIDIA/open-gpu-kernel-modules #424: Audio sometimes cuts out on monitor output

Pixirora, 2022-12-08:
### NVIDIA Open GPU Kernel Modules Version

525.60.11-2

### Does this happen with the proprietary driver (of the same version) as well?

I cannot test this

### Operating System and Version

Arch Linux

### Kernel Release

6.0.11-arch1-1

### Hardware: GPU

NVIDIA GeForce RTX 2060 SUPER

### Describe the bug

Audio output to monitors plugged into both the Display Port and HDMI adapter of my GPU sometimes cuts out for a short moment (less than 100 ms). This happens to audio from browsers, games, and virtual machines. This bug seems to occur randomly when the GPU is under any sort of load, and does not happen with the nouveau driver installed. I have tested this on pulseaudio and pipewire, and the glitch seems to persist on both.

### To Reproduce

Use the audio output on your monitor.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10189169/nvidia-bug-report.log.gz)

sTiKyt, 2023-01-27:
Have same issue and had to use audio port on motherboard for my headphones because of that instead monitor

dcelasun, 2023-11-11:
Same issue here. RTX 4070, driver 545.29.02. Notably, the bug doesn't occur with my 1080 Ti.

Plet53, 2024-05-10:
Currently experiencing the same issue.
Card: RTX 3060
nvidia pacakge version: 550.78-2
kernel: 6.8.9.arch1-2

Dawud18854, 2026-04-26:
Same issue here:

6.19.13-arch1-1
RTX 5090
595.58.03