NVIDIA/open-gpu-kernel-modules #428: Unproportional brightness slider with 525.x

aslary, 2022-12-22:
### NVIDIA Open GPU Kernel Modules Version

Any version of 525

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Tested on Fedora 37 and PopOS 22.04 

### Kernel Release

6.0 and 5.17

### Hardware: GPU

GTX 1660Ti

### Describe the bug

The lowest two steps of the display brightness regulation slider max out the brightness. A demo .gif can be found 

### To Reproduce

Update your GPU driver from 515 to 525

### Bug Incidence

Always

### nvidia-bug-report.log.gz

None

QuestionMark001, 2022-12-24:
I meet the same problem.  
Nvidia Driver: `Proprietary Driver`  
Driver Version: `525.60.11`  
Operating System: `Arch Linux x86_64`  
Linux Kernel Release: `6.1.1`  
Nvidia GPU: `NVIDIA GeForce RTX 3060 Laptop GPU`  
[image]  
[image]

GrayYip, 2023-01-18:
your problem is the same as mine

https://github.com/NVIDIA/open-gpu-kernel-modules/issues/409

ValZapod, 2023-01-23:
Why is brightness slider even works in gif? Is not brightness supposed to work in the display itself?

LRitzdorf, 2023-02-28:
I've encountered this as well, and have posted further details of the observed backlight behavior at https://bbs.archlinux.org/viewtopic.php?pid=2072664.

My solution until recently was to use an old driver version via DKMS, but after the recent kernel 6.2 release (on Arch), the old module no longer builds. I'm thus forced to use a driver that manifests this issue, or hold back my kernel. With this change, it would really be nice if this issue received some proper developer attention.

All-The-Foxes, 2023-04-20:
This issue persists on driver version 530.