NVIDIA/open-gpu-kernel-modules #429: 6.2rc1 dkms is failing to build

ptr1337, 2022-12-27:
### NVIDIA Open GPU Kernel Modules Version

525.60.11

### Operating System and Version

CachyOS (archlinux based)

### Kernel Release

6.2.rc1

### Build Command

Closed and open drivers are failing when building dkms.

### Terminal output/Build Log

  SYMLINK /var/lib/dkms/nvidia/525.60.11/build/nvidia/nv-kernel.o
  SYMLINK /var/lib/dkms/nvidia/525.60.11/build/nvidia-modeset/nv-modeset-kernel.o
 CONFTEST: hash__remap_4k_pfn
 CONFTEST: set_pages_uc
 CONFTEST: list_is_first
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: set_pages_array_uc
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: get_num_physpages
 CONFTEST: pde_data
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: dma_map_page_attrs
 CONFTEST: dma_attr_macros
 CONFTEST: write_cr4
 CONFTEST: of_find_node_by_phandle
 CONFTEST: of_node_to_nid
 CONFTEST: pnv_pci_get_npu_dev
 CONFTEST: of_get_ibm_chip_id
 CONFTEST: pci_bus_address
 CONFTEST: pci_stop_and_remove_bus_device
 CONFTEST: register_cpu_notifier
 CONFTEST: dma_map_resource
 CONFTEST: cpuhp_setup_state
 CONFTEST: get_backlight_device_by_name
 CONFTEST: timer_setup
 CONFTEST: pci_enable_msix_range
 CONFTEST: kernel_read_has_pointer_pos_arg
 CONFTEST: kernel_write_has_pointer_pos_arg
 CONFTEST: dma_direct_map_resource
 CONFTEST: tegra_get_platform
 CONFTEST: tegra_bpmp_send_receive
 CONFTEST: flush_cache_all
 CONFTEST: vmf_insert_pfn
 CONFTEST: jiffies_to_timespec
 CONFTEST: ktime_get_raw_ts64
 CONFTEST: ktime_get_real_ts64
 CONFTEST: full_name_hash
 CONFTEST: pci_enable_atomic_ops_to_root
 CONFTEST: vga_tryget
 CONFTEST: pgprot_decrypted
 CONFTEST: cc_mkdec
 CONFTEST: seq_read_iter
 CONFTEST: unsafe_follow_pfn
 CONFTEST: drm_gem_object_get
 CONFTEST: add_memory_driver_managed
 CONFTEST: drm_gem_object_put_unlocked
 CONFTEST: device_property_read_u64
[... 541 lines cut ...]
/var/lib/dkms/nvidia/525.60.11/build/nvidia-uvm/uvm_rb_tree_test.c:361:13: note: insert 'break;' to avoid fall-through
            default:
            ^
            break; 
ld.lld -r -o /var/lib/dkms/nvidia/525.60.11/build/nvidia-modeset/nv-modeset-interface.o /var/lib/dkms/nvidia/525.60.11/build/nvidia-modeset/nvidia-modeset-linux.o /var/lib/dkms/nvidia/525.60.11/build/nvidia-modeset/nv-kthread-q.o
  LD [M]  /var/lib/dkms/nvidia/525.60.11/build/nvidia-modeset.o
/var/lib/dkms/nvidia/525.60.11/build/nvidia-drm/nvidia-drm-drv.c:262:22: error: no member named 'fb_base' in 'struct drm_mode_config'
    dev->mode_config.fb_base = 0;
    ~~~~~~~~~~~~~~~~ ^
1 error generated.
make[2]: *** [scripts/Makefile.build:252: /var/lib/dkms/nvidia/525.60.11/build/nvidia-drm/nvidia-drm-drv.o] Error 1
make[2]: *** Waiting for unfinished jobs....
1 warning generated.

### More Info

The 6.2 rc1 Kernel got released on sunday and nvidia-dkms is not building anymore correctly.
Also there are many warnings about the rethunks:
```
/var/lib/dkms/nvidia/525.60.11/build/nvidia-modeset.o: warning: objtool: _nv000094kms+0x4f: 'naked' return found in RETHUNK build
```

Related commit from kernel:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/include/drm/drm_mode_config.h?id=7c99616e3fe7f35fe25bf6f5797267da29b4751e
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/gpu/drm/drm_edid.c?id=90b575f52c6ab35979968e2e4d9cbd9f1eb3901c

loqs, 2022-12-29:
> ```
> /var/lib/dkms/nvidia/525.60.11/build/nvidia-modeset.o: warning: objtool: _nv000094kms+0x4f: 'naked' return found in RETHUNK build
> ```
Have you tried disabling asynchronous-unwind-tables in src/nvidia-modeset/Makefile?  The patch below also address SLS and RETHUNK warnings.
```patch
diff --git a/src/nvidia-modeset/Makefile b/src/nvidia-modeset/Makefile
index 9ef3c62..a6e3dc7 100644
--- a/src/nvidia-modeset/Makefile
+++ b/src/nvidia-modeset/Makefile
@@ -140,8 +140,11 @@ ifeq ($(TARGET_ARCH),x86_64)
     CONDITIONAL_CFLAGS += $(call TEST_CC_ARG, -fcf-protection=branch)
   endif
   CONDITIONAL_CFLAGS += $(call TEST_CC_ARG, -fno-jump-tables)
+  CONDITIONAL_CFLAGS += $(call TEST_CC_ARG, -fno-asynchronous-unwind-tables)
   CONDITIONAL_CFLAGS += $(call TEST_CC_ARG, -mindirect-branch=thunk-extern)
   CONDITIONAL_CFLAGS += $(call TEST_CC_ARG, -mindirect-branch-register)
+  CONDITIONAL_CFLAGS += $(call TEST_CC_ARG, -mharden-sls=all)
+  CONDITIONAL_CFLAGS += $(call TEST_CC_ARG, -mfunction-return=thunk-extern)
 endif
 
 CFLAGS += $(CONDITIONAL_CFLAGS)
diff --git a/src/nvidia/Makefile b/src/nvidia/Makefile
index b7ca09b..dc747bd 100644
--- a/src/nvidia/Makefile
+++ b/src/nvidia/Makefile
@@ -164,6 +164,8 @@ ifeq ($(TARGET_ARCH),x86_64)
   CONDITIONAL_CFLAGS += $(call TEST_CC_ARG, -fno-jump-tables)
   CONDITIONAL_CFLAGS += $(call TEST_CC_ARG, -mindirect-branch-register)
   CONDITIONAL_CFLAGS += $(call TEST_CC_ARG, -mindirect-branch=thunk-extern)
+  CONDITIONAL_CFLAGS += $(call TEST_CC_ARG, -mharden-sls=all)
+  CONDITIONAL_CFLAGS += $(call TEST_CC_ARG, -mfunction-return=thunk-extern)
 endif
 
 CFLAGS += $(CONDITIONAL_CFLAGS)
```

ptr1337, 2022-12-29:
Actually I already got nvidia-dkms patched for 6.2.
This patch actually fixes it:
```
From ac78402fe504591eeb98a4212e695731755d6b25 Mon Sep 17 00:00:00 2001
From: Peter Jung <admin@ptr1337.dev>
Date: Tue, 27 Dec 2022 18:08:48 +0100
Subject: [PATCH] Tentative fix for NVIDIA 525 driver for Linux 6.2-rc1

---
 kernel-dkms/nvidia-drm/nvidia-drm-connector.c | 22 +++++++++++++++++++
 kernel-dkms/nvidia-drm/nvidia-drm-drv.c       |  4 ++++
 2 files changed, 26 insertions(+)

diff --git a/kernel-dkms/nvidia-drm/nvidia-drm-connector.c b/kernel-dkms/nvidia-drm/nvidia-drm-connector.c
index 77a6120..cd92bb2 100644
--- a/kernel-dkms/nvidia-drm/nvidia-drm-connector.c
+++ b/kernel-dkms/nvidia-drm/nvidia-drm-connector.c
@@ -20,6 +20,8 @@
  * DEALINGS IN THE SOFTWARE.
  */
 
+#include <linux/version.h>
+#include <drm/drm_edid.h>
 #include "nvidia-drm-conftest.h" /* NV_DRM_ATOMIC_MODESET_AVAILABLE */
 
 #if defined(NV_DRM_ATOMIC_MODESET_AVAILABLE)
@@ -98,6 +100,7 @@ __nv_drm_detect_encoder(struct NvKmsKapiDynamicDisplayParams *pDetectParams,
             break;
     }
 
+#if LINUX_VERSION_CODE < KERNEL_VERSION(6, 2, 0)
     if (connector->override_edid) {
         const struct drm_property_blob *edid = connector->edid_blob_ptr;
 
@@ -110,6 +113,25 @@ __nv_drm_detect_encoder(struct NvKmsKapiDynamicDisplayParams *pDetectParams,
                     sizeof(pDetectParams->edid.buffer));
         }
     }
+#else
+    // Rel. commit "drm/edid: detach debugfs EDID override from EDID property update" (Jani Nikula, 24 Oct 2022)
+    // NOTE: HUGE HACK!
+    mutex_lock(&connector->edid_override_mutex);
+    if (connector->edid_override) {
+        const struct edid *edid = drm_edid_raw(connector->edid_override);
+        size_t edid_length = EDID_LENGTH * (edid->extensions + 1);
+        if (edid_length <= sizeof(pDetectParams->edid.buffer)) {
+            memcpy(pDetectParams->edid.buffer, edid, edid_length);
+            pDetectParams->edid.bufferSize = edid_length;
+            pDetectParams->overrideEdid = NV_TRUE;
+        } else {
+            WARN_ON(edid_length >
+                    sizeof(pDetectParams->edid.buffer));
+        }
+    }
+    mutex_unlock(&connector->edid_override_mutex);
+
+#endif
 
     if (!nvKms->getDynamicDisplayInfo(nv_dev->pDevice, pDetectParams)) {
         NV_DRM_DEV_LOG_ERR(
diff --git a/kernel-dkms/nvidia-drm/nvidia-drm-drv.c b/kernel-dkms/nvidia-drm/nvidia-drm-drv.c
index ac504ec..75ef54d 100644
--- a/kernel-dkms/nvidia-drm/nvidia-drm-drv.c
+++ b/kernel-dkms/nvidia-drm/nvidia-drm-drv.c
@@ -20,6 +20,7 @@
  * DEALINGS IN THE SOFTWARE.
  */
 
+#include <linux/version.h>
 #include "nvidia-drm-conftest.h" /* NV_DRM_AVAILABLE and NV_DRM_DRM_GEM_H_PRESENT */
 
 #include "nvidia-drm-priv.h"
@@ -257,9 +258,12 @@ nv_drm_init_mode_config(struct nv_drm_device *nv_dev,
     dev->mode_config.preferred_depth = 24;
     dev->mode_config.prefer_shadow = 1;
 
+// Rel. commit "drm: Remove drm_mode_config::fb_base" (Zack Rusin, 18 Oct 2022)
+#if defined(CONFIG_FB) && LINUX_VERSION_CODE < KERNEL_VERSION(6, 2, 0)
     /* Currently unused. Update when needed. */
 
     dev->mode_config.fb_base = 0;
+#endif
 
 #if defined(NV_DRM_CRTC_STATE_HAS_ASYNC_FLIP) || \
     defined(NV_DRM_CRTC_STATE_HAS_PAGEFLIP_FLAGS)
-- 
2.39.0
```

The rethunk part is not needed to get fixed for now. So far it also works without the above patch. I think nvidia will fix this with their new driver hopefully.

xevilstar, 2023-01-03:
this patch fixes also the not-open driver sources

xevilstar, 2023-01-03:
leave the "should" it is working on my laptop ;)  thanks

`root@ghost:~# nvidia-smi 
Tue Jan  3 16:55:13 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.60.13    Driver Version: 525.60.13    CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  On   | 00000000:01:00.0  On |                  N/A |
| N/A   41C    P8    18W /  N/A |    705MiB /  8192MiB |     17%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A     10334      G   /usr/lib/xorg/Xorg                328MiB |
|    0   N/A  N/A     10613      G   /usr/bin/gnome-shell              100MiB |
|    0   N/A  N/A     11593      G   ...plications/Zoiper5/zoiper       14MiB |
|    0   N/A  N/A     11623    C+G   ...161843668523319335,131072      125MiB |
|    0   N/A  N/A     11923      G   ...RendererForSitePerProcess       47MiB |
|    0   N/A  N/A     23144    C+G   ...lib/vmware/bin/mksSandbox       76MiB |
|    0   N/A  N/A     73834      G   /usr/bin/deja-dup                   6MiB |
+-----------------------------------------------------------------------------+
root@ghost:~# uname -a
Linux ghost 6.2.0-rc2 #1 SMP PREEMPT_DYNAMIC Tue Jan  3 09:06:34 CET 2023 x86_64 GNU/Linux`

xevilstar, 2023-01-05:
news ?

ptr1337, 2023-01-05:
> news ?

Its fixed in `525.78.01 `

closed 2023-01-05 by ptr1337