NVIDIA/open-gpu-kernel-modules #433: Open-source kernel modules are not working

Monsterovich, 2022-12-30:
### NVIDIA Open GPU Kernel Modules Version

525.60.11

### Does this happen with the proprietary driver (of the same version) as well?

No

### Operating System and Version

Linux Mint 21.1

### Kernel Release

Linux pc 5.19.15-xanmod1 #0~20221012.git85103f7 SMP Wed Oct 12 15:04:07 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

### Hardware: GPU

RTX 3060 Ti

### Describe the bug

The kernel modules are loaded, but the GPU is not detected.

`dmesg` is filled with the same errors:

```
[  217.545140] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ kernel_gsp_ga102.c:235
[  217.545142] NVRM kgspInitRm_IMPL: cannot bootstrap riscv/gsp: 0xffff
[  217.545145] NVRM RmInitAdapter: Cannot initialize GSP firmware RM
[  217.545832] NVRM: GPU 0000:05:00.0: RmInitAdapter failed! (0x62:0xffff:1622)
[  217.546288] NVRM: GPU 0000:05:00.0: rm_init_adapter failed, device minor number 0
[  217.729940] NVRM s_executeBooterUcode_TU102: Booter failed with non-zero error code: 0xa
[  217.729943] NVRM kgspExecuteBooterUnloadIfNeeded_TU102: failed to execute Booter Unload: 0xffff
[  217.738211] NVRM s_executeFwsec_TU102: failed to execute FWSEC for FRTS: FRTS error code 0xbe
[  217.738213] NVRM nvAssertOkFailedNoLog: Assertion failed: Failure: Generic Error [NV_ERR_GENERIC] (0x0000FFFF) returned from kgspExecuteFwsecFrts_HAL(pGpu, pKernelGsp, pKernelGsp->pFwsecUcode, pKernelGsp->pWprMeta->frtsOffset) @ kernel_gsp_ga102.c:164
```

### To Reproduce

```
sudo apt install nvidia-driver-525-open
```

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10324835/nvidia-bug-report.log.gz)

Vaderpucong, 2023-01-03:
I met the same problem.

Vaderpucong, 2023-01-04:
Is acpi selected as the hardware description mode?

JaouadROS, 2023-01-12:
@Vaderpucong did you solve the issue? I have a similar issue with the same card RTX 3060.

constroy, 2023-02-08:
Did you try `NVreg_OpenRmEnableUnsupportedGpus=1` kernel module parameter?

ichergui, 2023-02-08:
Agree with @constroy 
Also, check the GSP firmware in this location: `/lib/firmware/nvidia/525.60.11/`

sTiKyt, 2023-02-08:
> @Vaderpucong did you solve the issue? I have a similar issue with the same card RTX 3060.

RTX 3060 or 3060 Ti? Because they are not the same

JaouadROS, 2023-02-08:
I did solve the problem using a different Nvidia driver. It is RTX 3060.

y-vectorfield, 2023-02-12:
@constroy 
I tried NVreg_OpenRmEnableUnsupportedGpus=1 kernel module parameter, then this trouble was solved!!
By the way, why is this option invalid by default?
This troubleshooting task is very very very complicated!!

koltenfluckiger, 2023-02-16:
> I did solve the problem using a different Nvidia driver. It is RTX 3060.
What driver did you end up using?

JaouadROS, 2023-02-16:
> > I did solve the problem using a different Nvidia driver. It is RTX 3060.
> > What driver did you end up using?

Now I'm using 525.78.01. Just try between recent ones, proprietary, then (proprietary, tested), etc. restart every time and see if any version works for you.

y-vectorfield, 2023-02-16:
I am also using version 525.78.01.

Monsterovich, 2023-03-11:
> @Monsterovich I met the same problem.
> 
> @constroy I tried NVreg_OpenRmEnableUnsupportedGpus=1 kernel module parameter, then this trouble was solved!! By the way, why is this option invalid by default? This troubleshooting task is very very very complicated!!

I agree. This problem doesn't appear in other driver, only in open-source. However, this option eliminates the problem.

j3soon, 2023-03-18:
> Did you try `NVreg_OpenRmEnableUnsupportedGpus=1` kernel module parameter?

Specifically, you can run the following command:

```sh
echo "options nvidia NVreg_OpenRmEnableUnsupportedGpus=1" | sudo tee /etc/modprobe.d/nvreg_fix.conf > /dev/null
```

as discussed here: https://answers.launchpad.net/ubuntu/+source/ubuntu-drivers-common/+question/705811

aritger, 2023-05-30:
If I'm reading the comments correctly, I believe this is resolved by setting the NVreg_OpenRmEnableUnsupportedGpus=1 kernel module parameter.  For better or worse, that is the intended behavior.  We're working to close the remaining feature and performance gaps between the closed drivers and the open drivers, at which point we can hopefully remove the need for that kernel module option.  Until then, this is behaving as intended.  Closing.  Sorry.

closed 2023-05-30 by aritger

y-vectorfield, 2023-05-31:
We hope the installation step becomes simple.