NVIDIA/open-gpu-kernel-modules #435: Dynamic Boost still not using all the power it is supposed to

Kleidis, 2023-01-02:
### NVIDIA Open GPU Kernel Modules Version

525.60.11

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Pop!_OS 22.04 LTS

### Kernel Release

Linux pop-os 6.0.12-76060006-generic #202212290932~1671652965~22.04~452ea9d SMP PREEMPT_DYNAMIC Wed D x86_64 x86_64 x86_64 GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3050 Laptop GPU (UUID: GPU-b2f720a1-2b4e-9b17-4383-76f9361248a2)

### Describe the bug

The maximum power my GPU can use is 85W , On Windows that is the case under heavy workloads 
But on Linux with dynamic boost enabled and nvidia powerd running the power is limited to 64W 
Or 75W if i switch my laptop to performance mode and reboot my system 

(Linux dsitro doesn't really matter since this is a driver bug)

### To Reproduce

I used "Portal with RTX" title menu to showcase this bug but this bug happens on any graphically demanding application 
The full power of my GPU is just not being utilized
On Windows this game was using 85W at the same area 
Mangohud was used to measure the power usage on linux 
 
[image]

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10332343/nvidia-bug-report.log.gz)

kleidiss, 2023-04-14:
This issue is kinda fixed with 530 drivers , Now if you switch to performance mode and restart thhis laptop gets 85W (We have the same laptop)

But it should not need a restart to enable 85W , it should be able to go from 60W to 85W just by pressing fn + Q and going to performance mode , I might make a new issue for this

kleidiss, 2023-04-15:
https://github.com/NVIDIA/open-gpu-kernel-modules/issues/492

Updated version of this issue

amrit1711, 2023-10-11:
This issue was fixed and integrated in drivers which are released already.
Please let us know if anyone still facing the same issue by any chance.