NVIDIA/open-gpu-kernel-modules #436: Use git properly for versioning

orowith2os, 2023-01-03:
### NVIDIA Open GPU Kernel Modules Version

Any

### Does this happen with the proprietary driver (of the same version) as well?

I cannot test this

### Operating System and Version

Any

### Kernel Release

Any

### Hardware: GPU

Any

### Describe the bug

NVIDIA currently uses git *commits* for tracking versions of the NVIDIA driver, with large git commits squashed all into one. 

Git tags exist for this use case, and should be used rather than individual commits. Additionally, not using git properly makes it harder to track down bugs as you can only find if an issue happens in one version or another. 

If managing this and the proprietary kernel module tree is an issue, you can also cherry pick commits and apply them to this repo that way.

### To Reproduce

View the git logs of the repository.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

none

### More Info

The management of this git repository is the worst I've seen, especially one managed by a multibillion dollar company. How you're managing this git repository currently can only cause harm, and downplays the contributions of everybody that contributes to this driver.

aritger, 2023-01-03:
Thanks for the feedback.  I completely agree with your assessment.

Unfortunately, I don't believe this is something we can feasibly change in the near future.  There are a number of challenges that make this difficult to unravel:

* Currently, NVIDIA-internal commits potentially change both open-gpu-kernel-modules and proprietary NVIDIA driver code.  Thus, today, we can't publish complete commits as-is.  In a perfect world, we would better segregate changes, and I think we're working towards that, but it isn't the case today.

* Currently, NVIDIA-internal commits utilize a variety of code generation tools.  The output of these tools may affect both open-gpu-kernel-modules and proprietary NVIDIA driver code, which makes the previous bullet harder to resolve.

* The rate of change is high enough that we can't simply cherry-pick individual changes, hand editing to just the portion of each commit that applies to open-gpu-kernel-modules.

These design choices are a result of evolving our pre-existing proprietary code base into something that we could release as source, while maintaining both our existing proprietary NVIDIA driver and open-gpu-kernel-modules.  Internally, we'll continue to evolve, and hopefully someday get to a model where we can provide actual git history.

But, sadly, for the foreseeable future, I think we're stuck with monolithic commits (one-per-release), and all the downsides of that which you identified.

Marking as "Close as not planned".  Sorry.

aritger, 2023-05-30:
Re-closing, with the explanation provided on January 3.  Sorry.

closed 2023-05-30 by aritger