NVIDIA/open-gpu-kernel-modules #440: KDE and GNOME wayland sessions high CPU usage with external displays

Enverbalalic, 2023-01-08:
### NVIDIA Open GPU Kernel Modules Version

525.78.01

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

6.1.4

### Hardware: GPU

NVIDIA GeForce GTX 1660 Ti (UUID: GPU-1039f844-290a-8383-6bb3-75f30db7d7c5)

### Describe the bug

When using a laptop and an external display both KDE and GNOME Wayland sessions have high CPU usage when doing anything/refreshing the external display

### To Reproduce

- Connect external display
- See high CPU usage

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10369410/nvidia-bug-report.log.gz)

Lynet101, 2023-02-28:
Not just KDE and gnome, just wayland in general... Sway, hyprland, wayfire, you name it!

Enverbalalic, 2023-03-01:
Yes, DMABUF import is broken on wayland :(

ankurdhama, 2023-05-27:
This issue is still present in 530. Kwin and Mutter still have to fallback to CPU copy as the nvidia driver fails to import the dma buffer.

relief-melone, 2023-11-07:
I am having the same issue since updating to the nvidia drivers **545.29.02** for my GeForce 3080Ti (Laptop). Happens here when starting CS2 which ran fine before. 

I am running NixOS on Kernel Version 6.6.0. When killing the process this is the output from *dmesg*. 

>[12740.753015] NVRM unixCallVideoBIOS: int10h(4f02, 0000) vesa call failed! (4f02, 0000)
[12740.753421] NVRM nvCheckOkFailedNoLog: Check failed: Failure: Generic Error [NV_ERR_GENERIC] (0x0000FFFF) returned from pRmApi->Control(pRmApi, nv->rmapi.hClient, nv->rmapi.hSubDevice, NV2080_CTRL_CMD_INTERNAL_DISPLAY_POST_RESTORE, &restoreParams, sizeof(restoreParams)) @ unix_console.c:197
[12740.899571] NVRM nvAssertFailedNoLog: Assertion failed: !nvp->pGpu || !rmDeviceGpuLockIsOwner(nvp->pGpu->gpuInstance) @ dynamic-power.c:150
[12740.987953] NVRM unixCallVideoBIOS: int10h(4f02, 0000) vesa call failed! (4f02, 0000)
[12740.988331] NVRM nvCheckOkFailedNoLog: Check failed: Failure: Generic Error [NV_ERR_GENERIC] (0x0000FFFF) returned from pRmApi->Control(pRmApi, nv->rmapi.hClient, nv->rmapi.hSubDevice, NV2080_CTRL_CMD_INTERNAL_DISPLAY_POST_RESTORE, &restoreParams, sizeof(restoreParams)) @ unix_console.c:197

After the process kill I am thrown back to the login screen

ioquatix, 2023-12-02:
This could be related to https://github.com/obsproject/obs-studio/issues/9895 as well.