NVIDIA/open-gpu-kernel-modules #441: Build fails with "/lib/modules/*kernel_version*/build: No such file or directory"

Wooty-B, 2023-01-15:
### NVIDIA Open GPU Kernel Modules Version

525.78.01

### Operating System and Version

Debian GNU/Linux 11

### Kernel Release

5.10.0-20-arm64

### Build Command

`cd ~/Build`
`wget https://us.download.nvidia.com/XFree86/aarch64/525.78.01/NVIDIA-Linux-aarch64-525.78.01.run`
`sudo sh ./NVIDIA-Linux-aarch64-525.78.01.run --no-kernel-modules`
`git clone https://github.com/NVIDIA/open-gpu-kernel-modules.git`
`cd open-gpu-kernel-modules`
`make modules -j16`

### Terminal output/Build Log

```
make -C src/nvidia
make -C src/nvidia-modeset
make[1]: Entering directory '/home/wooty/Build/open-gpu-kernel-modules/src/nvidia'
make[1]: Entering directory '/home/wooty/Build/open-gpu-kernel-modules/src/nvidia-modeset'
 [ nvidia-modeset    ]  XZ           _out/Linux_aarch64/maxwell_shaders.xz
 [ nvidia-modeset    ]  XZ           _out/Linux_aarch64/pascal_shaders.xz
 [ nvidia-modeset    ]  XZ           _out/Linux_aarch64/volta_shaders.xz
...
...
...
 [ nvidia            ]  CC           _out/Linux_aarch64/g_nvid_string.c
 [ nvidia            ]  LD           _out/Linux_aarch64/nv-kernel.o
 [ nvidia            ]  OBJCOPY      _out/Linux_aarch64/nv-kernel.o
make[1]: Leaving directory '/home/wooty/Build/open-gpu-kernel-modules/src/nvidia'
cd kernel-open/nvidia/ && ln -sf ../../src/nvidia/_out/Linux_aarch64/nv-kernel.o nv-kernel.o_binary
make -C kernel-open modules
make[1]: Entering directory '/home/wooty/Build/open-gpu-kernel-modules/kernel-open'
make[2]: Entering directory '/home/wooty/Build/open-gpu-kernel-modules/kernel-open'
make[2]: *** /lib/modules/5.10.0-20-arm64/build: No such file or directory.  Stop.
make[2]: Leaving directory '/home/wooty/Build/open-gpu-kernel-modules/kernel-open'
make[1]: *** [Makefile:82: modules] Error 2
make[1]: Leaving directory '/home/wooty/Build/open-gpu-kernel-modules/kernel-open'
make: *** [Makefile:59: modules] Error 2

```

### More Info

This is after a clean install of Debian 11.6 image, the only things having been done was update `apt/sources.list` and install `build-essential` and `git` packages. Appears to be looking for a non-existent directory located here: `/lib/modules/5.10.0-20-arm64/build`

Not sure how I should proceed. Thanks.

Wooty-B, 2023-01-15:
Just a follow up: I installed the NVIDIA 525.78.01 driver package and let it build the modules, I also updated initramfs after blacklisting. Xorg appears to be unable to find a "drm" device per systemd, I checked and nouveau is unloaded and the following 3 modules were loaded; nvidia, nvidia_drm and nvidia_modeset.

I'm assuming this isn't the recommended way to install based off the setup instructions, but thought I'd post.

aritger, 2023-01-17:
Hi Wooty-B.  I think you need to install the package for your kernel's configured sources, so that open-gpu-kernel-modules can build against it.

Check, for example, `dpkg -S /lib/modules/5.10.0-20-arm64/build` to find the package to install.

For this:
```
Xorg appears to be unable to find a "drm" device per systemd
```
What is the symptom?  Does Xorg fail to start?  Something else?

Anyway, I think that is unrelated to the build problem tracked in this Issue.

Wooty-B, 2023-01-19:
I apologize for that oversight, and after installing headers I was able to build and sucessfully install. I am now getting a different issue that yields no information online, so I will post that to the appropriate category. Thanks again. Closing.

closed 2023-01-19 by Wooty-B

sysyushuo, 2023-07-30:
hi，I came across the same problem ，how to installing headers？thanks

PawBud, 2024-06-07:
> hi，I came across the same problem ，how to installing headers？thanks

For bash on Ubuntu/Debian Based Systems

`
sudo apt update &&
sudo apt install build-essential dkms linux-headers-$(uname -r)
`