NVIDIA/open-gpu-kernel-modules #446: Timeout waiting for RPC from GSP!

875d, 2023-01-26:
### NVIDIA Open GPU Kernel Modules Version

525.85.05

### Does this happen with the proprietary driver (of the same version) as well?

I cannot test this

### Operating System and Version

Arch Linux

### Kernel Release

Linux [HOSTNAME] 6.1.8-arch1-1 #1 SMP PREEMPT_DYNAMIC Tue, 24 Jan 2023 21:07:04 +0000 x86_64 GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3050 Laptop GPU (UUID: GPU-071149ae-386e-0017-3b5b-7ea80801f725)

### Describe the bug

When I open a OpenGL application, like Yamagi Quake II, at a certain point the whole system freezes, and run in like 1 FPS per second. I generally have to REISUB when this happens.

### To Reproduce

1. Open Yamagi Quake II
2. Change workspace, open pavucontrol to select a new audio sink for the game, switch back

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10510199/nvidia-bug-report.log.gz)

### More Info

Related: #272

yfaimisaka, 2023-02-21:
same issue.

nvidia:
`NVIDIA GeForce RTX 3050 Mobile`
driver:
`nvidia-open-dkms 525.89.02-3` 
kernel:
` Linux [HOSTNAME] 6.1.12-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Tue, 14 Feb 2023 22:08:11 +0000 x86_64 GNU/Linux`

some thing repeatedly like this:
```
NVRM: Xid (PCI:0000:01:00): 119, pid=2061, name=alacritty, Timeout waiting for RPC from GSP! Expected function 103 (GSP_RM_ALLOC) (0x80 0x38).
NVRM _issueRpcAndWait: rpcRecvPoll timedout for fn 103!
NVRM nvAssertFailedNoLog: Assertion failed: 0 @ rpc.c:205
NVRM rpcRmApiAlloc_GSP: GspRmAlloc failed: hClient=0xc1d00013; hParent=0xc1d00013; hObject=0xfade0001; hClass=0x00000080; paramsSize=0x00000038; paramsStatus=0x00000000; status=0x00000065
NVRM serverFreeResourceTree: hObject 0xfade0002 not found for client 0xc1d00013
NVRM: Xid (PCI:0000:01:00): 119, pid=2061, name=alacritty, Timeout waiting for RPC from GSP! Expected function 10 (FREE) (0xfade0001 0x0).
NVRM _issueRpcAndWait: rpcRecvPoll timedout for fn 10!
NVRM nvAssertFailedNoLog: Assertion failed: 0 @ rpc.c:205
NVRM rpcRmApiFree_GSP: GspRmFree failed: hClient=0xc1d00013; hObject=0xfade0001; paramsStatus=0x00000000; status=0x00000065
```

fighterhit, 2023-02-23:
I met this in `525.85.12`.

jelmd, 2023-02-25:
This issue seems to exists on `525.60.13` for **A40** and **A100** as well. Please fix ASAP!
[bug-nv-smi.txt](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10832170/bug-nv-smi.txt)
[dmesg.txt](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10832171/dmesg.txt)

fighterhit, 2023-02-27:
Hi @aritger, is there any solution about this?

aritger, 2023-02-27:
I think "Timeout waiting for RPC from GSP!" is a pretty generic symptom, with many possible causes.  The reproduction steps that lead up to it will matter to help distinguish different bugs, as will the other NVRM dmesg spew around it.

I don't have specific reason to expect it is already fixed, but it may be worth testing the most recent 525.89.02 driver.  530.xx drivers will hopefully be released soon, and they will have a lot of changes relative to 525.xx, so that will also be worth testing.

Beyond that, if you see similar "Timeout waiting for RPC from GSP!" messages, it is worth attaching a complete nvidia-bug-report.log.gz, and describing the steps that led to it, so that we can compare instances of the symptom.

fighterhit, 2023-02-27:
Thanks @aritger，[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10836852/nvidia-bug-report.log.gz).

The problem occurs after running in the kubernetes environment for a period of time, and `nvidia-smi` will get stuck for a while. The specific error phenomenon is similar to @jelmd https://github.com/NVIDIA/open-gpu-kernel-modules/issues/446#issuecomment-1445190598. FYI, I got some help from nvidia-docker community(https://github.com/NVIDIA/nvidia-docker/issues/1648#issuecomment-1441139460), but not sure if the root cause of this problem is driver or NVIDIA-docker related.

jelmd, 2023-02-27:
Happend again on another machine:
```
...
[  +0.000094]  ? _nv011159rm+0x62/0x2e0 [nvidia]
[  +0.000090]  ? _nv039897rm+0xdb/0x140 [nvidia]
[  +0.000073]  ? _nv041022rm+0x2ce/0x3a0 [nvidia]
[  +0.000103]  ? _nv015438rm+0x788/0x800 [nvidia]
[  +0.000064]  ? _nv039416rm+0xac/0xe0 [nvidia]
[  +0.000092]  ? _nv041024rm+0xac/0x140 [nvidia]
[  +0.000095]  ? _nv041023rm+0x37a/0x4d0 [nvidia]
[  +0.000070]  ? _nv039319rm+0xc9/0x150 [nvidia]
[  +0.000151]  ? _nv039320rm+0x42/0x70 [nvidia]
[  +0.000180]  ? _nv000552rm+0x49/0x60 [nvidia]
[  +0.000219]  ? _nv000694rm+0x7fb/0xc80 [nvidia]
[  +0.000195]  ? rm_ioctl+0x54/0xb0 [nvidia]
[  +0.000132]  ? nvidia_ioctl+0x6e3/0x850 [nvidia]
[  +0.000003]  ? get_max_files+0x20/0x20
[  +0.000134]  ? nvidia_frontend_unlocked_ioctl+0x3b/0x50 [nvidia]
[  +0.000002]  ? do_vfs_ioctl+0x407/0x670
[  +0.000003]  ? __secure_computing+0xa4/0x110
[  +0.000002]  ? ksys_ioctl+0x67/0x90
[  +0.000002]  ? __x64_sys_ioctl+0x1a/0x20
[  +0.000002]  ? do_syscall_64+0x57/0x190
[  +0.000002]  ? entry_SYSCALL_64_after_hwframe+0x5c/0xc1
[  +6.010643] NVRM: Xid (PCI:0000:83:00): 119, pid=2710030, name=nvidia-smi, Timeout waiting for RPC from GSP! Expected function 10 (FREE) (0xa55a0050 0x0).
...
Mon Feb 27 16:08:33 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.60.13    Driver Version: 525.60.13    CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA A40          On   | 00000000:03:00.0 Off |                    0 |
|  0%   24C    P8    14W / 300W |      0MiB / 46068MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   1  NVIDIA A40          On   | 00000000:04:00.0 Off |                    0 |
|  0%   25C    P8    13W / 300W |      0MiB / 46068MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   2  NVIDIA A40          On   | 00000000:43:00.0 Off |                    0 |
|  0%   40C    P0    81W / 300W |    821MiB / 46068MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   3  NVIDIA A40          On   | 00000000:44:00.0 Off |                    0 |
|  0%   30C    P8    14W / 300W |      0MiB / 46068MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   4  ERR!                On   | 00000000:83:00.0 Off |                 ERR! |
|ERR!  ERR! ERR!    ERR! / ERR! |      0MiB / 46068MiB |    ERR!      Default |
|                               |                      |                 ERR! |
+-------------------------------+----------------------+----------------------+
|   5  NVIDIA A40          On   | 00000000:84:00.0 Off |                    0 |
|  0%   23C    P8    14W / 300W |      3MiB / 46068MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   6  NVIDIA A40          On   | 00000000:C3:00.0 Off |                    0 |
|  0%   22C    P8    15W / 300W |      3MiB / 46068MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   7  NVIDIA A40          On   | 00000000:C4:00.0 Off |                    0 |
|  0%   21C    P8    14W / 300W |      3MiB / 46068MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    2   N/A  N/A   2111354      C   ...eratornet/venv/bin/python      818MiB |
+-----------------------------------------------------------------------------+
```

fighterhit, 2023-03-03:
@jelmd +1, I ran into this problem again. Hi @aritger maybe this is a Driver issue, please take a look.
```
Fri Mar  3 14:41:55 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.85.12    Driver Version: 525.85.12    CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA A30          Off  | 00000000:01:00.0 Off |                    0 |
| N/A   26C    P0    29W / 165W |      0MiB / 24576MiB |      0%      Default |
|                               |                      |             Disabled |
+-------------------------------+----------------------+----------------------+
|   1  NVIDIA A30          Off  | 00000000:22:00.0 Off |                    0 |
| N/A   40C    P0    92W / 165W |  16096MiB / 24576MiB |     88%      Default |
|                               |                      |             Disabled |
+-------------------------------+----------------------+----------------------+
|   2  NVIDIA A30          Off  | 00000000:41:00.0 Off |                    0 |
| N/A   42C    P0   141W / 165W |  14992MiB / 24576MiB |     33%      Default |
|                               |                      |             Disabled |
+-------------------------------+----------------------+----------------------+
|   3  ERR!                Off  | 00000000:61:00.0 Off |                 ERR! |
|ERR!  ERR! ERR!    ERR! / ERR! |  23025MiB / 24576MiB |    ERR!      Default |
|                               |                      |                 ERR! |
+-------------------------------+----------------------+----------------------+
|   4  NVIDIA A30          Off  | 00000000:81:00.0 Off |                    0 |
| N/A   26C    P0    26W / 165W |      0MiB / 24576MiB |      0%      Default |
|                               |                      |             Disabled |
+-------------------------------+----------------------+----------------------+
|   5  NVIDIA A30          Off  | 00000000:A1:00.0 Off |                    0 |
| N/A   26C    P0    28W / 165W |      0MiB / 24576MiB |      0%      Default |
|                               |                      |             Disabled |
+-------------------------------+----------------------+----------------------+
|   6  NVIDIA A30          Off  | 00000000:C1:00.0 Off |                    0 |
| N/A   25C    P0    29W / 165W |      0MiB / 24576MiB |      0%      Default |
|                               |                      |             Disabled |
+-------------------------------+----------------------+----------------------+
|   7  NVIDIA A30          Off  | 00000000:E1:00.0 Off |                    0 |
| N/A   24C    P0    25W / 165W |      0MiB / 24576MiB |      0%      Default |
|                               |                      |             Disabled |
+-------------------------------+----------------------+----------------------+
                                                                               
```

```
[Fri Mar  3 04:23:22 2023] NVRM: GPU at PCI:0000:61:00: GPU-e59ce3f9-af53-a0dd-1d2c-8beaa74aa635
[Fri Mar  3 04:23:22 2023] NVRM: GPU Board Serial Number: 1322621149782
[Fri Mar  3 04:23:22 2023] NVRM: Xid (PCI:0000:61:00): 119, pid=1344368, name=nvidia-smi, Timeout waiting for RPC from GSP! Expected function 10 (FREE) (0xa55a00a0 0x0).
[Fri Mar  3 04:23:22 2023] CPU: 72 PID: 1344368 Comm: nvidia-smi Tainted: P           OE     5.10.0-20-amd64 #1 Debian 5.10.158-2
[Fri Mar  3 04:23:22 2023] Hardware name: Inspur NF5468A5/YZMB-02382-101, BIOS 4.02.12 01/28/2022
[Fri Mar  3 04:23:22 2023] Call Trace:
[Fri Mar  3 04:23:22 2023]  dump_stack+0x6b/0x83
[Fri Mar  3 04:23:22 2023]  _nv011231rm+0x39d/0x470 [nvidia]
[Fri Mar  3 04:23:22 2023]  ? _nv011168rm+0x62/0x2e0 [nvidia]
[Fri Mar  3 04:23:22 2023]  ? _nv040022rm+0xdb/0x140 [nvidia]
[Fri Mar  3 04:23:22 2023]  ? _nv041148rm+0x2ce/0x3a0 [nvidia]
[Fri Mar  3 04:23:22 2023]  ? _nv015451rm+0x788/0x800 [nvidia]
[Fri Mar  3 04:23:22 2023]  ? _nv039541rm+0xac/0xe0 [nvidia]
[Fri Mar  3 04:23:22 2023]  ? _nv041150rm+0xac/0x140 [nvidia]
[Fri Mar  3 04:23:22 2023]  ? _nv041149rm+0x37a/0x4d0 [nvidia]
[Fri Mar  3 04:23:22 2023]  ? _nv039443rm+0xc9/0x150 [nvidia]
[Fri Mar  3 04:23:22 2023]  ? _nv039444rm+0x42/0x70 [nvidia]
[Fri Mar  3 04:23:22 2023]  ? _nv000554rm+0x49/0x60 [nvidia]
[Fri Mar  3 04:23:22 2023]  ? _nv000694rm+0x7fb/0xc80 [nvidia]
[Fri Mar  3 04:23:22 2023]  ? rm_ioctl+0x54/0xb0 [nvidia]
[Fri Mar  3 04:23:22 2023]  ? nvidia_ioctl+0x6cd/0x830 [nvidia]
[Fri Mar  3 04:23:22 2023]  ? nvidia_frontend_unlocked_ioctl+0x37/0x50 [nvidia]
[Fri Mar  3 04:23:22 2023]  ? __x64_sys_ioctl+0x8b/0xc0
[Fri Mar  3 04:23:22 2023]  ? do_syscall_64+0x33/0x80
[... 62 lines cut ...]
[Fri Mar  3 04:50:24 2023] NVRM: Xid (PCI:0000:61:00): 119, pid=1365232, name=nvidia-smi, Timeout waiting for RPC from GSP! Expected function 76 (GSP_RM_CONTROL) (0x2080014b 0x5).
[Fri Mar  3 04:51:09 2023] NVRM: Xid (PCI:0000:61:00): 119, pid=1365232, name=nvidia-smi, Timeout waiting for RPC from GSP! Expected function 76 (GSP_RM_CONTROL) (0x2080014b 0x5).
[... cut ...]
[Fri Mar  3 05:06:11 2023] NVRM: Xid (PCI:0000:61:00): 119, pid=1385412, name=nvidia-smi, Timeout waiting for RPC from GSP! Expected function 76 (GSP_RM_CONTROL) (0x20800110 0x84).
[Fri Mar  3 05:06:56 2023] NVRM: Xid (PCI:0000:61:00): 119, pid=1385412, name=nvidia-smi, Timeout waiting for RPC from GSP! Expected function 76 (GSP_RM_CONTROL) (0x2080014b 0x5).
[Fri Mar  3 05:07:41 2023] NVRM: Xid (PCI:0000:61:00): 119, pid=1385412, name=nvidia-smi, Timeout waiting for RPC from GSP! Expected function 76 (GSP_RM_CONTROL) (0x20800157 0x0).
```

lpla, 2023-03-03:
Also happening here on a A100-PCIE-40GB using driver 530.30.02 and CUDA 12.1.

fighterhit, 2023-03-03:
> 

Hi @lpla , what's your use case environment? In kubernetes?

lpla, 2023-03-03:
There is no environment. It triggered the bug several times in both 525 and 530 driver. It is a Machine Learning inference command line written in PyTorch.

fighterhit, 2023-03-03:
> There is no environment. It triggered the bug several times in both 525 and 530 driver. It is a Machine Learning inference command line written in PyTorch.

Have you tried the 520.* driver ? Is it work?

jelmd, 2023-03-03:
FWIW: Most of our users use PyTorch as well. Perhaps it tortures GPUs too hard ;-)

jelmd, 2023-03-03:
Yepp. np with 470, too.

lpla, 2023-03-03:
> > There is no environment. It triggered the bug several times in both 525 and 530 driver. It is a Machine Learning inference command line written in PyTorch.
> 
> Have you tried the 520.* driver ? Is it work?

That's my next test. In fact, that's exactly the version I was using before upgrading from Ubuntu 20.04 with kernel 5.15 and driver 520 to Ubuntu 22.04 with kernel 5.19 and driver 525 last month. It was working perfect with that previous setup.

jelmd, 2023-03-07:
Same thing on another machine. FWIW: I removed `/usr/lib/firmware/nvidia/525.60.13` - perhaps this fixes the problem.

fighterhit, 2023-03-07:
Hi @jelmd @lpla  , as the NVIDIA customer, we communicate with the NVIDIA support team today and according to the `nvidia-bug-report.log.gz`  they advised us to disable `GSP-RM`. 
1. To disable GSP-RM: 
```
sudo su -c 'echo options nvidia NVreg_EnableGpuFirmware=0 > /etc/modprobe.d/nvidia-gsp.conf'
```
2.  Enable the kernel
```
# if ubuntu
sudo update-initramfs -u

# if centos
dracut -f 
```
3. Reboot
4. Check if work. If `EnableGpuFirmware: 0` then `GSP-RM` is disabled.
```
cat /proc/driver/nvidia/params | grep EnableGpuFirmware
```

Since our problem node still has tasks running, I haven't tried it yet, I will try this method tonight or tomorrow morning, just for reference. :)

stephenroller, 2023-03-08:
I'm also seeing XID 119s on some 520 drivers. Have not tried 525.

liming5619, 2023-03-09:
Driver 525.60.13 with A40 GSP disable.But nvidia-but-report also have GSP timeout error.

fighterhit, 2023-03-09:
Hi @stephenroller @liming5619 , maybe it's better to downgrade the driver version. On one hand, the GSP feature was introduced by NVIDIA since 510 but has not been fixed yet. On the other hand, 470 is an [LTS version](https://docs.nvidia.com/datacenter/tesla/drivers/index.html#:~:text=Term%20Support%20Branch-,Long%20Term%20Support%20Branch,-Production%20Branch) and has been running stably in our production environment for a long time. I have already downgraded the driver of the problematic node to 470.82.01 to match our other production nodes, just for your reference. :)

stephenroller, 2023-03-10:
So far disabling GSP seems to have mitigated, but maybe I've just been lucky since. Will report back if I see counter-evidence.

jelmd, 2023-03-10:
Yepp, removing /usr/lib/firmware/nvidia/5xx.* seems to fix the problem, too (did not use NVreg_EnableGpuFirmware=0).

mdrasheek, 2023-04-12:
I have a similar issue, after disabling GSP, It took more than 5 minutes to give output as "True".

```
# cat /etc/modprobe.d/nvidia-gsp.conf
options nvidia NVreg_EnableGpuFirmware=0
# cat /proc/driver/nvidia/params | grep EnableGpuFirmware
EnableGpuFirmware: 0
EnableGpuFirmwareLogs: 2
```
Strange thing is,  I'm booting up vms with images which has GPU driver pre-installed, on a host with 4 cards, 2 out of 4 cards ends up with a similar issue.

Please suggest a fix as its hampering our prod environments. Please let me know if any additional commands or log output that i should provide.

yskan, 2023-04-12:
@mdrasheek it could be that the driver config inside the VMs is overriding your changes to GSP. 
As it is, disabling GSP should resolve the problem. Check in your logs which xid error you are having.

mdrasheek, 2023-04-12:
> @mdrasheek it could be that the driver config inside the VMs is overriding your changes to GSP. As it is, disabling GSP should resolve the problem. Check in your logs which xid error you are having.

Before disabling GSP the error was same as in this post:
```
NVRM: Xid (PCI:0000:01:01): 119, pid=8019, name=python3, Timeout waiting for RPC from GSP0! Expected function 103 (GSP_RM_ALL
OC) (0x0 0x6c).
```
But after disabling it, i couldn't find any logs, but it takes lot of time to say cuda is "True".

mdrasheek, 2023-04-13:
> Before disabling GSP the error was same as in this post:
>
> ```
> NVRM: Xid (PCI:0000:01:01): 119, pid=8019, name=python3, Timeout waiting for RPC from GSP0! Expected function 103 (GSP_RM_ALL
> OC) (0x0 [...]

Is there something where I can enable trace, or increase the log level to know the cause of this delay?

fighterhit, 2023-04-13:
Hello everyone, I found that the NVIDIA official driver [525.105.17](https://docs.nvidia.com/datacenter/tesla/tesla-release-notes-525-105-17/index.html#:~:text=Fixed%20an%20issue%20specific%20to%20GSP%2DRM%20that%20could%20lead%20to%20GSP%20RPC%20timeout%20errors%20(Xid%20119).%20The%20issue%20was%20introduced%20in%20the%20first%20525%20driver%20release%20and%20was%20not%20present%20in%20earlier%20drivers.) recently released seems to have fixed this bug. Anyone who is interested could try it and give feedback on whether it works. :)

mdrasheek, 2023-04-14:
> Hello everyone, I found that the NVIDIA official driver [...]

Nope, facing the same error

```# modinfo nvidia|grep version
version:        525.105.17
rhelversion:    8.8
srcversion:     98F82D76E0EF3952EEE57A7
vermagic:       4.18.0-448.el8.x86_64 SMP mod_unload modversions

Apr 14 04:20:41 e2e-105-143 kernel: NVRM: Xid (PCI:0000:01:01): 119, pid=26071, name=python, Timeout waiting for RPC from GSP! Expected function 10 (FREE) (0x5c000001 0x0).```

fighterhit, 2023-04-14:
> 

Hi @mdrasheek ,  what's your use case environment? In kubernetes?

stephenroller, 2023-04-14:
Did you reboot after applying the mitigation? I haven't seen any issues in pytorch since the first reboot without GSP.

mdrasheek, 2023-04-16:
> Did you reboot after applying the mitigation? I haven't seen any issues in pytorch since the first reboot without GSP.

Yes reboot was done, you can also see the output posted earlier

```
cat /proc/driver/nvidia/params | grep EnableGpuFirmware
EnableGpuFirmware: 0
EnableGpuFirmwareLogs: 2
```

As i have already mentioned this happens in 2 cards out of 4. In the base machine i have 4 cards, and each vm has 1 card. 2 cards are working as expected,  but the other 2 cards shows up the delay to say "True". I agree this is strange, but this is what i have observed.

mdrasheek, 2023-04-17:
> 

May I know which mitigation you mean?

mdrasheek, 2023-04-17:
> Did you reboot after applying the mitigation? I haven't seen any issues in pytorch since the first reboot without GSP.

May I know which mitigation you mean?

pankajsahtech, 2023-04-19:
Hello 

New driver has bug fixed by Nvidia.
Fixed an issue specific to GSP-RM that could lead to GSP RPC timeout errors (Xid
119). The issue was introduced in the first 525 driver release and was not present in
earlier drivers.
https://docs.nvidia.com/datacenter/tesla/pdf/NVIDIA_Data_Center_GPU_Driver_Release_Notes_525_v3.0.pdf

fighterhit, 2023-04-19:
> Hello
> 
> New driver 525.105.17 (Linux) has bug fixed by Nvidia. Fixed an issue specific to GSP-RM that could lead to GSP RPC timeout errors (Xid 119). The issue was introduced in the first 525 driver release and was not present in earlier drivers. https://docs.nvidia.com/datacenter/tesla/pdf/NVIDIA_Data_Center_GPU_Driver_Release_Notes_525_v3.0.pdf

Yes @pankajsahtech, I noticed this but @mdrasheek said he still has the same problem(https://github.com/NVIDIA/open-gpu-kernel-modules/issues/446#issuecomment-1508127902).

pankajsahtech, 2023-04-19:
@mdrasheek , @fighterhit 
Is it DGX  box.
Pls check the matrix for xid errors. It could be that GPU having H/W error.
https://docs.nvidia.com/deploy/xid-errors/index.html#topic_4

Tverous, 2023-04-19:
Same issues with the following drivers on NVIDIA A40

- nvidia-driver-cuda.x86_64: 530.30.02
- nvidia-container-toolkit.x86_64: 1.13.0-1

pankajsahtech, 2023-04-20:
Looks you are using 530.30.02 beta version, this version not support A40 Card.
https://www.nvidia.com/download/driverResults.aspx/199985/en-us/

Pls check version 525.105.17.

mdrasheek, 2023-04-25:
> @mdrasheek , @fighterhit Is it DGX box. Pls check the matrix for xid errors. It could be that GPU having H/W error. https://docs.nvidia.com/deploy/xid-errors/index.html#topic_4

No, Its not DGX. Its A100 80 GB PCI, a VM is using it through pass through.

mikelou2012, 2023-05-17:
> I'm also seeing XID 119s on some 510 drivers. Have not tried 525 or 520.

The same XID 119 issue for 520.105 driver.

zldrobit, 2023-06-09:
I still encounter this problem on 1 out of 2 A100s after upgrading drive to 530.41.03.

mdrasheek, 2023-06-15:
I want to highlight a point here, post enabling MIG and creating a full MIG Slice the GPU works without lag.

nvidia-smi -mig 1
nvidia-smi mig -cgi 0 -C

The above commands will create slice 0 which is the entire 80GB MIG slice. But with MIG mode disabled the lag comes again. I hope this may lead a way for the resolution for some one.

paulraines68, 2023-06-19:
We recently got a ThinkSystem SR670-V2 with 8 A100 80GB cards.  We are running 530.30.02  from yum CUDA repo in a SLURM batch.  We have had these Xid 119 errors on 3 different GPUs so far running for about 3 weeks with typical TensorFlow jobs.

Going to try disabling GSP as mentioned above.

Davidrjx, 2023-06-28:
> I still encounter this problem on 1 out of 2 A100s after upgrading drive to 530.41.03.

@mikelou2012 maybe have a try on 525.125.06 with GSP enable

mdrasheek, 2023-07-08:
Hi everyone, please try with the following driver:

wget https://us.download.nvidia.com/tesla/535.54.03/NVIDIA-Linux-x86_64-535.54.03.run

As tested from our side, It works as expected without any changes to GSP firmware and MIG in disabled state.

mdrasheek, 2023-07-11:
Unforunately ended up with similar issue in 535.54 in one of our machines.

```Jul 11 11:42:02 e2e-85-84 kernel: [ 1406.886236] NVRM: Xid (PCI:0000:01:01): 119, pid=3395, name=python3, Timeout waiting for RPC from GSP0! Expected function 103 (GSP_RM_ALLOC) (0x0 0x6c).
Jul 11 11:42:08 e2e-85-84 kernel: [ 1412.890184] NVRM: Xid (PCI:0000:01:01): 119, pid=3395, name=python3, Timeout waiting for RPC from GSP0! Expected function 103 (GSP_RM_ALLOC) (0x80 0x38).
Jul 11 11:42:14 e2e-85-84 kernel: [ 1418.894227] NVRM: Rate limiting GSP RPC error prints for GPU at PCI:0000:01:01 (printing 1 of every 30).  The GPU likely needs to be reset.
Jul 11 11:42:38 e2e-85-84 kernel: [ 1442.922569] NVRM: Xid (PCI:0000:01:02): 119, pid=3395, name=python3, Timeout waiting for RPC from GSP1! Expected function 10 (FREE) (0x5c00000c 0x0).
Jul 11 11:42:44 e2e-85-84 kernel: [ 1448.926199] NVRM: Xid (PCI:0000:01:02): 119, pid=3395, name=python3, Timeout waiting for RPC from GSP1! Expected function 10 (FREE) (0x5c00000b 0x0).
Jul 11 11:42:50 e2e-85-84 kernel: [ 1454.930190] NVRM: Rate limiting GSP RPC error prints for GPU at PCI:0000:01:02 (printing 1 of every 30).  The GPU likely needs to be reset.```

mdrasheek, 2023-08-16:
Hi everyone, We have identified the issue related to our case. In our machines, all the GPU cards are connected via NVLinks and as per the Nvidia doc [here](https://docs.nvidia.com/grid/13.0/grid-vgpu-user-guide/index.html#using-gpu-pass-through:~:text=All%20GPUs%20directly%20connected%20to%20each%20other%20through%20NVLink%20must%20be%20assigned%20to%20the%20same%20VM) 

```All GPUs directly connected to each other through NVLink must be assigned to the same VM```

So, after removing NVlinks, its working as expected. Hope this helps some one.

paulraines68, 2023-08-16:
@mdrasheek We were running on a box with 8 GPUS and NVLinks.  We are not doing VMs but running SLURM jobs that allocate cores/memory by cgroups and somehow isolate GPUs assigned too.  My guess is that is similar to what the VMs do.  

Still we don't want to remove the NVLinks as jobs that allocate mulitple GPUs should still be able to take advantage of them.

mdrasheek, 2023-08-19:
@paulraines68 As you have mentioned you are using the driver version 530.x, which is not a datacenter driver. Check the supported products of driver 530.x in [this page](https://www.nvidia.com/download/driverResults.aspx/199985/en-us/) you will not find A100. You should use data center drivers on which they say the Xid119 errors are fixed. Refer [here](https://docs.nvidia.com/datacenter/tesla/drivers/index.html#cuda-drivers) to know more about datacenter drivers.

If you want to install using yum, you can refer to official installation instructions [here](https://developer.nvidia.com/cuda-downloads?target_os=Linux&target_arch=x86_64&Distribution=CentOS&target_version=7&target_type=rpm_network).

alpiquero, 2023-09-06:
I can confirm the issue on our machines running the 535 driver version.

mdrasheek, 2023-09-11:
> I can confirm the issue on our machines running the 535 driver version. Disabling GSP workarounds the problem.

You can check which cards are interconnected with NVlinks using the command ```nvidia-smi topo -m```. For example, GPU0 and GPU1 is connected with NV8, GPU2 and GPU3 is connected with NV12, if you allocate jobs to GPU0&1 it shouldn't have the issue. But when you mix up (like allocating jobs to GPU1 and GPU2) you might have an issue.

Hope this helps!

Qubitium, 2023-09-15:
Just encountered this bug on A100 PCIE 80GB on all latest 525/530/535 official Linux drivers. Nvidia, this is a disgrace of a support for a 10K, or 13K (due to price gouging) gpu. Please fix this.

mckbrchill, 2023-10-02:
> Hi everyone, We have identified the issue related to our case with the help of Nvidia enterprise support team. In our machines, all the GPU cards are connected via NVLinks and as per the Nvidia doc [here](https://docs.nvidia.com/grid/13.0/grid-vgpu-user-guide/index.html#using-gpu-pass-through:~:text=All%20GPUs%20directly%20connected%20to%20each%20other%20through%20NVLink%20must%20be%20assigned%20to%20the%20same%20VM)
> 
> `All GPUs directly connected to each other through NVLink must be assigned to the same VM`
> 
> So, after removing NVlinks, its working as expected. Hope this helps some one.

Hi @mdrasheek ! 
I'm facing the same issue on A100: timeout from RPC and when I disable it, torch.cuda.is_available() takes 3 minutes to execute and other torch code is very slow on first initialization.
Does removing NVlinks help to fix pytoch speed? And may be you've found some other workarounds?
Thanks!

mdrasheek, 2023-10-13:
> Hi @mdrasheek ! I'm facing the same issue on A100: timeout from RPC and when I disable it, torch.cuda.is_available() takes 3 minutes to execute and other torch code is very slow on first [...]

Yes, when we remove NVLinks we are able to use the cards in separate VMs.

Or check the NVLinks connectivity using ```nvidia-smi topo -m```. If 2 cards interconnected using the same NVLinks, you can use those 2 cards in a single VM. 

A sample screenshot is given below from 4x VM:

[image]

You can see the GPU 0 and 1 is connected to an NVlink and GPU 2 and 3 are connected with another NVlink. So, the following deployment options will work

1. VM launched with GPU 0 and 1
2. VM launched with GPU 2 and 3
3. VM with all the 4 cards.

 Or as I mentioned earlier, if you are using single GPU for single process you can enable MIG, create MIG slices and use it acccordingly.

mckbrchill, 2023-10-13:
> Or check the NVLinks connectivity using nvidia-smi topo -m. If 2 cards interconnected using the same NVLinks, you can use those 2 cards in a single VM.

I rent VM with a single GPU from a cloud provider and `nvidia-smi topo -m` check shows that the connectivity for my GPU is 'Self' (X). Unfortunately, I'm still experiencing 3+ minutes delay on first `torch.cuda.is_available()' in a process.
Although I can see in `nvidia-smi -i 0 -q' that `GPU Virtualization Mode` is `Pass-Through`. And the cloud works so, that my VM is not always assigned to some specific GPU, but it can vary on which one GPU from the A100s pool is free now. 
Not sure, but may be there is something behind the proccess of how it's organized on the cloud provider side, and I just can't directly see if there is NVLink.
I also tested at other cloud provider VM and there were no problems with GSP timeout errors. The difference in terms of GPU on these VMs was: on a problematic VM there is A100-SXM4-80GB and `GPU Virtualization Mode` is `Pass-Through`, on the second one I had A100-PCIE-40GB and `GPU Virtualization Mode` - `None`

Haven't tried MIG slices yet, may be it will help.
Thanks for your answer!

mdrasheek, 2023-10-16:
> I rent VM with a single GPU from a cloud provider and `nvidia-smi topo -m` check shows that the connectivity for my GPU is 'Self' (X). Unfortunately, I'm still experiencing 3+ minutes delay on first [...]

SXM4 mostly comes in a bundle of 4 or 8 cards connected with NVlinks and it is possible that your cloud provider gives 1 card from it. You have to go with MIG and it should help you to overcome the lag issue.

neon-ninja, 2023-12-05:
> I want to highlight a point here, post enabling MIG and creating a full MIG Slice the GPU works without lag.
>
> nvidia-smi -mig 1 nvidia-smi mig -cgi 0 -C
>
> The above commands will create slice 0 which [...]

Thanks, this worked for me

robkooper, 2024-03-22:
I noticed the following, running in openstack with 4 A100 GPUs. I created 2 VMs, each with a GPU in there. I installed conda and pytorch for testing. When testing I open a python terminal and run the following code, do not close python leave it open.

```
import torch
torch.cuda.is_available()
```

Both driver 470, everything works, does not matter what order  you do this.

Both driver 535, the first machine to run the python code has access to the GPU, the second machine fails.

Machine 1 has driver 470, Machine 2 has driver 535, if you start on machine 1 (470) and then machine 2 (535) the second machine can not get access to the GPU. If you first do machine 2 (535) and then machine 1 (470) both machines have access to the GPU.

Took a while to find the exact sequence, but hopefully this might help somebody to find out what the issue is. But it feels like driver 535 is locking all the GPUs, and driver 470 does not look at the lock, or just ignores it.

hansbogert, 2024-05-23:
The RPC timout is just a red-herring in my opinion. Something is just wrong in our setup, removing the timeout just makes the environment wait longer;
Every invocation of a cuda kernel takes 5 minutes in my case in a kubernetes cluster.

jelmd, 2024-05-23:
>...
> However, every invocation of a cuda kernel takes 5 minutes in my case in a kubernetes cluster.
> 
> During those 5 minutes, the host application is at 100% CPU, doing ... well, I don't know, strace did give anything meaningful.

FWIW: We do not use kubernetes but lxc, and observed similar strange behavior with H100 pcie based boxes, sometimes 1 or 2 of 4 GPUs were not even usable (launch took forever). Often a bare metal box reboot got it back to normal behavior, but not always. A distro upgrade from Ubuntu 20.04.06 to 22.04.4 fixed this problem (upgraded the nv driver from 535.86.10 to 535.161.08 as well).  No problems anymore (for ~ 8 weeks).

[7 more replies]