NVIDIA/open-gpu-kernel-modules #447: Metro Exodus Crashes 525.xx

adolfotregosa, 2023-01-27:
### NVIDIA Open GPU Kernel Modules Version

Any 525

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch 

### Kernel Release

Linux VMArch 6.1.8-arch1-1 

### Hardware: GPU

3090

### Describe the bug

Start metro exodus ray tracing edition.

steam launch options:

PROTON_ENABLE_NVAPI=1 VKD3D_CONFIG=dxr,dxr11 PROTON_ENABLE_NGX_UPDATER=1 %command%

dmesg error:

NVRM: GPU at PCI:0000:00:0b: GPU-7008f4ca-0928-d2ff-80ae-9fc652eb3a5a
[51871.867694] NVRM: Xid (PCI:0000:00:0b): 13, pid='<unknown>', name=<unknown>, Graphics SM Warp Exception on (GPC 2, TPC 4, SM 0): Illegal Instruction Parameter

NVRM: Xid (PCI:0000:00:0b): 13, pid='<unknown>', name=<unknown>, Graphics Exception: ESR 0x516730=0x9000b 0x516734=0x0 0x516728=0xf812b60 0x51672c=0x1104

NVRM: Xid (PCI:0000:00:0b): 109, pid=188145, name=MetroExodus.exe, Ch 000000be, errorString CTX SWITCH TIMEOUT, Info 0x5cc059

### To Reproduce

start metro exodus ray tracing edition on steam NOT in safe mode.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10515375/nvidia-bug-report.log.gz)

adolfotregosa, 2023-01-27:
Witcher 3 when enabling Ray Tracing 

NVRM: Xid (PCI:0000:00:0b): 109, pid=190733, name=witcher3.exe, Ch 000000be, errorString CTX SWITCH TIMEOUT, Info 0x16c05c

adolfotregosa, 2023-01-27:
Horizon Zero Dawn 

NVRM: Xid (PCI:0000:00:0b): 31, pid=191455, name=HorizonZeroDawn, Ch 000000c9, intr 00000000. MMU Fault: ENGINE GRAPHICS GPCCLIENT_GCC faulted @ 0x0_00000000. Fault is of type FAULT_PDE ACCESS_TYPE_VIRT_READ

kellerman, 2023-02-09:
Have exactly the same errors. Playing Stray(dxvk), Forza Horizon(vkd3d), there are ocassional crashes. CSGO with dxvk doesn't crash. Sometimes this doesnt happen, sometimes does. Latest 525 driver, arch x64, linux-lqx kernel. One more user also claims to have same issue.
https://bbs.archlinux.org/viewtopic.php?id=283219

amrit1711, 2023-02-10:
There is similar issue reported on Linux Dev Forum, below is the link for reference.
https://forums.developer.nvidia.com/t/multiple-cuda-rtx-vulkan-application-crashing-with-xid-13-109-errors/235459
We have also filed a bug 3959156 internally for tracking purpose and have local repro for debugging.
Since this bug is not limited to Open GPU kernel modules, I will close this thread.

closed 2023-02-10 by amrit1711