NVIDIA/open-gpu-kernel-modules #448: 525.47.06 (vulkan dev) dkms build fails llvm 15

tisim, 2023-01-27:
### NVIDIA Open GPU Kernel Modules Version

525.47.06

### Operating System and Version

arch

### Kernel Release

6.1.7 tkg pds llvm

### Build Command

automatic dkms build command

### Terminal output/Build Log

DKMS make.log for nvidia-525.47.06 for kernel 6.1.8-273-tkg-pds-llvm (x86_64)
Fr 27. Jan 22:34:30 CET 2023
make[1]: Verzeichnis „/usr/lib/modules/6.1.8-273-tkg-pds-llvm/build“ wird betreten

Warning: Compiler version check failed:

The major and minor number of the compiler used to
compile the kernel:

clang version 15.0.7, LLD 15.0.7

does not match the compiler used here:

clang version 15.0.7
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin

It is recommended to set the CC environment variable
to the compiler that was used to compile the kernel.

To skip the test and silence this warning message, set
the IGNORE_CC_MISMATCH environment variable to "1".
However, mixing compiler versions between the kernel
and kernel modules can result in subtle bugs that are
difficult to diagnose.

*** Failed CC version check. ***

  SYMLINK /var/lib/dkms/nvidia/525.47.06/build/nvidia/nv-kernel.o
  SYMLINK /var/lib/dkms/nvidia/525.47.06/build/nvidia-modeset/nv-modeset-kernel.o
 CONFTEST: hash__remap_4k_pfn
 CONFTEST: set_pages_uc
 CONFTEST: list_is_first
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: set_pages_array_uc
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: get_num_physpages
 CONFTEST: pde_data
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: dma_attr_macros
 CONFTEST: dma_map_page_attrs
 CONFTEST: write_cr4
 CONFTEST: of_find_node_by_phandle
 CONFTEST: of_node_to_nid
 CONFTEST: pnv_pci_get_npu_dev
 CONFTEST: of_get_ibm_chip_id
[... 224 lines cut ...]
/var/lib/dkms/nvidia/525.47.06/build/nvidia/os-interface.c:1272:31: error: a function declaration without a prototype is deprecated in all versions of C [-Werror,-Wstrict-prototypes]
void NV_API_CALL os_dump_stack()
                              ^
                               void
/var/lib/dkms/nvidia/525.47.06/build/nvidia/os-interface.c:1412:11: warning: format string is not a string literal (potentially insecure) [-Wformat-security]
    panic(bugCodeStr);
          ^~~~~~~~~~
/var/lib/dkms/nvidia/525.47.06/build/nvidia/os-interface.c:1412:11: note: treat the string as an argument to avoid this
    panic(bugCodeStr);
          ^
          "%s", 
1 warning and 3 errors generated.
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/nvidia/525.47.06/build/nvidia/os-interface.o] Fehler 1
make[2]: *** Es wird auf noch nicht beendete Prozesse gewartet …
make[1]: *** [Makefile:1992: /var/lib/dkms/nvidia/525.47.06/build] Fehler 2
make[1]: Verzeichnis „/usr/lib/modules/6.1.8-273-tkg-pds-llvm/build“ wird verlassen
make: *** [Makefile:82: modules] Fehler 2

### More Info

Apparently this has already been fixed in the normal drivers.
LLVM enforces [-Werror,-Wstrict-prototypes] with version 15.

If it is according to the log, then adding the 3 function Prototypes to the code should fix it. 
Dont know if there is more to it, but appaently the normal drivers (non-vulkan-dev) had the same issue and it was fixed there with the last release.

ptr1337, 2023-02-05:
Should be fixed in the latest one :)

tisim, 2023-02-05:
worked with 525.47.07

closed 2023-02-05 by tisim