NVIDIA/open-gpu-kernel-modules #450: Black screen (monitor doesn't wake) after waking up from suspend

lufog, 2023-01-30:
### NVIDIA Open GPU Kernel Modules Version

525.85.05-2

### Does this happen with the proprietary driver (of the same version) as well?

No

### Operating System and Version

Arch Linux

### Kernel Release

6.1.8-arch1-1 #1 SMP PREEMPT_DYNAMIC Tue, 24 Jan 2023 21:07:04 +0000 x86_64 GNU/Linux

### Hardware: GPU

NVIDIA GeForce GTX 1660

### Describe the bug

After wake up from suspend, monitor doesn't wake (no signal), the system unit wakes up, the coolers make noise, the backlight on the keyboard turns on.

After wake up from suspend, the monitor does not wake up (no signal), the system unit wakes up, the coolers make noise, the keyboard backlight turns on. The indicator changes its state when the num-lock key is pressed, but only for the first 5-10 seconds, then the keyboard stops responding to pressing. The only way to get out of this state is to do a hard reset.

The problem appears on both Wayland and Xorg, but only on an open driver (nvidia-open), on a proprietary driver (nvidia) the computer wakes up without problems.

### To Reproduce

(It is enough to do the first two steps, reboot, and send the computer to sleep. But I will describe all the actions I have done.)

* Install `nvidia-open` package;
* remove `kms` from the `HOOKS` array in `/etc/mkinitcpio.conf` and regenerate the initramfs.
* add the `nvidia_drm.modeset=1` kernel parameter;
* create `/etc/modprobe.d/nvidia-power-management.conf` with content:
```
options nvidia NVreg_PreserveVideoMemoryAllocations=1 NVreg_TemporaryFilePath=/path/to/tmp-nvidia
```
* enable services `nvidia-suspend.service`, `nvidia-hibernate.service`, `nvidia-resume.service`

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10540124/nvidia-bug-report.log.gz)

### More Info

The problem and symptoms are the same as described by another user [here](https://bbs.archlinux.org/viewtopic.php?pid=2047692). Perhaps some of the information he provided will be useful.

Fxzzi, 2023-02-01:
I also experience this exact problem. RTX 3070 FE. Only occurs on open drivers too.

BurntToast01, 2023-02-02:
On a ryzen 2700x Kernel 6.1
I had the issue happen this morning after having rtcwake run. The PC woke up, but the screen remained black after returning from a suspended mode.
The amusing bit is that driver 525.85.05 patch notes made a point in addressing this issue:
"Improved the reliability of suspend and resume on UEFI systems when using certain display panels. "
https://www.nvidia.com/download/driverResults.aspx/198554/en-us/

jrgiacone, 2023-02-07:
This is the reason I have switched back to the proprietary drivers. The suspend issue is only present on the open kernel modules.

BurntToast01, 2023-02-07:
> This is the reason I have switched back to the proprietary drivers. The suspend issue is only present on the open kernel modules.

I solved this issue by installing the proprietary drivers from the Nvidia website and running the .run file in the terminal.
Installing the drivers via the systems autoinstall causes this current issue. Installing manually via the .run file provided on the Nvidia website fixed my problem. I don't have a clue why, but I am currently running driver 525.85.05 with no sleep or a black screen issue

scorpion-26, 2023-02-08:
See https://github.com/NVIDIA/open-gpu-kernel-modules/issues/223: Suspend/Resume is currently not supported on the open-source drivers. Unfortunately Nvidia didn't mention when exactly that's getting implemented.

Lucretia, 2023-02-10:
How can you turn off pm to stop it going into suspend then?

scorpion-26, 2023-02-12:
> How can you turn off pm to stop it going into suspend then?

According to nvidia's documentation(https://download.nvidia.com/XFree86/Linux-x86_64/515.65.01/README/powermanagement.html) the power management is either triggered through systemd or the kernel itself. There is probably an obscure way to prevent the gpu to go to suspend with these methods(For systemd probably removing the nvidia services and disabling some sort of ACPI device, but don't quote me on that), when the CPU and the rest of the system is sleeping. I doubt however that such workarounds will fix the suspend issues on the open source drivers. It is however probably possible to never even suspend the whole system in the first place.

Lucretia, 2023-02-12:
> According to nvidia's documentation(https://download.nvidia.com/XFree86/Linux-x86_64/515.65.01/README/powermanagement.html) the power management is either triggered through systemd or the kernel [...]

I have enabled ```NVreg_PreserveVideoMemoryAllocations=1``` in ```/etc/modprobe.d/nvidia.conf``` and also ```HandleNvidiaSleep=yes``` in ```/etc/elogind/logind.conf``` and I'm now using the kernel-open drivers in Gentoo.

amrit1711, 2023-03-27:
Hi All,
Could you please test with driver 530.41.03 and share test results.

Fxzzi, 2023-03-27:
I would test 530 however I'm experiencing other issues with that driver. I'm unable to use my main monitor on wayland, which is 1440p, at its highest refresh rate @ 170hz. It works fine on 120hz but I'm not willing to test until it's fixed. Here's an arch bugtracker link to the issue: [Arch BugTracker](https://bugs.archlinux.org/task/77975?string=nvidia&project=1&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=)

And also a Reddit post experiencing a similar issue: [reddit](https://www.reddit.com/r/archlinux/comments/1206ffa/is_nvidia53041031_borked_just_for_me_or_everyone/)

Hellzbellz123, 2023-03-28:
From the testing on my arch system, suspend still doesn't work properly on the new driver,
I'm running nvidia-open 530.41.03, and systemctl suspend still kills my displays on resume, I have Nvidia systemd services enabled

amrit1711, 2023-03-29:
@Fxzzi 
This is known issue internally and we are working on it. You can refer below thread for more updates.
https://forums.developer.nvidia.com/t/cant-boot-into-kde-wayland-session-with-530-30-02/244559/4

amrit1711, 2023-03-29:
@Hellzbellz123 
Could you please share nvidia bug report from repro state.
If by any chance system is inaccessible post suspend/resume action, please hard reboot it and then capture bug report and share with us.

Lucretia, 2023-05-01:
> From the testing on my arch system, suspend still doesn't work properly on the new driver, I'm running nvidia-open 530.41.03, and systemctl suspend leaved my displays blank on resume, with no way too access tty, I have Nvidia systemd services enabled

I'm on this, still have the settings above and seems to be fine waking up.

Fxzzi, 2023-05-27:
> @Fxzzi This is known issue internally and we are working on it. You can refer below thread for more updates. https://forums.developer.nvidia.com/t/cant-boot-into-kde-wayland-session-with-530-30-02/244559/4

bumping here because y'all are doing jack crap about this major bug

Lucretia, 2023-05-27:
@Fxzzi Are you using openrc or systemd? What about elogind?

Fxzzi, 2023-05-27:
> @Fxzzi Are you using openrc or systemd? What about elogind?

SystemD, so probably elogind also.

Lucretia, 2023-05-28:
Systemd should handle it for you as they bothered to write scripts for that, elogind had to be extended to do the same thing.

Lucretia, 2023-05-30:
Have you put logging statements into ```/usr/bin/nvidia-sleep.sh``` to see what it's doing, or if it's even called?

lufog, 2024-09-04:
I haven't used this driver for a long time, because the problem with "suspend" prevented me from using the computer normally. And I simply didn't have time to experiment/try to figure out the cause of the problem. Now I have some free time and I decided to try this driver again. And now everything works, the computer wakes up correctly!

closed 2024-09-04 by lufog