NVIDIA/open-gpu-kernel-modules #455: Brightness control not working

Maldus512, 2023-02-09:
### NVIDIA Open GPU Kernel Modules Version

525.85.05-4

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

Linux nadja 6.1.9-arch1-2 #1 SMP PREEMPT_DYNAMIC Fri, 03 Feb 2023 18:49:53 +0000 x86_64 GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA GeForce GTX 1060 (UUID: GPU-2c84c995-f03e-ef7d-6713-ffef5d0a67fd)

### Describe the bug

Hardware:

    Clevo P7xxTM1 laptop
    BIOS Version 1.07.20
    EC firmware version 1.07.12

What happens:

When I start my X11 server a device appears under `/sys/class/backlight/nvidia_0`, but the screen brightness is stuck at 100%. I can change it manually or by third party tools like xbacklight or brightnessctl and the content of `/sys/class/backlight/nvidia_0/brightness` changes accordingly, but the light doesn’t fade. I can't control the brightness whatsoever.

### To Reproduce

Just boot up the computer and launch X11.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10694669/nvidia-bug-report.log.gz)

amrit1711, 2023-03-27:
We have similar issue reported on below linux forum thread 
https://forums.developer.nvidia.com/t/brightness-on-a-lenovo-thinkpad-p52-no-longer-controllable-after-updating-nvidia-drivers-to-525-xx-series/236008/16
Since this issue is not restricted to open GPU kernel modules, I will close this thread and request you post such generic issues in future on DevTalk forum.

closed 2023-03-27 by amrit1711

err0r2k, 2024-02-01:
Same problem here but I´m running Linux Mint 21.3 (Ubuntu 22.04) with installed BIOS Version 1.07.29 and RTX2080 Mobile.