NVIDIA/open-gpu-kernel-modules #458: No Text mode Virtul Terminal / boot console on MSI NVIDIA GeForce RTX 3070 Ti GAMING X TRIO 8GB GDDR6X

Lucretia, 2023-02-10:
### NVIDIA Open GPU Kernel Modules Version

525.89.02 and now 525.47.07 (for vulkan/RT support)

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Gentoo Linux

### Kernel Release

Linux rogue 6.1.11-gentoo-x86_64 #1 SMP PREEMPT_DYNAMIC Fri Feb 10 18:15:34 GMT 2023 x86_64 AMD FX(tm)-8350 Eight-Core Processor AuthenticAMD GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3070 Ti (UUID: GPU-ce7d1b2c-cbfe-df2c-3de0-e688477e7043)

### Describe the bug

I finally managed to get a 1920x1080x32 Grub screen, no idea how as it was originally reporting only 3 modes and that wasn't one, it says is prefers a 4k res, but I don't get that from videoinfo nor does it work.

Anyway, after selecting which kernel to boot, I don't get any output from the boot process, I get no taking over of the FB_EFI module. The last thing displayed on the screen before X takes over is the booting initiramfs message from grub.

I have a W541 running the 470.161.03, and I've tried the legacy drivers, which has a working text console. I've compared the modules with the desktop and see no difference really.

### To Reproduce

1. emerge -av x11-drivers/nvidia-drivers
2. configure the kernel as per instructions,
3. genkernel (rebuilds the above kernel modules)
4. reboot.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10710694/nvidia-bug-report.log.gz)

### More Info

[dot-config.txt](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10710711/dot-config.txt)

I have also tested including the modules into the initramfs.

aaronp24, 2023-02-10:
If you're not seeing any console messages even before the NVIDIA driver loads, then this must be a kernel problem. Looking at your log, it looks like the kernel does see an EFI framebuffer but for some reason the usual messages about efifb initializing aren't there.
> [    1.253416] pci 0000:01:00.0: BAR 1: assigned to efifb

And that's it. There should be at least this message if efifb were being initialized properly:

> efifb: probing for efifb

You do have `CONFIG_FB_EFI=y` and `CONFIG_SYSFB=y` enabled, so it's not obvious why it's not getting initialized. I wonder if kgdboc is getting in the way. Can you please try removing that from the kernel boot arguments?

Lucretia, 2023-02-10:
Thought I'd give sys-kernel/gentoo-sources-5.15.93 a go with the config from above, and I get a text console/vt.

Lucretia, 2023-02-11:
> > efifb: probing for efifb
> 
> You do have `CONFIG_FB_EFI=y` and `CONFIG_SYSFB=y` enabled, so it's not obvious why it's not getting initialized. I wonder if kgdboc is getting in the way. Can you please try removing that from the kernel boot arguments?

I tried without that option and then without the crashkernel option as well and both had the same result, no console.

Lucretia, 2023-02-23:
Still the same on 6.2.0.

Lucretia, 2023-03-01:
Still the same on 6.2.1 with x11-drivers/nvidia-drivers-530.30.02.

scottcarey, 2023-04-20:
I am experiencing a very similar issue.  

Do you have an iGPU on your CPU?  In my case, there are no text consoles output on the NVidia GPU, but If I plug the monitor into my motherboard, it seems they are initialized on the iGPU and available there (in the ctrl-f keys).

Perhaps your efifb is initialized on the iGPU instead of the primary GPU?

With my previous non-Nvidia GPU I did not have this issue.      I have not yet found a solution that allows the console to print on the GPU instead of the CPU's iGPU.

Lucretia, 2023-04-30:
Same on 6.3.0.

Lucretia, 2023-04-30:
> Perhaps your efifb is initialized on the iGPU instead of the primary GPU?

I don't have an iGPU, this is a desktop with dGPU only.

Lucretia, 2023-05-02:
Same on 6.3.1.

Lucretia, 2023-05-16:
Same on 6.3.2.

[nvidia-bug-report.log-6.3.2.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11486729/nvidia-bug-report.log-6.3.2.gz)
[dot-config-6.3.2.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11486745/dot-config-6.3.2.gz)

Lucretia, 2023-05-25:
Aaand, same on 6.3.3, will try 6.3.4 later.

Lucretia, 2023-05-29:
Yup, 6.3.4 is the same non-consoleness.

Lucretia, 2023-05-29:
Is there something I'm missing here or have you just broken the console with no intention of replacing it?

traitman, 2023-06-06:
6.3.5-200.fc38.x86_64

no console, no /dev/fb* , /proc/fb is empty

can not switch vtt now

traitman, 2023-06-06:
> I am experiencing a very similar issue.
>
> Do you have an iGPU on your CPU? In my case, there are no text consoles output on the NVidia GPU, but If I plug the monitor into my motherboard, it seems [...]

yours is another issue.  you have two solutions:

1. disable the intergrated GPU in BIOS
2.  use `pci-stub.ids=xxxx:xxxx` kernel param to block the iGPU
you can get the id by `lspci -nn | grep VGA`, it is something like like `8086:4680`, so add `pci-stub.ids=8086:4680`

Lucretia, 2023-06-06:
I thought I'd take another look around and found #341 I tried the [modeset=1 fbdev=1](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/341#issuecomment-1401092032) options first on 6.3.6 and that didn't work, so I tried commenting out the [sysfb_disable()](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/341#issuecomment-1367591000) line and that did work.

mvrk69, 2023-06-09:
Hi,

I have the same issue as @scottcarey, i tried with pci-stub.ids it disables the onboard vga card, but then i have no VT consoles at all, still don't show in the NVIDIA card.

MY ENVIRONMENT:

OS: Fedora 37 (6.3.4-101.fc37.x86_64)
MB: Supermicro H12SSL-i
CPU: AMD EPYC 7313P
GPU0: ASPEED Technology, Inc. ASPEED Graphics Family [1a03:2000]
GPU1: NVIDIA GeForce GTX 1070
GPU2: MSI Radeon RX 6700 XT MECH 2X 12G (only used in KVM W10 VM / vfio-pci takes over it on boot)

Lucretia, 2023-06-10:
> Hi,
> 
> I have the same issue as @scottcarey (VT console get displayed on onboard vga card instead of NVIDIA card), i 

Didn't you see the post from @traitman? Your issue is not *this* issue, post your issue in your own thread.

Lucretia, 2023-06-11:
> good news, kernel 6.3.6 on my machine, the fb device works again.

Did you use the patch I posted above? Because without it, 6.3.6 doesn't work.

Lucretia, 2023-06-14:
@ttys3 Have I missed something? I applied

```
diff --git a/drivers/firmware/sysfb.c b/drivers/firmware/sysfb.c
index 3c197db42..16e4a2e90 100644
--- a/drivers/firmware/sysfb.c
+++ b/drivers/firmware/sysfb.c
@@ -34,6 +34,22 @@
 #include <linux/screen_info.h>
 #include <linux/sysfb.h>
 
+static int skip_simpledrm;
+
+static int __init simpledrm_disable(char *opt)
+{
+	if (!opt)
+                return -EINVAL;
+
+	get_option(&opt, &skip_simpledrm);
+
+	if (skip_simpledrm)
+		pr_info("The simpledrm driver will not be probed\n");
+
+	return 0;
+}
+early_param("nvidia-drm.modeset", simpledrm_disable);
+
 static struct platform_device *pd;
 static DEFINE_MUTEX(disable_lock);
 static bool disabled;
@@ -85,7 +101,7 @@ static __init int sysfb_init(void)
 
 	/* try to create a simple-framebuffer device */
 	compatible = sysfb_parse_mode(si, &mode);
-	if (compatible) {
+	if (compatible && !skip_simpledrm) {
 		pd = sysfb_create_simplefb(si, &mode);
 		if (!IS_ERR(pd))
 			goto unlock_mutex;
```

to my kernel and tried without any modeset specific parameters, tried `nvidia_drm.modeset=1`, `nvidia-drm.modeset=1` and also tried `fbdev=1` with those and still no console.

Is there another part?

Lucretia, 2023-06-14:
@ttys3 I checked that link, that's where I got that patch from. I applied the patch by hand and I posted the diff above, it's identical.

As I said, I tried numerous different ways including setting that option, modeset is Y by default. I've even tried enabling simpledrm again as a module with that option and nothing.

Lucretia, 2023-06-15:
Yeah, doesn't work for me with any combination of parameters.

```
$ dmesg |grep -e modes -e efifb -e simpledr
[    0.000000] Command line: BOOT_IMAGE=/@/boot/vmlinuz-6.3.7-gentoo-x86_64+ root=/dev/mapper/vg0_ssd-ws_rootfs ro rootflags=subvol=@ dolvm dobtrfs quiet amd_iommu=on iommu=pt hugepages=4608 pci=noats vfio-pci.ids=1002:67df,1002:aaf0,1b73:1100 crashkernel=@128M kgdboc=/dev/ttyS0,11520 module_blacklist=evbug,nouveau rootfstype=btrfs nvidia-drm.modeset=1 initcall_blacklist=simpledrm_platform_driver_init
[    0.000000] The simpledrm driver will not be probed
[    0.110505] Kernel command line: BOOT_IMAGE=/@/boot/vmlinuz-6.3.7-gentoo-x86_64+ root=/dev/mapper/vg0_ssd-ws_rootfs ro rootflags=subvol=@ dolvm dobtrfs quiet amd_iommu=on iommu=pt hugepages=4608 pci=noats vfio-pci.ids=1002:67df,1002:aaf0,1b73:1100 crashkernel=@128M kgdboc=/dev/ttyS0,11520 module_blacklist=evbug,nouveau rootfstype=btrfs nvidia-drm.modeset=1 initcall_blacklist=simpledrm_platform_driver_init
[    0.110699] blacklisting initcall simpledrm_platform_driver_init
[    1.275842] pci 0000:01:00.0: BAR 1: assigned to efifb
[   13.689114] nvidia-modeset: Loading NVIDIA UNIX Open Kernel Mode Setting Driver for x86_64  535.54.03  Release Build  (portage@localhost)  Thu 15 Jun 10:06:57 BST 2023

$ cat /proc/fb

$ ls -l /dev/fb
ls: cannot access '/dev/fb': No such file or directory
```

Lucretia, 2023-06-15:
I've started a parallel bug on [gentoo](https://bugs.gentoo.org/908532)

HeikintaloJukka, 2023-06-27:
I have the same bug and I noticed you too are using vfio-pci. Removing vfio-pci from loading temp fixes it for me.

mvrk69, 2023-07-09:
> I have the same bug and I noticed you too are using vfio-pci. Removing vfio-pci from loading temp fixes it for me.

Yup, i can confirm i removed the vfio from initramfs with dracut and in my case i also had to use pci-stub.ids to disable the onboard gpu and now i have boot messages and virtual terminal working on nvidia.

I wonder why vfio is causing this issue.

Lucretia, 2023-07-13:
Interesting, but annoying. Yes, this seems to be the cause, but removing that from the command line and using ```/etc/modprobe.d/vfio.conf``` with 

```
options vfio-pci ids=1002:67df,1002:aaf0,1b73:1100
```

I get this:

```
$ lspci -nnk
...
07:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Ellesmere [Radeon RX 470/480/570/570X/580/580X/590] [1002:67df] (rev e7)
        Subsystem: Sapphire Technology Limited Radeon RX 580 Pulse 4GB [1da2:e387]
07:00.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Ellesmere HDMI Audio [Radeon RX 470/480 / 570/580/590] [1002:aaf0]
        Subsystem: Sapphire Technology Limited Ellesmere HDMI Audio [Radeon RX 470/480 / 570/580/590] [1da2:aaf0]
        Kernel driver in use: snd_hda_intel
        Kernel modules: snd_hda_intel
```

So, vfio-pci isn't taking hold of the audio part of the card.

mvrk69, 2023-07-18:
> Interesting, but annoying. Yes, this seems to be the cause, but removing that from the command line and using `/etc/modprobe.d/vfio.conf` with
> 
> ```
> options vfio-pci ids=1002:67df,1002:aaf0,1b73:1100
> ```
> 
> I get this:
> 
> ```
> $ lspci -nnk
> ...
> 07:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Ellesmere [Radeon RX 470/480/570/570X/580/580X/590] [1002:67df] (rev e7)
>         Subsystem: Sapphire Technology Limited Radeon RX 580 Pulse 4GB [1da2:e387]
> 07:00.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Ellesmere HDMI Audio [Radeon RX 470/480 / 570/580/590] [1002:aaf0]
>         Subsystem: Sapphire Technology Limited Ellesmere HDMI Audio [Radeon RX 470/480 / 570/580/590] [1da2:aaf0]
>         Kernel driver in use: snd_hda_intel
>         Kernel modules: snd_hda_intel
> ```
> 
> So, vfio-pci isn't taking hold of the card at all.

Yes, but as soon as you do "modprobe vfio-pci" it will take hold of the card, but you will loose text console right away :(

aaronp24, 2023-07-20:
This sounds like the same problem as https://github.com/NVIDIA/open-gpu-kernel-modules/issues/341 because the vfio-pci driver calls the same problematic `aperture_remove_conflicting_pci_devices` function as DRM does, triggering the kernel bug where it disables the framebuffer console even on devices that don't conflict with the one being virtualized.

My changes to work around this by supporting `fbdev=1` in nvidia-drm should be in the next major driver release. Please note that the feature is still experimental so please use caution when enabling it and let me know if you run into any problems with it. I don't have an ETA for when a driver with that feature will be released.

Given the root cause of this issue, I'm closing this as a duplicate of #341.

closed 2023-07-20 by aaronp24