NVIDIA/open-gpu-kernel-modules #461: [ARM64] GPU has fallen off the bus; System crash

Wooty-B, 2023-02-18:
### NVIDIA Open GPU Kernel Modules Version

525.85.05 (with matching drivers from nvidia.com)

### Does this happen with the proprietary driver (of the same version) as well?

I cannot test this

### Operating System and Version

Debian GNU/Linux 11 (bullseye)

### Kernel Release

6.1.3+ aarch64 GNU/Linux (SolidRun patched kernel: https://github.com/SolidRun/linux-stable)

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3060

### Describe the bug

After an indeterminate amount of time (15secs - 36hours) the system slows to a halt and locks up, forcing a hard reboot. This also happens on the stock Debian kernel 5.10.0-21, as well as the latest stable kernel from kernel.org, 6.1.12

Here is a sample output of `sudo journalctl -b -1|tail -20`:

```
Feb 17 18:59:35 wootydebi-arm64 rtkit-daemon[689]: Supervising 9 threads of 5 processes of 1 users.
Feb 17 18:59:38 wootydebi-arm64 rtkit-daemon[689]: Supervising 9 threads of 5 processes of 1 users.
Feb 17 18:59:38 wootydebi-arm64 rtkit-daemon[689]: Supervising 9 threads of 5 processes of 1 users.
Feb 17 18:59:43 wootydebi-arm64 rtkit-daemon[689]: Supervising 9 threads of 5 processes of 1 users.
Feb 17 18:59:43 wootydebi-arm64 rtkit-daemon[689]: Supervising 9 threads of 5 processes of 1 users.
Feb 17 19:00:16 wootydebi-arm64 kernel: NVRM: GPU at PCI:0004:01:00: GPU-d8742144-60aa-ba60-e216-233309878ed8
Feb 17 19:00:17 wootydebi-arm64 kernel: NVRM: Xid (PCI:0004:01:00): 79, pid='<unknown>', name=<unknown>, GPU has fallen off the bus.
Feb 17 19:00:17 wootydebi-arm64 kernel: NVRM: GPU 0004:01:00.0: GPU has fallen off the bus.
Feb 17 19:00:17 wootydebi-arm64 kernel: NVRM prbEncStartAlloc: Can't allocate memory for protocol buffers.
Feb 17 19:00:17 wootydebi-arm64 kernel: NVRM: A GPU crash dump has been created. If possible, please run
                                        NVRM: nvidia-bug-report.sh as root to collect this data before
                                        NVRM: the NVIDIA kernel module is unloaded.
Feb 17 19:00:17 wootydebi-arm64 kernel: NVRM nvAssertOkFailedNoLog: Assertion failed: GPU lost from the bus [NV_ERR_GPU_IS_LOST] (0x0000000F) returned from intrGetPendingStall_GM107(pGpu, pIntr, pEngines, pThreadState) @ intr_gp100.c:193
Feb 17 19:00:17 wootydebi-arm64 kernel: NVRM intrServiceStall_IMPL: Failed GPU reg read : 0xffffffff. Check whether GPU is present on the bus
Feb 17 19:00:22 wootydebi-arm64 kernel: NVRM: Xid (PCI:0004:01:00): 119, pid=430, name=nv_queue, Timeout waiting for RPC from GSP! Expected function 78 (DUMP_PROTOBUF_COMPONENT) (0x0 0x0).
Feb 17 19:00:22 wootydebi-arm64 kernel: NVRM _kgspLogXid119: RPC history (CPU -> GSP):
Feb 17 19:00:22 wootydebi-arm64 kernel: NVRM _kgspLogXid119:         entry        func                                data
Feb 17 19:00:22 wootydebi-arm64 kernel: NVRM _kgspLogXid119:          0         78 DUMP_PROTOBUF_COMPONENT        0x00000000 0x00000000
Feb 17 19:00:22 wootydebi-arm64 kernel: NVRM _kgspLogXid119:         -1         76 GSP_RM_CONTROL                0x00730109 0x0000000c
Feb 17 19:00:22 wootydebi-arm64 kernel: NVRM _kgspLogXid119:         -2         76 GSP_RM_CONTROL                0x00730109 0x0000000c

```
Attached is the full dmesg log.
[nvidia_aarch64_lx2k_525.85.05.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10772724/nvidia_aarch64_lx2k_525.85.05.log)

### To Reproduce

After the system boots, the computer will rapidly slow to a crawl until it hangs. This could be before the login manager appears, after sitting idle for a day or two, or after moving around several applications (regardless of performance requirements). 

Playing games or switching through many browser tabs appears to exacerbate the issue, however appears to correlate more to constant screen draw than being graphically intensive as I can play 15min of Doom 3 without a crash whereas scrolling an article for 5min renders my PC unusable.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10772753/nvidia-bug-report.log.gz)

### More Info

This is an aarch64 Solidrun HoneyComb @ 2.2Ghz w/ 8x PCIe 3.0, 650W PSU. Looking for options to play with to try and narrow down the issue.

I have ran the following as well: `nvidia-smi -pm 1`

Current Linux boot flags: idle=nomwait pcie_aspm=off

Redsandro, 2023-05-12:
I came here from Google with a similar issue, but it only happens occasionally. [Someone says](https://forums.developer.nvidia.com/t/gpu-has-fallen-off-the-bus/71058/7?u=tim.evi) the issue only happens after a cold boot.

> I have discovered something interesting. The issue never reappears after a soft reboot.
>
> So I do a Sync, Unmount and reBoot SysRequest (i.e. hold Alt + SysRq while pressing S, U, and B in slow succession) and the problem is gone until I do a cold boot.

amrit1711, 2023-06-05:
@Wooty-B 
You’re getting a XID 79, fallen off the bus. Most common reasons for this issue are overheating or lack of power or outdated SBIOS.
Kindly monitor temperatures, reseat power connectors/the card in its slot, check/replace PSU and update to latest SBIOS.
To check for power issues, you can use nvidia-smi -lgc to prevent boost situations, e.g.
nvidia-smi -lgc 300,1500

Since this issue is not restricted to open gpu kernel modules, I will close this thread and request you to post new issues on below forum or email to us at "linux-bugs@nvidia.com" for generic issues.
https://forums.developer.nvidia.com/c/gpu-graphics/145

closed 2023-06-05 by amrit1711

DanaGoyette, 2023-08-04:
Actually, this board IS restricted to the open kernel modules, because the old kernel module refuses to allow loading on this board, because it's not one of the approved SOCs.

I've been having an RTX2060 Super fall off the bus in my Honeycomb, whereas the same GPU is fine in any x86 machine, with either the open or the previous closed kernel module.  And the GPU works fine in a Thunderbolt enclosure on an x86 machine.