NVIDIA/open-gpu-kernel-modules #467: Pipewire screen sharing under Wayland results in black canvas with cursors

kotatsuyaki, 2023-02-26:
### NVIDIA Open GPU Kernel Modules Version

525.89.02

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

NixOS (Nixpkgs commit `806075b`)

### Kernel Release

6.1.13

### Hardware: GPU

NVIDIA GeForce RTX 3070

### Describe the bug

When using Pipewire screen sharing on Chromium or Firefox on Wayland, the shared stream is black, with the cursor visible and moving. Here's a screenshot of it.

[image]

This bug [was reported to Pipewire](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/2242#note_1793891). All affected users are using the proprietary Nvidia driver, and the bug is reproducible with the open kernel module.

Note that while on Chromium and Firefox the results were black, I was able to record the screen with OBS Studio, so it's unclear whether this bug belongs to here.

### To Reproduce

1. Open Chromium or Firefox.
2. Open any website that uses WebRTC screen sharing (tested on https://www.webrtc-experiment.com) and share the screen.
3. From the xdg-desktop-portal popup dialog, select the entire screen.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

Due to Nixpkgs missing the `nvidia-bug-report.sh` script (NixOS/nixpkgs#172739), this part is left out.

otacke, 2023-04-01:
Not sure if this helps, but I am in the same situation: black screen and only mouse pointer visible when sharing the screen on Wayland.

```
Kernel: 6.2.6-1-default x86_64

VGA compatible controller [0300]: NVIDIA Corporation TU117 [GeForce GTX 1650] [10de:1f82] (rev a1)
	Subsystem: NVIDIA Corporation TU117 [GeForce GTX 1650] [10de:1f82]
	Kernel driver in use: nouveau
```

otacke, 2023-04-08:
I fixed the issue for me.
```
VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Navi 22 [Radeon RX 6700/6700 XT/6750 XT / 6800M/6850M XT] (rev c5)
```

ayoubedd, 2023-04-13:
> I fixed the issue for me.
> 
> ```
> VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Navi 22 [Radeon RX 6700/6700 XT/6750 XT / 6800M/6850M XT] (rev c5)
> ```

How ?

otacke, 2023-04-13:
@ayoubedd I replaced the NVIDIA card with an ATI board ;-) I've had other trouble with NVIDIA before, so I thought it might be time for a change eventually.

Cylon-interrogator, 2023-04-13:
I absolutely agree, Nvidia just sucks on Linux. However, I've got ton of NVidia GPUs lying around left over from 2021 crypto craze waiting for the next bull run, and I've put one of the to my secondary rig in my second home ... At least, now when they've opened the drivers, there is hope for the future ...

ar-nadeem, 2023-04-16:
Leaving Nvidia is not a choice for me. AI, Cuda, and other software implementations of Nvidia are of great use to me. Nvidia wants nothing to do with Linux. The bare minimum drivers with billion of bugs are all they do. This might change because of Steam Deck, why wouldn't Nvidia want their GPU in the steam deck instead of AMDs? But for now, I shifted back to Windows.

nikzasel, 2023-05-11:
Problem still occurs on 530.41.03-1

I have `KDE 5.27.5-1` and `kwin 5.27.5-2` and have partially (can share windows, but not whole screen) made screen sharing working by:
1. Install `xwaylandvideobridge`. You need it to stream wayland windows to x11 apps (like to screenshare in discord or telegram, you dont need it for wayland window to wayland app streaming)
2. Add following string to `~/.profile` file to use glx instead of egl backend.
```
#O2: OpenGL 2 backend
#O2ES: OpenGL ES 2 backend (since 5.5)
**export KWIN_COMPOSE=O2**
```
3. (optional) Remove if set `--use-gl=egl` flag from browser launch flags to fix GPU acceleration.

I think you can fix it for other window manager if you somehow make make them not work with egl

I think recent NVIDIA driver have somehow broke OpenGL ES because if i fetch info about it via `eglinfo -B` i receive:

```
GBM platform:
EGL API version: 1.5
EGL vendor string: NVIDIA
EGL version string: 1.5
EGL client APIs: OpenGL_ES OpenGL
corrupted size vs. prev_size
fish: Job 1, 'eglinfo -B' terminated by signal SIGABRT (Abort)
```
and following entry in journal:

```
Stack trace of thread 65909:
#0  0x00007f2eef60526c n/a (libc.so.6 + 0x8926c)
#1  0x00007f2eef5b5a08 raise (libc.so.6 + 0x39a08)
#2  0x00007f2eef59e538 abort (libc.so.6 + 0x22538)
#3  0x00007f2eef59f2db n/a (libc.so.6 + 0x232db)
#4  0x00007f2eef60f1b7 n/a (libc.so.6 + 0x931b7)
#5  0x00007f2eef60fcce n/a (libc.so.6 + 0x93cce)
#6  0x00007f2eef60fe40 n/a (libc.so.6 + 0x93e40)
#7  0x00007f2eef6124b8 n/a (libc.so.6 + 0x964b8)
#8  0x00007f2eef61382a malloc (libc.so.6 + 0x9782a)
#9  0x00007f2eed696155 n/a (libnvidia-eglcore.so.530.41.03 + 0x1496155)
#10 0x00007f2eed689bfd n/a (libnvidia-eglcore.so.530.41.03 + 0x1489bfd)
#11 0x00007f2eed68a91c n/a (libnvidia-eglcore.so.530.41.03 + 0x148a91c)
#12 0x00007f2eed68aa3c n/a (libnvidia-eglcore.so.530.41.03 + 0x148aa3c)
#13 0x00007f2eed6a457b n/a (libnvidia-eglcore.so.530.41.03 + 0x14a457b)
#14 0x00007f2eed6a4700 n/a (libnvidia-eglcore.so.530.41.03 + 0x14a4700)
#15 0x00007f2eef042d12 n/a (libEGL_nvidia.so.0 + 0x42d12)
#16 0x00007f2eef048344 n/a (libEGL_nvidia.so.0 + 0x48344)
#17 0x00005604a1407823 n/a (eglinfo + 0x8823)
#18 0x00005604a140b3a5 n/a (eglinfo + 0xc3a5)
#19 0x00005604a140326e n/a (eglinfo + 0x426e)
#20 0x00007f2eef59f850 n/a (libc.so.6 + 0x23850)
#21 0x00007f2eef59f90a __libc_start_main (libc.so.6 + 0x2390a)
#22 0x00005604a14036e5 n/a (eglinfo + 0x46e5)
ELF object binary architecture: AMD x86-64
```

But for `glxinfo -B` i receive:

```
name of display: :1
display: :1  screen: 0
direct rendering: Yes
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 24564 MB
    Total available memory: 24564 MB
    Currently available dedicated video memory: 23235 MB
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: NVIDIA GeForce RTX 4090/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 530.41.03
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.6.0 NVIDIA 530.41.03
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)

OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 530.41.03
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

TheM1Stery, 2023-05-13:
Idk how but it works now for me. I can screen share the whole screen through browser. I tried with xwayland video bridge but it didn't work there(old version of pipewire in the flatpak maybe?).
KDE 5.27.5 
Nvidia version:
[image]
pipewire version: 0.3.69

Screen sharing whole screen through Firefox on Discord Web:
[image]

TheM1Stery, 2023-05-13:
Idk how but it works now for me. I can screen share the whole screen through browser. I tried with xwayland video bridge but it didn't work there(old version of pipewire in the flatpak maybe?).
KDE 5.27.5
Nvidia version: 525.116.04
pipewire version: 0.3.69

Screen sharing whole screen through Firefox on Discord Web:
[image]

I can share individual windows as well

dannkunt, 2023-05-28:
> I have KDE 5.27.5-1 and kwin 5.27.5-2 and have partially (can share windows, but not whole screen) made screen sharing working by:

>    Install xwaylandvideobridge. You need it to stream wayland windows to x11 apps (like to screenshare in discord or telegram, you dont need it for wayland window to wayland app streaming)
    Add following string to ~/.profile file to use glx instead of egl backend.

>#O2: OpenGL 2 backend
#O2ES: OpenGL ES 2 backend (since 5.5)
export KWIN_COMPOSE=O2

   > (optional) Remove if set --use-gl=egl flag from browser launch flags to fix GPU acceleration.

> I think you can fix it for other window manager if you somehow make make them not work with egl

No changes in telegram screen sharing on KDE Plasma 5.27.4, RTX 3060. But I experience same bug with eglinfo and glxinfo commands.

ar-nadeem, 2023-06-18:
Soo still not fixed? Are we sure the problem is with the drivers not playing nice with Wayland screen-sharing policies?

rwiggins, 2023-07-06:
The problem in https://github.com/NVIDIA/open-gpu-kernel-modules/issues/467#issuecomment-1544340511 still occurs for me with version 535.54.03.

```
 ~ % eglinfo -B
GBM platform:
EGL API version: 1.5
EGL vendor string: NVIDIA
EGL version string: 1.5
EGL client APIs: OpenGL_ES OpenGL
corrupted size vs. prev_size
[1]    178090 IOT instruction (core dumped)  eglinfo -B
 ~ % glxinfo -B
name of display: :1
display: :1  screen: 0
direct rendering: Yes
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 24564 MB
    Total available memory: 24564 MB
    Currently available dedicated video memory: 20202 MB
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: NVIDIA GeForce RTX 4090/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 535.54.03
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.6.0 NVIDIA 535.54.03
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)

OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 535.54.03
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
 ~ % uname -a
Linux kryptos 6.4.1-arch1-1 #1 SMP PREEMPT_DYNAMIC Sat, 01 Jul 2023 16:17:21 +0000 x86_64 GNU/Linux
```

I can share the core dump with NVIDIA if it's helpful.

```
Core was generated by `eglinfo -B'.
Program terminated with signal SIGABRT, Aborted.
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
Downloading source file /usr/src/debug/glibc/glibc/nptl/pthread_kill.c
44            return INTERNAL_SYSCALL_ERROR_P (ret) ? INTERNAL_SYSCALL_ERRNO (ret) : 0;                                                        
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007fc2495042d3 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007fc2494b4a08 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007fc24949d538 in __GI_abort () at abort.c:79
#4  0x00007fc24949e2db in __libc_message (fmt=fmt@entry=0x7fc249618515 "%s\n") at ../sysdeps/posix/libc_fatal.c:150
#5  0x00007fc24950e1b7 in malloc_printerr (str=str@entry=0x7fc249615f79 "corrupted size vs. prev_size") at malloc.c:5651
#6  0x00007fc24950ecce in unlink_chunk (p=<optimized out>, av=<optimized out>) at malloc.c:1605
#7  0x00007fc24950ee40 in malloc_consolidate (av=av@entry=0x7fc249656ba0 <main_arena>) at malloc.c:4766
#8  0x00007fc2495114b8 in _int_malloc (av=av@entry=0x7fc249656ba0 <main_arena>, bytes=1472) at malloc.c:3951
#9  0x00007fc24951282a in __GI___libc_malloc (bytes=<optimized out>) at malloc.c:3297
#10 0x00007fc247530825 in ?? () from /usr/lib/libnvidia-eglcore.so.535.54.03
#11 0x00007fc2475241ed in ?? () from /usr/lib/libnvidia-eglcore.so.535.54.03
#12 0x00007fc247524f3c in ?? () from /usr/lib/libnvidia-eglcore.so.535.54.03
#13 0x00007fc24752505c in ?? () from /usr/lib/libnvidia-eglcore.so.535.54.03
#14 0x00007fc24753eeeb in ?? () from /usr/lib/libnvidia-eglcore.so.535.54.03
#15 0x00007fc24753f070 in ?? () from /usr/lib/libnvidia-eglcore.so.535.54.03
#16 0x00007fc249042f72 in ?? () from /usr/lib/libEGL_nvidia.so.0
#17 0x00007fc2490485a4 in ?? () from /usr/lib/libEGL_nvidia.so.0
#18 0x0000558fabdaf823 in createEGLContext (d=d@entry=0x558fad65ab00, conf=conf@entry=0xcaf32c, api=api@entry=12450, 
    khr_create_context=khr_create_context@entry=1, core_profile=core_profile@entry=1, context_version=context_version@entry=0x7ffdb30e24ac)
    at ../mesa-demos-9.0.0/src/egl/opengl/eglinfo.c:436
#19 0x0000558fabdb33a5 in doOneDisplay (d=0x558fad65ab00, name=<optimized out>, opts=...) at ../mesa-demos-9.0.0/src/egl/opengl/eglinfo.c:575
#20 0x0000558fabdab26e in main (argc=<optimized out>, argv=<optimized out>) at ../mesa-demos-9.0.0/src/egl/opengl/eglinfo.c:850
(gdb) i threads
  Id   Target Id                          Frame 
* 1    Thread 0x7fc2493c0740 (LWP 178090) __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0)
    at pthread_kill.c:44
```
I suspect this may be the cause of a crash I'm getting when launching a different program, but it's hard for me to say for sure.

ar-nadeem, 2023-07-25:
Is this Fixed ? This is the only thing stopping me from shifting to Linux from Windows.

GrabbenD, 2023-07-25:
I fixed all of my Wayland- and hardware acceleration issues by switching my GPU to AMD @ar-nadeem 
Don't bother using hardware which gets minimal support on Linux

Cylon-interrogator, 2023-07-25:
That's just like when you ask how to fix an issue with a game under Linux and someone replies: "Dude, just use Windows!" ... Why haven't we thought of that, I wonder ... Anyhow, I don't think that's very helpful answer. Yes, everybody knows that Nvidia (still!) sucks under Linux, although it has improved a lot during last year (as corroborated by various Linux users). Some people still prefer or need to use Nvidia for various reasons - better ray tracing support, CUDA, DLSS ... I personally use it on my secondary rig (my primary PC has RX 6700XT) because, I've got a lot of Nvidia GPUs lying around from 2021 crypto bull run ...

Cylon-interrogator, 2023-07-25:
> I fixed all of my Wayland- and hardware acceleration issues by switching my GPU to AMD @ar-nadeem Don't bother using hardware which gets minimal support on Linux

That's just like when you ask how to fix an issue with a game under Linux and someone replies: "Dude, just use Windows!" ... Why haven't we thought of that, I wonder ... Anyhow, I don't think that's very helpful answer. Yes, everybody knows that Nvidia (still!) sucks under Linux, although it has improved a lot during last year (as corroborated by various Linux users). Some people still prefer or need to use Nvidia for various reasons - better ray tracing support, CUDA, DLSS ... I personally use it on my secondary rig (my primary PC has RX 6700XT) because, I've got a lot of Nvidia GPUs lying around from 2021 crypto bull run ...

GrabbenD, 2023-07-25:
> Dude, just use Windows

Not a fair comparison..
We can't fix issues in Windows
We can't fix issues with hardware behind closed binary blobs

> everybody knows that Nvidia (still!)
> although it has improved a lot during last year (as corroborated by various Linux users).

They have sucked for the past decade.
What's your point?

Don't support evil companies
**Vote with your wallet**

regalialong, 2023-07-25:
@GrabbenD 
> Yes, they have sucked for the past decade We shouldn't expect too much from a ~multi billion dollar comapny~ [sic] small company like NVDIA though

Do you know how multi billion dollar companies stay billion dollar companies? They don't throw money into things that they think won't bring them money back.

Reality is that the crowd running Nvidia on Linux, followed by the crowd that runs Wayland is pretty small (although steadily growing). I also dislike that they don't invest more attention into Linux but that's just company priority for you.

> I fixed all of my Wayland- and hardware acceleration issues by switching my GPU to AMD

Comments like these are just unhelpful for everybody involved, it's off-topic (this comment too). 
Everybody here knows driver support is better on AMD (or should at least), some people can't buy another GPU (like being too poor for another GPU) or it doesn't make sense for them to switch (like people who have AI workloads).

Ideally using Wayland should be device agnostic.

But Cylon already told you this and you didn't address their strong arguments so I'll default to you not being here for a good faith discussion. Please just don't comment this, it clogs up the issue which might make a dev less interested working on this.

ar-nadeem, 2023-07-30:
Hey, good news. I think it is fixed. 
[image]

Driver: 535.86.05
GPU: RTX 3070
01:00.0 VGA compatible controller: NVIDIA Corporation GA104 [GeForce RTX 3070] (rev a1)
        Subsystem: Palit Microsystems Inc. GA104 [GeForce RTX 3070]
        Kernel driver in use: nvidia
        Kernel modules: nouveau, nvidia_drm, nvidia

glaeqen, 2023-07-30:
With `wlroots` bad luck still:

`/usr/lib/xdg-desktop-portal-wlr -l TRACE -r` && trying to start the screen sharing both in Chromium and Firefox.
```
2023/07/30 14:58:53 [INFO] - dbus: start method invoked
2023/07/30 14:58:53 [INFO] - dbus: request_handle: /org/freedesktop/portal/desktop/request/1_24/webrtc2043827389
2023/07/30 14:58:53 [INFO] - dbus: session_handle: /org/freedesktop/portal/desktop/session/1_24/webrtc_session1289938188
2023/07/30 14:58:53 [INFO] - dbus: app_id:
2023/07/30 14:58:53 [INFO] - dbus: parent_window:
2023/07/30 14:58:53 [DEBUG] - dbus: start: found matching session /org/freedesktop/portal/desktop/session/1_24/webrtc_session1289938188
2023/07/30 14:58:53 [TRACE] - wlroots: callbacks registered
2023/07/30 14:58:53 [TRACE] - wlroots: buffer event handler
2023/07/30 14:58:53 [ERROR] - xdg-desktop-portal-wlr: unsupported wl_shm format 0x34324742
fish: Job 1, '/usr/lib/xdg-desktop-portal-wlr…' terminated by signal SIGABRT (Abort)
```

ar-nadeem, 2023-07-30:
I am using xdg-desktop-portal-kde
Let me know if you would like any more information on logs.

potatogod3000, 2023-07-30:
WebRTC is working fine in Firefox (both Window and Screen Sharing). I get the black screen with just a cursor in both screen and window capture in Telegram Wayland screen sharing though. I have attached the image below. (@ar-nadeem)
My Specs:
Fedora 38 (Up-to-date)
KDE Plasma 5.27.6 (with both xdg-desktop-portal-kde and xdg-desktop-portal-gtk installed)
NVIDIA Proprietary Driver version 535.86.05
All non-free codecs from RPMFusion
WebRTC in Firefox:
[image]
Telegram Desktop (rpm package) screensharing not working:
[image]

ar-nadeem, 2023-08-06:
I yet have to test with other applications. But this also now works on gnome.
You have to install pipewire-media-session and all other pipewire related stuff. Then reboot and it will work.

glaeqen, 2023-08-07:
Small nitpick

> You have to install pipewire-media-session and all other pipewire related stuff. Then reboot and it will work.

`pipewire-media-session` is not supposed to be used, it was just a demo session manager for which `wireplumber` is a fully featured replacement. FYI

erik-kz, 2023-10-18:
For sway, naively adding support for DRM_FORMAT_BGR888 to xdg-desktop-portal-wlr appears to work
```
diff --git a/src/screencast/screencast_common.c b/src/screencast/screencast_common.c
index d6d13db..69bd83c 100644
--- a/src/screencast/screencast_common.c
+++ b/src/screencast/screencast_common.c
@@ -264,6 +264,7 @@ enum wl_shm_format xdpw_format_wl_shm_from_drm_fourcc(uint32_t format) {
                return WL_SHM_FORMAT_ARGB8888;
        case DRM_FORMAT_XRGB8888:
                return WL_SHM_FORMAT_XRGB8888;
+       case DRM_FORMAT_BGR888:
        case DRM_FORMAT_RGBA8888:
        case DRM_FORMAT_RGBX8888:
        case DRM_FORMAT_ABGR8888:
@@ -293,6 +294,7 @@ uint32_t xdpw_format_drm_fourcc_from_wl_shm(enum wl_shm_format format) {
                return DRM_FORMAT_ARGB8888;
        case WL_SHM_FORMAT_XRGB8888:
                return DRM_FORMAT_XRGB8888;
+       case WL_SHM_FORMAT_BGR888:
        case WL_SHM_FORMAT_RGBA8888:
        case WL_SHM_FORMAT_RGBX8888:
        case WL_SHM_FORMAT_ABGR8888:
@@ -318,6 +320,8 @@ uint32_t xdpw_format_drm_fourcc_from_wl_shm(enum wl_shm_format format) {
 
 enum spa_video_format xdpw_format_pw_from_drm_fourcc(uint32_t format) {
        switch (format) {
+       case DRM_FORMAT_BGR888:
+               return SPA_VIDEO_FORMAT_BGR;
        case DRM_FORMAT_ARGB8888:
                return SPA_VIDEO_FORMAT_BGRA;
        case DRM_FORMAT_XRGB8888:
```

erik-kz, 2023-10-19:
Ha, it looks like someone made such a change just yesterday https://github.com/emersion/xdg-desktop-portal-wlr/commit/84a9cf40fb6e5b2900c7d245a1ac8acad828a88f

So screencast with sway/wlroots ought to be working with the latest xdg-desktop-portal-wlr