NVIDIA/open-gpu-kernel-modules #468: Failure to build kernel modules after 530.30.02

apprehensions, 2023-03-01:
### NVIDIA Open GPU Kernel Modules Version

530.30.02

### Operating System and Version

KISS

### Kernel Release

6.1.13

### Build Command

```
sh NVIDIA-Linux-x86_64-530.30.02-no-compat32.run --extract-only
cd kernel
make modules
```

### Terminal output/Build Log

```
Creating directory NVIDIA-Linux-x86_64-530.30.02-no-compat32
Verifying archive integrity... OK
Uncompressing NVIDIA Accelerated Graphics Driver for Linux-x86_64 530.30.02..............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 12.2.0
  You are using:           cc (GCC) 12.2.0
 CONFTEST: hash__remap_4k_pfn
 CONFTEST: set_pages_uc
 CONFTEST: list_is_first
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: set_pages_array_uc
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: get_num_physpages
 CONFTEST: pde_data
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: dma_attr_macros
 CONFTEST: dma_map_page_attrs
 CONFTEST: write_cr4
 CONFTEST: of_find_node_by_phandle
 CONFTEST: of_node_to_nid
 CONFTEST: pnv_pci_get_npu_dev
 CONFTEST: of_get_ibm_chip_id
 CONFTEST: pci_bus_address
 CONFTEST: pci_stop_and_remove_bus_device
 CONFTEST: pci_rebar_get_possible_sizes
 CONFTEST: register_cpu_notifier
 CONFTEST: cpuhp_setup_state
 CONFTEST: dma_map_resource
 CONFTEST: get_backlight_device_by_name
 CONFTEST: timer_setup
 CONFTEST: pci_enable_msix_range
 CONFTEST: kernel_read_has_pointer_pos_arg
 CONFTEST: kernel_write_has_pointer_pos_arg
 CONFTEST: dma_direct_map_resource
 CONFTEST: tegra_get_platform
 CONFTEST: tegra_bpmp_send_receive
 CONFTEST: flush_cache_all
 CONFTEST: vmf_insert_pfn
 CONFTEST: jiffies_to_timespec
 CONFTEST: ktime_get_raw_ts64
 CONFTEST: ktime_get_real_ts64
 CONFTEST: full_name_hash
[... 456 lines cut ...]
In file included from /tmp/23233/build/nvidia/NVIDIA-Linux-x86_64-530.30.02-no-compat32/kernel/common/inc/conftest.h:28,
                 from /tmp/23233/build/nvidia/NVIDIA-Linux-x86_64-530.30.02-no-compat32/kernel/common/inc/nv_stdarg.h:29,
                 from /tmp/23233/build/nvidia/NVIDIA-Linux-x86_64-530.30.02-no-compat32/kernel/common/inc/os-interface.h:40,
                 from /tmp/23233/build/nvidia/NVIDIA-Linux-x86_64-530.30.02-no-compat32/kernel/nvidia/nv-i2c.c:28:
/tmp/23233/build/nvidia/NVIDIA-Linux-x86_64-530.30.02-no-compat32/kernel/conftest/functions.h:71:2: error: #error dma_buf_export() conftest failed!
   71 | #error dma_buf_export() conftest failed!
      |  ^~~~~
/tmp/23233/build/nvidia/NVIDIA-Linux-x86_64-530.30.02-no-compat32/kernel/conftest/functions.h:83:2: error: #error wait_on_bit_lock() conftest failed!
   83 | #error wait_on_bit_lock() conftest failed!
      |  ^~~~~
/tmp/23233/build/nvidia/NVIDIA-Linux-x86_64-530.30.02-no-compat32/kernel/conftest/functions.h:86:2: error: #error radix_tree_replace_slot() conftest failed!
   86 | #error radix_tree_replace_slot() conftest failed!
      |  ^~~~~
/tmp/23233/build/nvidia/NVIDIA-Linux-x86_64-530.30.02-no-compat32/kernel/common/inc/nv_stdarg.h:33:14: fatal error: stdarg.h: No such file or directory
   33 |     #include <stdarg.h>
      |              ^~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:250: /tmp/23233/build/nvidia/NVIDIA-Linux-x86_64-530.30.02-no-compat32/kernel/nvidia/nv-i2c.o] Error 1
make[1]: *** [Makefile:2005: /tmp/23233/build/nvidia/NVIDIA-Linux-x86_64-530.30.02-no-compat32/kernel] Error 2
make: *** [Makefile:82: modules] Error 2
```

### More Info

this only occured after updating to 530.30.02 from  525.89.02

aritger, 2023-03-01:
Thanks for the report.  That is curious. Just to double check: building the 525.89.02 kernel modules with the same 6.1.13 target kernel works?

530.xx changed a few things with how conftest handles CFLAGS.  That's the only thing I can think of that would cause this.

Could I trouble you to perform the following experiment?

(1) Make the following edit to conftest.sh:

```
diff --git a/kernel-open/conftest.sh b/kernel-open/conftest.sh
index b6a152af6986..66f5fc502f94 100755
--- a/kernel-open/conftest.sh
+++ b/kernel-open/conftest.sh
@@ -5937,6 +5937,9 @@ case "$5" in
     ;;

     compile_tests)
+
+        echo "$*" > conftest.log.$$
+
         #
         # Run a series of compile tests to determine the set of interfaces
         # and features available in the target kernel.
```

(that should apply to kernel-open/conftest.sh as well as kernel/conftest.sh)

(2) Do `make modules` as in your repro steps.

(3) Grep the conftest.log.* files for "dma_buf_export".  Paste the contents of the dma_buf_export log file here.

(4) Repeat steps 1-3 with 525.89.02.

I'm curious to see how the CFLAGS in the log files differ between 530.30.02 and 525.89.02.

Thanks.

apprehensions, 2023-03-02:
530.30.02:
```
cc x86_64 /lib/modules/6.1.13teapot/build /lib/modules/6.1.13teapot/build compile_tests -O2 -D__KERNEL__ -DKBUILD_BASENAME="#conftest8170" -DKBUILD_MODNAME="#conftest8170" -nostdinc -isystem /usr/lib/gcc/x86_64-pc-linux-gnu/12.2.0/include -Wno-implicit-function-declaration -Wno-strict-prototypes -I/lib/modules/6.1.13teapot/build/include/asm-x86/mach-default -I/lib/modules/6.1.13teapot/build/arch/x86/include/asm/mach-default -include /lib/modules/6.1.13teapot/build/include/generated/autoconf.h -I/lib/modules/6.1.13teapot/build/include -I/lib/modules/6.1.13teapot/build/include/uapi -I/lib/modules/6.1.13teapot/build/include/xen -I/lib/modules/6.1.13teapot/build/include/generated/uapi -I/lib/modules/6.1.13teapot/build/arch/x86/include -I/lib/modules/6.1.13teapot/build/arch/x86/include/uapi -I/lib/modules/6.1.13teapot/build/arch/x86/include/generated -I/lib/modules/6.1.13teapot/build/arch/x86/include/generated/uapi -mfentry -DCC_USING_FENTRY -I/tmp/7668/build/nvidia/NVIDIA-Linux-x86_64-530.30.02-no-compat32/kernel/common/inc -I/tmp/7668/build/nvidia/NVIDIA-Linux-x86_64-530.30.02-no-compat32/kernel -Wall   -Wno-cast-qual -Wno-error -Wno-format-extra-args -D__KERNEL__ -DMODULE -DNVRM -DNV_VERSION_STRING=\"530.30.02\" -Wno-unused-function -Wuninitialized -fno-strict-aliasing -ffreestanding -mno-red-zone -mcmodel=kernel -DNV_UVM_ENABLE -Werror=undef -DNV_SPECTRE_V2=0 -DNV_KERNEL_INTERFACE_LAYER -fno-pie dma_buf_export_args
```
525.89.02:
```
cc x86_64 /lib/modules/6.1.13teapot/build /lib/modules/6.1.13teapot/build compile_tests -O2 -D__KERNEL__ -DKBUILD_BASENAME="#conftest12651" -DKBUILD_MODNAME="#conftest12651" -nostdinc -isystem /usr/lib/gcc/x86_64-pc-linux-gnu/12.2.0/include -Wno-implicit-function-declaration -Wno-strict-prototypes -I/lib/modules/6.1.13teapot/build/include/asm-x86/mach-default -I/lib/modules/6.1.13teapot/build/arch/x86/include/asm/mach-default -include /lib/modules/6.1.13teapot/build/include/generated/autoconf.h -I/lib/modules/6.1.13teapot/build/include -I/lib/modules/6.1.13teapot/build/include/uapi -I/lib/modules/6.1.13teapot/build/include/xen -I/lib/modules/6.1.13teapot/build/include/generated/uapi -I/lib/modules/6.1.13teapot/build/arch/x86/include -I/lib/modules/6.1.13teapot/build/arch/x86/include/uapi -I/lib/modules/6.1.13teapot/build/arch/x86/include/generated -I/lib/modules/6.1.13teapot/build/arch/x86/include/generated/uapi -mfentry -DCC_USING_FENTRY -I/tmp/11993/build/nvidia/NVIDIA-Linux-x86_64-525.89.02-no-compat32/kernel/common/inc -I/tmp/11993/build/nvidia/NVIDIA-Linux-x86_64-525.89.02-no-compat32/kernel -Wall -MD   -Wno-cast-qual -Wno-error -Wno-format-extra-args -D__KERNEL__ -DMODULE -DNVRM -DNV_VERSION_STRING=\"525.89.02\" -Wno-unused-function -Wuninitialized -fno-strict-aliasing -ffreestanding -mno-red-zone -mcmodel=kernel -DNV_UVM_ENABLE -Werror=undef -DNV_SPECTRE_V2=0 -DNV_KERNEL_INTERFACE_LAYER -fno-pie dma_buf_export_args
```

there is NO difference:
```
--- 530
+++ 525
@@ -5,8 +5,8 @@
 compile_tests
 -O2
 -D__KERNEL__
--DKBUILD_BASENAME="#conftest8170"
--DKBUILD_MODNAME="#conftest8170"
+-DKBUILD_BASENAME="#conftest12651"
+-DKBUILD_MODNAME="#conftest12651"
 -nostdinc
 -isystem
 /usr/lib/gcc/x86_64-pc-linux-gnu/12.2.0/include
@@ -26,9 +26,10 @@
 -I/lib/modules/6.1.13teapot/build/arch/x86/include/generated/uapi
 -mfentry
 -DCC_USING_FENTRY
--I/tmp/7668/build/nvidia/NVIDIA-Linux-x86_64-530.30.02-no-compat32/kernel/common/inc
--I/tmp/7668/build/nvidia/NVIDIA-Linux-x86_64-530.30.02-no-compat32/kernel
+-I/tmp/11993/build/nvidia/NVIDIA-Linux-x86_64-525.89.02-no-compat32/kernel/common/inc
+-I/tmp/11993/build/nvidia/NVIDIA-Linux-x86_64-525.89.02-no-compat32/kernel
 -Wall
+-MD

 -Wno-cast-qual
@@ -37,7 +38,7 @@
 -D__KERNEL__
 -DMODULE
 -DNVRM
--DNV_VERSION_STRING=\"530.30.02\"
+-DNV_VERSION_STRING=\"525.89.02\"
 -Wno-unused-function
 -Wuninitialized
 -fno-strict-aliasing
```
well, other than `-MD`.
### i'd also like to clarify i am on linux-headers version `5.15.41-1`

retardtank, 2023-03-08:
I'd suspect I may have the same problem.

On the same setup 525.89.02 builds fine, while on 530.30.02 the following result:

```
# cd /dist/NVIDIA-Linux-x86_64-530.30.02/kernel/
/dist/NVIDIA-Linux-x86_64-530.30.02/kernel # make modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.86-0-lts'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Alpine 11.2.1_git20220219) 11.2.1 20220219
  You are using:           cc (Alpine 11.2.1_git20220219) 11.2.1 20220219
 CONFTEST: drm_dev_unref
 CONFTEST: drm_reinit_primary_mode_group
 CONFTEST: drm_gem_object_lookup
 CONFTEST: drm_atomic_state_ref_counting
 CONFTEST: drm_driver_has_gem_prime_res_obj
 CONFTEST: drm_atomic_helper_connector_dpms
 CONFTEST: drm_connector_funcs_have_mode_in_name
 CONFTEST: drm_connector_has_vrr_capable_property
 CONFTEST: drm_framebuffer_get
 CONFTEST: drm_dev_put
 CONFTEST: drm_format_num_planes
 CONFTEST: drm_connector_for_each_possible_encoder
 CONFTEST: drm_rotation_available
 CONFTEST: drm_vma_offset_exact_lookup_locked
 CONFTEST: nvhost_dma_fence_unpack
  CC [M]  /dist/NVIDIA-Linux-x86_64-530.30.02/kernel/nvidia/nv.o
In file included from /dist/NVIDIA-Linux-x86_64-530.30.02/kernel/common/inc/conftest.h:28,
                 from /dist/NVIDIA-Linux-x86_64-530.30.02/kernel/common/inc/nv_stdarg.h:29,
                 from /dist/NVIDIA-Linux-x86_64-530.30.02/kernel/common/inc/os-interface.h:40,
                 from /dist/NVIDIA-Linux-x86_64-530.30.02/kernel/nvidia/nv.c:33:
/dist/NVIDIA-Linux-x86_64-530.30.02/kernel/conftest/functions.h:71:2: error: #error dma_buf_export() conftest failed!
   71 | #error dma_buf_export() conftest failed!
      |  ^~~~~
/dist/NVIDIA-Linux-x86_64-530.30.02/kernel/conftest/functions.h:83:2: error: #error wait_on_bit_lock() conftest failed!
   83 | #error wait_on_bit_lock() conftest failed!
      |  ^~~~~
/dist/NVIDIA-Linux-x86_64-530.30.02/kernel/conftest/functions.h:86:2: error: #error radix_tree_replace_slot() conftest failed!
   86 | #error radix_tree_replace_slot() conftest failed!
      |  ^~~~~
In file included from /dist/NVIDIA-Linux-x86_64-530.30.02/kernel/common/inc/nv-linux.h:32,
                 from /dist/NVIDIA-Linux-x86_64-530.30.02/kernel/nvidia/nv.c:34:
/dist/NVIDIA-Linux-x86_64-530.30.02/kernel/common/inc/nv-lock.h:40:10: fatal error: asm/semaphore.h: No such file or directory
   40 | #include <asm/semaphore.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:289: /dist/NVIDIA-Linux-x86_64-530.30.02/kernel/nvidia/nv.o] Error 1
make[1]: *** [Makefile:1902: /dist/NVIDIA-Linux-x86_64-530.30.02/kernel] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.86-0-lts'
make: *** [Makefile:82: modules] Error 2
/dist/NVIDIA-Linux-x86_64-530.30.02/kernel # uname -a
Linux hostname 5.15.86-0-lts #1-Alpine SMP Mon, 02 Jan 2023 10:18:15 +0000 x86_64 Linux
```

apprehensions, 2023-03-31:
fixed build on my machine with 525.105.17

apprehensions, 2023-04-01:
however, has not been fixed as of 530.41.03

aritger, 2023-05-30:
@apprehensions : are you still seeing this problem with the latest 535.43.02 release?

apprehensions, 2023-05-30:
Sorry, I no longer have an Nvidia graphics card.

aritger, 2023-05-31:
@apprehensions: your log snippets in https://github.com/NVIDIA/open-gpu-kernel-modules/issues/468#issuecomment-1451276145 are surprising.  You said you are on linux-headers version 5.15.41-1, but your log shows you building 6.1.13teapot kernel headers.  It would be good if you could test the `-MD` experiment described below (you should be able to test build, without having an NVIDIA GPU).

@retardtank: could you please conduct the same experiment described here: https://github.com/NVIDIA/open-gpu-kernel-modules/issues/468#issuecomment-1450828462

Then, as a separate experiment, if the problem is really that 530.xx and newer branches is missing -MD from CFLAGS, try making this change:
```
diff --git a/kernel-open/conftest.sh b/kernel-open/conftest.sh
index 538b0b2..d168702 100755
--- a/kernel-open/conftest.sh
+++ b/kernel-open/conftest.sh
@@ -6315,7 +6315,7 @@ case "$5" in
         #
         shift 5

-        CFLAGS=$1
+        CFLAGS="$1 -MD"
         shift

         for i in $*; do compile_test $i; done
```

retardtank, 2023-05-31:
@aritger

There was no difference adding `-MD` to CFLAGS as described, same error (with `make clean; make modules`).

535.43.02:
```
cc x86_64 /lib/modules/5.15.86-0-lts/build /lib/modules/5.15.86-0-lts/build compile_tests -O2 -D__KERNEL__ -DKBUILD_BASENAME="#conftest14957" -DKBUILD_MODNAME="#conftest14957" -nostdinc -isystem /usr/lib/gcc/x86_64-alpine-linux-musl/11.2.1/include -Wno-implicit-function-declaration -Wno-strict-prototypes -I/lib/modules/5.15.86-0-lts/build/include/asm-x86/mach-default -I/lib/modules/5.15.86-0-lts/build/arch/x86/include/asm/mach-default -include /lib/modules/5.15.86-0-lts/build/include/generated/autoconf.h -I/lib/modules/5.15.86-0-lts/build/include -I/lib/modules/5.15.86-0-lts/build/include/uapi -I/lib/modules/5.15.86-0-lts/build/include/xen -I/lib/modules/5.15.86-0-lts/build/include/generated/uapi -I/lib/modules/5.15.86-0-lts/build/arch/x86/include -I/lib/modules/5.15.86-0-lts/build/arch/x86/include/uapi -I/lib/modules/5.15.86-0-lts/build/arch/x86/include/generated -I/lib/modules/5.15.86-0-lts/build/arch/x86/include/generated/uapi -mfentry -DCC_USING_FENTRY -I/tmp/NVIDIA-Linux-x86_64-535.43.02/kernel-open/common/inc -I/tmp/NVIDIA-Linux-x86_64-535.43.02/kernel-open -Wall   -Wno-cast-qual -Wno-error -Wno-format-extra-args -D__KERNEL__ -DMODULE -DNVRM -DNV_VERSION_STRING=\"535.43.02\" -Wno-unused-function -Wuninitialized -fno-strict-aliasing -ffreestanding -mno-red-zone -mcmodel=kernel -DNV_UVM_ENABLE -Werror=undef -DNV_SPECTRE_V2=0 -DNV_KERNEL_INTERFACE_LAYER -fno-pie dma_buf_export_args
```

525.89.02:
```
cc x86_64 /lib/modules/5.15.86-0-lts/build /lib/modules/5.15.86-0-lts/build compile_tests -O2 -D__KERNEL__ -DKBUILD_BASENAME="#conftest18547" -DKBUILD_MODNAME="#conftest18547" -nostdinc -isystem /usr/lib/gcc/x86_64-alpine-linux-musl/11.2.1/include -Wno-implicit-function-declaration -Wno-strict-prototypes -I/lib/modules/5.15.86-0-lts/build/include/asm-x86/mach-default -I/lib/modules/5.15.86-0-lts/build/arch/x86/include/asm/mach-default -include /lib/modules/5.15.86-0-lts/build/include/generated/autoconf.h -I/lib/modules/5.15.86-0-lts/build/include -I/lib/modules/5.15.86-0-lts/build/include/uapi -I/lib/modules/5.15.86-0-lts/build/include/xen -I/lib/modules/5.15.86-0-lts/build/include/generated/uapi -I/lib/modules/5.15.86-0-lts/build/arch/x86/include -I/lib/modules/5.15.86-0-lts/build/arch/x86/include/uapi -I/lib/modules/5.15.86-0-lts/build/arch/x86/include/generated -I/lib/modules/5.15.86-0-lts/build/arch/x86/include/generated/uapi -mfentry -DCC_USING_FENTRY -I/dist/NVIDIA-Linux-x86_64-525.89.02/kernel-open/common/inc -I/dist/NVIDIA-Linux-x86_64-525.89.02/kernel-open -Wall -MD   -Wno-cast-qual -Wno-error -Wno-format-extra-args -D__KERNEL__ -DMODULE -DNVRM -DNV_VERSION_STRING=\"525.89.02\" -Wno-unused-function -Wuninitialized -fno-strict-aliasing -ffreestanding -mno-red-zone -mcmodel=kernel -DNV_UVM_ENABLE -Werror=undef -DNV_SPECTRE_V2=0 -DNV_KERNEL_INTERFACE_LAYER -fno-pie dma_buf_export_args
```

Thanks.

folknor, 2024-01-29:
Getting this exact problem on ubuntu 24.04 noble using the recent nvidia 550.40.07-0ubuntu0~gpu24.04.1 packages from the graphics-drivers ppa.

cottsak, 2024-02-08:
same problem on `ubuntu/images/hvm-ssd/ubuntu-jammy-22.04-amd64-server-20231207` using:
* `NVIDIA-Linux-x86_64-410.104.run`, and 
* `NVIDIA-Linux-x86_64-470.57.02.run`

retardtank, 2024-02-10:
I've solved this with 545.29.06 on Alpine Linux 3.19, by installing gnu coreutils.

On my system at least, I'd guess that there might been some deviant behavior in the busybox counterpart of some standard command that gets called on, eventually causing conftest to fail.

git-bruh, 2024-03-07:
https://github.com/NVIDIA/open-gpu-kernel-modules/pull/609

StidOfficial, 2024-05-13:
> #609

Work for me, on Alpine Linux v3.19.

flukejones, 2024-09-30:
I'm hitting this same error log on Fedora 41 with Kernel 6.11.

flukejones, 2024-10-28:
I hit this again. It appears that the cause was my compiler was not the same as the one used to build my kernel. Updating both fixed it.

laion01, 2024-11-12:
> fixed build on my machine with 525.105.17

Could you help me to build it on my machine? @apprehensions

apprehensions, 2024-11-12:
Sorry, I switched to AMD.