NVIDIA/open-gpu-kernel-modules #471: Wayland is broken on 530.30.02

solbjorn, 2023-03-09:
### NVIDIA Open GPU Kernel Modules Version

530.30.02 (4397463e738d)

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux rolling x86_64

### Kernel Release

6.2.0, local build

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3090 (UUID: GPU-cdf78b31-460a-c570-ea90-4e05c0a1c4d9)

### Describe the bug

The same here:

https://www.reddit.com/r/archlinux/comments/11ivucn/cant_get_any_de_working_on_wayland_since_updating/

and here:

https://forums.developer.nvidia.com/t/cant-boot-into-kde-wayland-session-with-530-30-02/244559

so this report is the third at least...

Basically Wayland doesn't work after upgrading to 530.30.02. Disabling modesetting or just disabling Wayland with enabled modeset makes system work (under X11).
I don't have any way to control the PC without the primary monitor, but in the reports above there's a couple logs. They refer to `-EINVAL` when an atomic commit happens.

### To Reproduce

* have modeset enabled;
* run any Wayland-capable DE (GNOME Shell, KDE are confirmed).

### Bug Incidence

Always

### nvidia-bug-report.log.gz

I have no graphics when this happens, so I personally can't :s

gilvbp, 2023-03-09:
I'm using it and only have problems using a monitor fresh rate above 60hz.

solbjorn, 2023-03-09:
@gilvbp, thanks for the hint. That actually gave me a bit.
I experimented with different fresh rates and have this:

- if GDM has a fresh rate <= 60 Hz, it works even if the DE has a higher one (144 works fine);
- if GDM has a higher fresh rate (set up via `/var/lib/gdm/.config/monitors.xml`), the graphics won't appear.

Curious one lol. If only NVIDIA had normal Git commit flow, not this "one commit per release", I'd bisect it quickly, but hard to say for now.

gilvbp, 2023-03-09:
@shbaskaran, you're welcome. Thank you for the tip.

amrit1711, 2023-03-13:
Thanks for reporting the issue. 
I have filed a bug internally 4026365 internally for tracking purpose. Shall try to reproduce issue locally which will help in debugging.
Since this issue is not limited to open gpu kernel module components, I will close this thread and request you all to follow Nvidia Developer forum for generic issues.
 https://forums.developer.nvidia.com/t/cant-boot-into-kde-wayland-session-with-530-30-02/244559

closed 2023-03-13 by amrit1711