NVIDIA/open-gpu-kernel-modules #472: Suspend doesn't work when PreserveVideoMemoryAllocations is enabled

Monsterovich, 2023-03-11:
### NVIDIA Open GPU Kernel Modules Version

525.85.05

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Linux Mint 21.1

### Kernel Release

Linux pc 5.19.15-xanmod1 #0~20221012.git85103f7 SMP Wed Oct 12 15:04:07 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3060 Ti (UUID: GPU-039a34f7-087c-dc8f-a607-e0d937443d0f)

### Describe the bug

An error occurs when trying to boot into the system after suspend if PreserveVideoMemoryAllocations is enabled:

```
[   10.255125] NVRM: GPU 0000:05:00.0: PreserveVideoMemoryAllocations module parameter is set. System Power Management attempted without driver procfs suspend interface. Please refer to the 'Configuring Power Management Support' section in the driver README.
[   10.255127] nvidia 0000:05:00.0: PM: pci_pm_freeze(): nv_pmops_freeze+0x0/0x20 [nvidia] returns -5
[   10.255200] nvidia 0000:05:00.0: PM: dpm_run_callback(): pci_pm_freeze+0x0/0xc0 returns -5
[   10.255203] nvidia 0000:05:00.0: PM: failed to quiesce async: error -5
```

### To Reproduce

1. Create a file `/etc/modprobe.d/nvidia-power-management.conf` with the following content:
```
options nvidia NVreg_PreserveVideoMemoryAllocations=1 NVreg_TemporaryFilePath=/tmp
```
2. `sudo update-initramfs -c -k all` to re-generate the kernels.
3. Reboot.
4. Go to suspend mode.
5. Get an error when trying to load a saved state.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10949264/nvidia-bug-report.log.gz)

### More Info

https://forums.developer.nvidia.com/t/cannot-suspend-gtx3060-laptop/244143/9
https://forums.developer.nvidia.com/t/preservevideomemoryallocations-systemd-services-causes-resume-from-hibernate-to-fail/233643/3

David7ce, 2023-03-12:
Same bug with suspend mode on Arch Linux, I suspend I can'get a black screen. Running:
```
linux 6.2.2.arch2-1
nvidia-open 525.89.02-7
xf86-video-nouveau 1.0.17-2
```

aaronp24, 2023-03-13:
How did you trigger suspend on this system? It looks like the relevant systemd units are enabled but I don't see any evidence that they were triggered. Something needs to run `/usr/bin/nvidia-sleep.sh suspend` before suspend and `/usr/bin/nvidia-sleep.sh resume` after resume.

For context, the video memory preservation process needs to happen before the kernel's low-level suspend process is triggered. The message from the driver implies that something bypassed  "System Power Management attempted without driver procfs suspend interface. Please refer to the 'Configuring Power Management Support' section in the driver README" implies that that bracketing didn't happen for some reason.

David7ce, 2023-03-14:
Whit NVIDiA propietary graphics suspend mode for me works fine.
And I suspend the systems with KDE GUI or via terminal typing systemctl suspend. And the problem was when I tried to wake up the system, always black screen...

Monsterovich, 2023-03-17:
> How did you trigger suspend on this system? It looks like the relevant systemd units are enabled but I don't see any evidence that they were triggered.

Nevertheless, all these services are enabled in my system, but they don't as they should.

```
systemctl list-unit-files | grep nvidia
nvidia-hibernate.service                                                  enabled         enabled
nvidia-persistenced.service                                               static          -
nvidia-resume.service                                                     enabled         enabled
nvidia-suspend.service                                                    enabled         enabled
```

amrit1711, 2023-06-05:
Hi All, 
Please test with latest released driver 535.43.02 and share test results.

Sid127, 2023-06-15:
still doesn't work with 535.43.02

Monsterovich, 2023-07-01:
> Hi All, Please test with latest released driver 535.43.02 and share test results.

@amrit1711

NVidia driver 535.54.03.

Regular suspend now doesn't work on open-source driver. The screen simply turns off after waking up.

[dmesg.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11927050/dmesg.log)

```
июл 01 22:08:37 pc kernel: NVRM rpcRmApiAlloc_GSP: GspRmAlloc failed: hClient=0xc1d00000; hParent=0x00010001; hObject=0x00010011; hClass=0x0000c670; paramsSize=0x00000000; paramsStatus=0x00000062; status=0x00000062
июл 01 22:08:37 pc kernel: nvidia-modeset: ERROR: GPU:0: Failed to initialize display engine: 0x62 (Reset required [NV_ERR_RESET_REQUIRED])
июл 01 22:08:37 pc kernel: NVRM serverFreeResourceTree: hObject 0x10011 not found for client 0xc1d00000
июл 01 22:08:37 pc kernel: NVRM unixCallVideoBIOS: int10h(4f02, 0177) vesa call failed! (02a2, 0000)
июл 01 22:08:37 pc kernel: NVRM rpcRmApiAlloc_GSP: GspRmAlloc failed: hClient=0xc1d00000; hParent=0x00010001; hObject=0x00010011; hClass=0x0000c670; paramsSize=0x00000000; paramsStatus=0x00000062; status=0x00000062
июл 01 22:08:37 pc kernel: nvidia-modeset: ERROR: GPU:0: Failed to initialize display engine: 0x62 (Reset required [NV_ERR_RESET_REQUIRED])
июл 01 22:08:37 pc kernel: NVRM serverFreeResourceTree: hObject 0x10011 not found for client 0xc1d00000
июл 01 22:08:37 pc kernel: NVRM unixCallVideoBIOS: int10h(4f02, 0177) vesa call failed! (02a2, 0000)
июл 01 22:08:37 pc kernel: NVRM rpcRmApiAlloc_GSP: GspRmAlloc failed: hClient=0xc1d00000; hParent=0x00010001; hObject=0x00010011; hClass=0x0000c670; paramsSize=0x00000000; paramsStatus=0x00000062; status=0x00000062
июл 01 22:08:37 pc kernel: nvidia-modeset: ERROR: GPU:0: Failed to initialize display engine: 0x62 (Reset required [NV_ERR_RESET_REQUIRED])
июл 01 22:08:37 pc kernel: NVRM serverFreeResourceTree: hObject 0x10011 not found for client 0xc1d00000
июл 01 22:08:37 pc kernel: NVRM unixCallVideoBIOS: int10h(4f02, 0177) vesa call failed! (02a2, 0000)
июл 01 22:08:37 pc kernel: NVRM rpcRmApiAlloc_GSP: GspRmAlloc failed: hClient=0xc1d00000; hParent=0x00010001; hObject=0x00010011; hClass=0x0000c670; paramsSize=0x00000000; paramsStatus=0x00000062; status=0x00000062
июл 01 22:08:37 pc kernel: nvidia-modeset: ERROR: GPU:0: Failed to initialize display engine: 0x62 (Reset required [NV_ERR_RESET_REQUIRED])
```

Suspend to disk doesn't work either. The same message as in the first post.

I also see garbage messages in dmesg:

```
[   12.759538] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 11 shares a pVAS with channel 12
[   12.760963] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 12 shares a pVAS with channel 13
[   12.762454] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 13 shares a pVAS with channel 14
[   12.763909] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 14 shares a pVAS with channel 15
[   12.765581] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 15 shares a pVAS with channel 16
[   12.767228] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 16 shares a pVAS with channel 17
[   12.768622] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 17 shares a pVAS with channel 18
[   12.920867] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 11 shares a pVAS with channel 12
[   12.922407] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 12 shares a pVAS with channel 13
[   12.923962] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 13 shares a pVAS with channel 14
[   12.925519] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 14 shares a pVAS with channel 15
[   12.926825] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 15 shares a pVAS with channel 16
[   12.928098] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 16 shares a pVAS with channel 17
[   12.929377] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 17 shares a pVAS with channel 18
[   13.608901] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 11 shares a pVAS with channel 12
[   13.611364] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 12 shares a pVAS with channel 13
[   13.613183] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 13 shares a pVAS with channel 14
[   13.614439] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 14 shares a pVAS with channel 15
[   13.616568] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 15 shares a pVAS with channel 16
[   13.620816] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 16 shares a pVAS with channel 17
[   13.622490] NVRM kgrctxUnmapAssociatedCtxBuffers_IMPL: TSG 9 Channel 17 shares a pVAS with channel 18
```

and

```
[    5.099933] NVRM cpuidInfoAMD: Unrecognized AMD processor in cpuidInfoAMD
```

But I guess they don't affect anything.

zaltysz, 2023-07-06:
Same problem on GTX 3080 with kernel 6.4.2 and open driver 535.54.03. 

During suspend preparation, after nvidia-suspend.service has been activated kernel complains:

```
kernel: NVRM nvAssertFailedNoLog: Assertion failed: 0 @ mem_list.c:293kernel: NVRM nvAssertFailedNoLog: Assertion failed: 0 @ mem_list.c:293
kernel: NVRM nvAssertOkFailedNoLog: Assertion failed: Call not supported [NV_ERR_NOT_SUPPORTED] (0x00000056) returned from pRmApi->Alloc(pRmApi, pMemoryManager->hClient>
kernel: NVRM nvAssertOkFailedNoLog: Assertion failed: Call not supported [NV_ERR_NOT_SUPPORTED] (0x00000056) returned from memdescSendMemDescToGSP(pGpu, pFbsr->pSysMemD>
kernel: NVRM nvAssertOkFailedNoLog: Assertion failed: Call not supported [NV_ERR_NOT_SUPPORTED] (0x00000056) returned from _fbsrInitGsp(pGpu, pFbsr) @ fbsr_gm107.c:548
```

Looks like some sort of GSP firmware and driver incompatibility/miscommunication.

hugh712, 2023-07-10:
I'm also has this issue in 
- Ubuntu 23.04
- NVIDIA Corporation TU116 [GeForce GTX 1660 SUPER] [10de:21c4]
- Kernel 6.2.0-24-generic
- Nvidia 525.116.04  /  535.54.03

FlorianFranzen, 2023-07-13:
Suspend leads to similar error logs before suspend and a kernel panic while waking up from suspend on my system.
It is a laptop with an AMD Ryzen 7 5800H and an RTX 3070 Mobile running kernel 6.3.9 and version 535.54.03. I boot with the following flags: `nvidia-drm.modeset=1 NVreg_EnableResizableBar=1 nvidia.NVreg_PreserveVideoMemoryAllocations=1 nvidia.NVreg_OpenRmEnableUnsupportedGpus=1`

Before suspend I find the following in the logs:
```
NVRM nvAssertFailedNoLog: Assertion failed: 0 @ mem_list.c:293
NVRM nvAssertOkFailedNoLog: Assertion failed: Call not supported [NV_ERR_NOT_SUPPORTED] (0x00000056) returned from pRmApi->Alloc(pRmApi, pMemoryManager->hClient, pMemoryManager->
NVRM nvAssertOkFailedNoLog: Assertion failed: Call not supported [NV_ERR_NOT_SUPPORTED] (0x00000056) returned from memdescSendMemDescToGSP(pGpu, pFbsr->pSysMemDesc, &hSysMem) @ >
NVRM nvAssertOkFailedNoLog: Assertion failed: Call not supported [NV_ERR_NOT_SUPPORTED] (0x00000056) returned from _fbsrInitGsp(pGpu, pFbsr) @ fbsr_gm107.c:548
```
After suspend I get more error logs and a kernel panic:
```
NVRM: GPU at PCI:0000:01:00: GPU-cd48b96b-c536-1a29-44c0-809f5b2531e4
NVRM: Xid (PCI:0000:01:00): 45, pid=975, name=(udev-worker), Ch 00000000
NVRM: Xid (PCI:0000:01:00): 45, pid=975, name=(udev-worker), Ch 00000001
NVRM: Xid (PCI:0000:01:00): 45, pid=3823, name=Renderer, Ch 00000002
NVRM: Xid (PCI:0000:01:00): 45, pid=3823, name=Renderer, Ch 00000003
NVRM: Xid (PCI:0000:01:00): 45, pid=3823, name=Renderer, Ch 00000004
NVRM: Xid (PCI:0000:01:00): 45, pid=3823, name=Renderer, Ch 00000005
NVRM: Xid (PCI:0000:01:00): 45, pid=3823, name=Renderer, Ch 00000006
NVRM: Xid (PCI:0000:01:00): 45, pid=3823, name=Renderer, Ch 00000007
NVRM: Xid (PCI:0000:01:00): 45, pid=2819, name=sway, Ch 00000008
NVRM: Xid (PCI:0000:01:00): 45, pid=2819, name=sway, Ch 00000009
NVRM: Xid (PCI:0000:01:00): 45, pid=2819, name=sway, Ch 0000000a
NVRM: Xid (PCI:0000:01:00): 45, pid=2905, name=Xwayland, Ch 0000000b
NVRM: Xid (PCI:0000:01:00): 45, pid=2905, name=Xwayland, Ch 0000000c
ata2: SATA link down (SStatus 0 SControl 300)
NVRM: Xid (PCI:0000:01:00): 45, pid=2905, name=Xwayland, Ch 0000000d
NVRM: Xid (PCI:0000:01:00): 45, pid=23536, name=slack, Ch 0000000e
NVRM: Xid (PCI:0000:01:00): 45, pid=23536, name=slack, Ch 0000000f
NVRM: Xid (PCI:0000:01:00): 45, pid=23536, name=slack, Ch 00000010
NVRM kbusVerifyBar2_GM107: MMUTest BAR0 window offset 0x70e000 returned garbage 0x0
NVRM nvAssertOkFailedNoLog: Assertion failed: Generic memory error [NV_ERR_MEMORY_ERROR] (0x00000072) returned from kbusVerifyBar2_HAL(pGpu, pKernelBus, NULL, NULL, 0, 0) @ kern_bus_gm107.c:457
NVRM nvAssertOkFailedNoLog: Assertion failed: Generic memory error [NV_ERR_MEMORY_ERROR] (0x00000072) returned from gpuStateLoad(pGpu, IS_GPU_GC6_STATE_EXITING(pGpu) ? GPU_STATE_FLAGS_PRESERVING | GPU_STATE_FL>
nvidia-modeset: ERROR: GPU:0: Failed to initialize display engine: 0x62 (Reset required [NV_ERR_RESET_REQUIRED])
NVRM serverFreeResourceTree: hObject 0x10011 not found for client 0xc1d00000
NVRM rpcRmApiAlloc_GSP: GspRmAlloc failed: hClient=0xc1d00000; hParent=0x00010001; hObject=0x00010011; hClass=0x0000c670; paramsSize=0x00000000; paramsStatus=0x00000062; status=0x00000062
nvidia-modeset: ERROR: GPU:0: Failed to initialize display engine: 0x62 (Reset required [NV_ERR_RESET_REQUIRED])
NVRM serverFreeResourceTree: hObject 0x10011 not found for client 0xc1d00000
BUG: kernel NULL pointer dereference, address: 000000000000026c
#PF: supervisor read access in kernel mode
#PF: error_code(0x0000) - not-present page
PGD 15dde0067 P4D 15dde0067 PUD 1270b5067 PMD 0 
Oops: 0000 [#1] PREEMPT SMP NOPTI
CPU: 3 PID: 2819 Comm: sway Tainted: G           O       6.3.9 #1-NixOS
Hardware name: LENOVO 82N6/LNVNB161216, BIOS GKCN60WW 03/07/2023
RIP: 0010:nvAssignDefaultUsageBounds+0x94/0x1a0 [nvidia_modeset]
Code: 43 0c 00 89 43 08 49 8b 84 24 60 0a 00 00 48 8b 80 98 01 00 00 48 85 c0 74 b4 4c 89 e7 e8 24 1a 8f ef ba 04 00 00 00 48 63 ca <8b> 74 c8 08 85 f6 74 5c 48 8d 04 c8 0f b7 48 08 66 89 4b 04 0f b7
RSP: 0018:ffffaa0c088a74c0 EFLAGS: 00010206
RAX: 0000000000000244 RBX: ffff9b5c26c5ea58 RCX: 0000000000000004
RDX: 0000000000000004 RSI: ffff9b5c26c5ea40 RDI: ffff9b5c3a8e0008
RBP: ffff9b5c3a8e06a0 R08: 0000000000000005 R09: 0000000000000000
R10: 00000000000006c4 R11: 0000000000000000 R12: ffff9b5c3a8e0008
R13: ffff9b5c26c5ea40 R14: ffff9b5c26c5ea58 R15: ffff9b5c3a8e07e0
FS:  00007ff08edc6f80(0000) GS:ffff9b630dcc0000(0000) knlGS:0000000000000000
CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
CR2: 000000000000026c CR3: 0000000152924000 CR4: 0000000000750ee0
PKRU: 55555554
Call Trace:
 <TASK>
 ? __die+0x23/0x70
 ? page_fault_oops+0x17d/0x4b0
 ? number+0x34a/0x3e0
 ? exc_page_fault+0x6a/0x150
 ? asm_exc_page_fault+0x26/0x30
 ? nvAssignDefaultUsageBounds+0x94/0x1a0 [nvidia_modeset]
 ConstructHwModeTimingsViewPort.constprop.0+0x118/0x160 [nvidia_modeset]
 nvConstructHwModeTimingsEvo+0x344/0x5e0 [nvidia_modeset]
 ? vsnprintf+0x2cb/0x550
 ? nvEvoLogInfoString+0xa0/0xc0 [nvidia_modeset]
[... 23 lines cut ...]
RSP: 002b:00007ffc649f08d0 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
RAX: ffffffffffffffda RBX: 0000000001a42690 RCX: 00007ff08fe3529f
RDX: 00007ffc649f0970 RSI: 00000000c05064a7 RDI: 000000000000000b
RBP: 00007ffc649f0970 R08: 0000000000000007 R09: 0000000001693b60
R10: 7909572cf32bdd3d R11: 0000000000000246 R12: 00000000c05064a7
R13: 000000000000000b R14: 00000000c05064a7 R15: 00007ffc649f0970
 </TASK>
[... cut ...]
RAX: 0000000000000244 RBX: ffff9b5c26c5ea58 RCX: 0000000000000004
RDX: 0000000000000004 RSI: ffff9b5c26c5ea40 RDI: ffff9b5c3a8e0008
RBP: ffff9b5c3a8e06a0 R08: 0000000000000005 R09: 0000000000000000
R10: 00000000000006c4 R11: 0000000000000000 R12: ffff9b5c3a8e0008
R13: ffff9b5c26c5ea40 R14: ffff9b5c26c5ea58 R15: ffff9b5c3a8e07e0
FS:  00007ff08edc6f80(0000) GS:ffff9b630dcc0000(0000) knlGS:0000000000000000
CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
CR2: 000000000000026c CR3: 0000000152924000 CR4: 0000000000750ee0
PKRU: 55555554
```

aritger, 2023-07-13:
Thank you for the reports and logs.  We understand the issue, and NVIDIA internal bug 4193761 tracks this.

Unfortunately it will probably be the release cycle after 535.xx, at the earliest, before NVreg_PreserveVideoMemoryAllocations works correctly with the open-gpu-kernel-modules.  Sorry, and thanks for your patience.

Longhao-Chen, 2023-09-29:
Is there any progress?

ifaigios, 2023-11-08:
Still doesn't work with nvidia-open 545.29.02

MXXXXXS, 2023-11-09:
**Found solution here!** See https://forums.developer.nvidia.com/t/cannot-suspend-gtx3060-laptop/244143/12

1. Download the [nvidia-services-install.zip](https://forums.developer.nvidia.com/uploads/short-url/2ZtALqhLyZlrEx0a5ulzFsc3CXD.zip)
2. `cd ~/Downloads/` then extract `sudo unzip nvidia-services-install.zip -d /usr/share/doc/nvidia-driver-545/` (change the driver version number to yours)
3. Modify the file `sudo vim setup-nvdia-suspend.sh`, change this line to 
`TMPL_PATH=/usr/share/doc/nvidia-driver-545`, save and quit
4. Then run `sudo ./setup-nvdia-suspend.sh`
5. Try suspend, it should works

veldenb, 2023-11-09:
> **Found solution here!** See https://forums.developer.nvidia.com/t/cannot-suspend-gtx3060-laptop/244143/12
>
>     1. Download the [...]

I think this is with the proprietary kernel modules and not the open kernel modules?

MXXXXXS, 2023-11-09:
> I think this is with the proprietary kernel modules and not the open kernel modules?

Oh... overlooked the title, yes it works on the proprietary driver, not tested on the open driver

edwloef, 2023-11-17:
> **Found solution here!** See https://forums.developer.nvidia.com/t/cannot-suspend-gtx3060-laptop/244143/12
>
>     1. Download the [...]

This has not fixed the issue for me, on the proprietary driver or the open driver.

veldenb, 2023-12-11:
Suspend seems to work sort of, but when waking up it fails by not turning the display on. The machine is sill responsible. While going into sleep mode I get error messages starting with:
```
[   61.128594] NVRM nvAssertFailedNoLog: Assertion failed: 0 @ mem_list.c:268
[   61.128750] NVRM nvAssertOkFailedNoLog: Assertion failed: Call not supported [NV_ERR_NOT_SUPPORTED] (0x00000056) returned from pRmApi->Alloc(pRmApi, pMemoryManager->hClient, pMemoryManager->hSubdevice, pHandle, hClass, &listAllocParams, sizeof(listAllocParams)) @ mem_desc.c:4841
[   61.128788] NVRM serverFreeResourceTree: hObject 0xcaf00003 not found for client 0xc1e00002
[   61.128803] NVRM nvAssertOkFailedNoLog: Assertion failed: Call not supported [NV_ERR_NOT_SUPPORTED] (0x00000056) returned from memdescSendMemDescToGSP(pGpu, pFbsr->pSysMemDesc, &hSysMem) @ fbsr_gm107.c:113
[   61.128830] NVRM nvAssertOkFailedNoLog: Assertion failed: Call not supported [NV_ERR_NOT_SUPPORTED] (0x00000056) returned from _fbsrInitGsp(pGpu, pFbsr) @ fbsr_gm107.c:550
```

And when waking up:
```
[   67.529695] NVRM kbusVerifyBar2_GM107: MMUTest BAR0 window offset 0x70f000 returned garbage 0x0
[   67.530492] NVRM kbusVerifyBar2_GM107: MMUTest BAR0 window offset 0x70f000 returned garbage 0x0
[   67.530999] NVRM kbusVerifyBar2_GM107: MMUTest BAR0 window offset 0x70f000 returned garbage 0x0
[   67.531515] NVRM kbusVerifyBar2_GM107: MMUTest BAR0 window offset 0x70f000 returned garbage 0x0
```
(full log below)

### Specs
```
Linux desktop 6.6.6-060606-generic #202312111032 SMP PREEMPT_DYNAMIC Mon Dec 11 10:41:57 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
nvidia-driver-545-open 545.29.06-0ubuntu0~gpu23.10.1
```

```
$ systemctl list-unit-files | grep nvidia
nvidia-hibernate.service                                           enabled         enabled
nvidia-persistenced.service                                        static          -
nvidia-resume.service                                              enabled         enabled
nvidia-suspend.service                                             enabled         enabled
```

```
$ nvidia-smi 
Mon Dec 11 22:37:59 2023       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 545.29.06              Driver Version: 545.29.06    CUDA Version: 12.3     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 3080        Off | 00000000:0B:00.0  On |                  N/A |
|  0%   43C    P8              23W / 320W |    498MiB / 10240MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
```

<details> 
  <summary>Syslog with dmesg output</summary>

```
2023-12-11T22:19:47.492445+01:00 desktop systemd[1]: Reached target sleep.target - Sleep.
2023-12-11T22:19:47.509928+01:00 desktop systemd[1]: Starting nvidia-suspend.service - NVIDIA system suspend actions...
2023-12-11T22:19:47.526542+01:00 desktop suspend: nvidia-suspend.service
2023-12-11T22:19:47.526641+01:00 desktop logger[3676]: <13>Dec 11 22:19:47 suspend: nvidia-suspend.service
2023-12-11T22:19:47.780492+01:00 desktop bluetoothd[1461]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/ldac
2023-12-11T22:19:47.780556+01:00 desktop bluetoothd[1461]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSink/aptx_hd
2023-12-11T22:19:47.780703+01:00 desktop bluetoothd[1461]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/aptx_hd
2023-12-11T22:19:47.780876+01:00 desktop bluetoothd[1461]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSink/aptx
2023-12-11T22:19:47.781048+01:00 desktop bluetoothd[1461]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/aptx
2023-12-11T22:19:47.781223+01:00 desktop bluetoothd[1461]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSink/sbc
2023-12-11T22:19:47.781397+01:00 desktop kernel: [   60.990030] rfkill: input handler enabled
2023-12-11T22:19:47.781448+01:00 desktop bluetoothd[1461]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/sbc
2023-12-11T22:19:47.781575+01:00 desktop bluetoothd[1461]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSink/sbc_xq
2023-12-11T22:19:47.781756+01:00 desktop bluetoothd[1461]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/sbc_xq
2023-12-11T22:19:47.781943+01:00 desktop bluetoothd[1461]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/aptx_ll_1
2023-12-11T22:19:47.782130+01:00 desktop bluetoothd[1461]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/aptx_ll_0
2023-12-11T22:19:47.782350+01:00 desktop bluetoothd[1461]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
2023-12-11T22:19:47.782552+01:00 desktop bluetoothd[1461]: Endpoint unregistered: sender=:1.42 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
[... 264 lines cut ...]
[... cut ...]
2023-12-11T22:20:08.989318+01:00 desktop kernel: [   67.573665] NVRM kbusVerifyBar2_GM107: MMUTest BAR0 window offset 0x70f000 returned garbage 0x0
2023-12-11T22:20:08.989318+01:00 desktop kernel: [   67.574726] NVRM kbusVerifyBar2_GM107: MMUTest BAR0 window offset 0x70f000 returned garbage 0x0
2023-12-11T22:20:08.989319+01:00 desktop kernel: [   67.575785] NVRM kbusVerifyBar2_GM107: MMUTest BAR0 window offset 0x70f000 returned garbage 0x0
2023-12-11T22:20:08.993312+01:00 desktop kernel: [   67.576843] NVRM kbusVerifyBar2_GM107: MMUTest BAR0 window offset 0x70f000 returned garbage 0x0
...this repeats a lot...
```

</details>

relief-melone, 2023-12-31:
I also have the same issue on with nixos in the unstable channel. Nvidia 545 and from all kernel versions 6.x so far.
I am also using the proprietary drivers

ElDavoo, 2024-01-14:
Same here , PCS laptop with RTX 3070 
Gentoo, 6.7.9 kernel, 545.29.06 driver

miguno, 2024-01-29:
Same issue here.  Suspend doesn't work.

Environment: NVIDIA proprietary driver on Debian 12, with X11 (not Wayland).

```
$ uname -a
Linux doom 6.1.0-17-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.69-1 (2023-12-30) x86_64 GNU/Linux

# From journalctl
Jan 29 19:31:08 doom kernel: NVRM: GPU 0000:54:00.0: PreserveVideoMemoryAllocations module parameter is set. System Power Management attempted without driver procfs sus
pend interface. Please refer to the 'Configuring Power Management Support' section in the driver README.
Jan 29 19:31:08 doom kernel: nvidia 0000:54:00.0: PM: pci_pm_suspend(): nv_pmops_suspend+0x0/0x20 [nvidia] returns -5
Jan 29 19:31:08 doom kernel: nvidia 0000:54:00.0: PM: dpm_run_callback(): pci_pm_suspend+0x0/0x170 returns -5
Jan 29 19:31:08 doom kernel: nvidia 0000:54:00.0: PM: failed to suspend async: error -5
```

NVIDIA driver information:

```
$ nvidia-smi
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 545.23.08              Driver Version: 545.23.08    CUDA Version: 12.3     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
```

jofr49, 2024-01-30:
Same problem with resuming from hibernation for me.
* Debian 12.4 on X11
  `Linux 6.1.0-17-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.69-1 (2023-12-30) x86_64 GNU/Linux`
* NVIDIA driver on a old GT640:
`NVIDIA-SMI 470.223.02   Driver Version: 470.223.02   CUDA Version: 11.4`

amrit1711, 2024-01-31:
We are actively working upon it, shall update once the fix is integrated into release drivers.

Fxzzi, 2024-05-16:
Hey @amrit1711, It's been quite a few months. Any updates on the fix? Considering the open kernel modules will soon be the default and recommended for newer GPUs, getting this solved will allow many users to finally make the switch. I've been stuck using the proprietary drivers due to this.

mtijanic, 2024-05-16:
Hey @Fxzzi , thanks for the vigilance. This is definitely on the radar and being actively worked on. Sorry, can't really give any concrete dates, but very much stay tuned :)

Fxzzi, 2024-05-16:
Thanks for the reply @mtijanic! Take your time, I'm just happy to see this is being worked on :)

mtijanic, 2024-05-21:
We missed calling it out in the changelog explicitly (oops), but this should be fixed with [555.42.02](https://github.com/NVIDIA/open-gpu-kernel-modules/tree/555.42.02). Please test. I'll leave this bug open while 555.xx is still in beta.

Fxzzi, 2024-05-21:
Can confirm the fix has indeed solved the issue on my end. Running latest 555 open kernel modules (dkms) and suspend and resume working great with PreserveVideoMemoryAllocations!!

WJKPK, 2024-05-24:
I tried Nvidia 555.42.02 with kernels: 6.6.31 and 6.8.10 and error remains the same. On which kernel did it work?

Zaggy1024, 2024-05-30:
Running 555.42.02 from [this AUR package](https://aur.archlinux.org/packages/nvidia-open-beta) with PreserveVideoMemoryAllocations, I would occasionally get some artifacts when resuming from suspend which looked like this:

[image]

The reason I initially noticed this was because my Logitech mouse's wireless receiver was causing spurious wakeups, but I was also able to reproduce the issue by repeatedly running `systemctl suspend` and then spamming a key to wake up the system until I saw artifacts.

After switching back to the stable closed drivers, I was unable to reproduce the issue.

All the changes I made to my system relating to the NVIDIA drivers:
- Installed `nvidia-open-beta` 555.42.02 from AUR
- Enabled `modeset`, `fbdev` in `/boot/loader/entries/*_linux.conf`
- Set `NVreg_PreserveVideoMemoryAllocations=1` and `NVreg_TemporaryPath=/var/tmp` in `/etc/modprobe.d/nvidia-power-management.conf` to allow Wayland to function
- Enabled `nvidia-suspend`, `nvidia-resume`, `nvidia-hibernate` services with systemctl
- Installed `libva-nvidia-driver` and set `MOZ_DISABLE_RDD_SANDBOX=1`, `NVD_BACKEND=direct`, and `LIBVA_DRIVER_NAME=nvidia` in `/etc/environment` to enable Firefox hardware acceleration
- Installed `gst-plugin-va` for GNOME Videos hardware acceleration

In the systemd journal, I didn't see anything that indicated any errors occurred when it invoked the suspend or resume scripts.

urbenlegend, 2024-06-03:
@Zaggy1024 I am also getting those squares on random occasions. Doesn't happen too frequently but it does happen.

I am also getting a crash on suspend once in a blue moon. Here's the logs whenever the crash happens: 
[suspend_hang.txt](https://github.com/user-attachments/files/15529703/suspend_hang.txt)

**System Info:**
Operating System: Arch Linux
KDE Plasma Version: 6.0.5
KDE Frameworks Version: 6.2.0
Qt Version: 6.7.1
Kernel Version: 6.9.2-arch1-1 (64-bit)
Graphics Platform: Wayland
Processors: 24 × AMD Ryzen 9 3900X 12-Core Processor
Memory: 31.3 GiB of RAM
Graphics Processor: NVIDIA GeForce RTX 3090/PCIe/SSE2 w/ 555.42.02 drivers + open kernel module
Product Name: X570 Taichi

amrit1711, 2024-06-04:
> Running 555.42.02 from [this AUR package](https://aur.archlinux.org/packages/nvidia-open-beta) with PreserveVideoMemoryAllocations, I would occasionally get some artifacts when resuming from suspend which looked like this:
> 
> ![Screenshot from 2024-05-29 18-29-32](https://private-user-images.githubusercontent.com/3149592/335077676-7075e462-75bd-47a4-bd32-e029344f12a9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc0Nzk1MTcsIm5iZiI6MTcxNzQ3OTIxNywicGF0aCI6Ii8zMTQ5NTkyLzMzNTA3NzY3Ni03MDc1ZTQ2Mi03NWJkLTQ3YTQtYmQzMi1lMDI5MzQ0ZjEyYTkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDRUMDUzMzM3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWNjNTcxYTU2YzExMjMyZjI5YjA0MzAwMzkwNmQ5NzIwMGY5MTZlMGE5MTY4ODYyYThhZGEzZmU3NTZkNjk1OSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.eX7MHubpoVhQPah9KSXI4ufBH7TFJ6dDpakhQfQMGE0)
> 
> The artifacts would also affect glyph atlases and any textures/framebuffers of applications that were open during the suspend, but when any affected textures were regenerated (i.e. by resizing a browser window), the issue would disappear.
> 
> The reason I initially noticed this was because my Logitech mouse's wireless receiver was causing spurious wakeups, but I was also able to reproduce the issue by repeatedly running `systemctl suspend` and then spamming a key to wake up the system until I saw artifacts.
> 
> After switching back to the stable closed drivers, I was unable to reproduce the issue.
> 
> All the changes I made to my system relating to the NVIDIA drivers:
> 
> * Installed `nvidia-open-beta` 555.42.02 from AUR
> * Enabled `modeset`, `fbdev` in `/boot/loader/entries/*_linux.conf`
> * Set `NVreg_PreserveVideoMemoryAllocations=1` and `NVreg_TemporaryPath=/var/tmp` in `/etc/modprobe.d/nvidia-power-management.conf` to allow Wayland to function
> * Enabled `nvidia-suspend`, `nvidia-resume`, `nvidia-hibernate` services with systemctl
> * Installed `libva-nvidia-driver` and set `MOZ_DISABLE_RDD_SANDBOX=1`, `NVD_BACKEND=direct`, and `LIBVA_DRIVER_NAME=nvidia` in `/etc/environment` to enable Firefox hardware acceleration
> * Installed `gst-plugin-va` for GNOME Videos hardware acceleration
> 
> In the systemd journal, I didn't see anything that indicated any errors occurred when it invoked the suspend or resume scripts.
> 
> I'm running X11 on GNOME:
> 
> ```
>                    -`                    zaggy1024@MYCOMPY 
>                   .o+`                   ----------------- 
>                  `ooo/                   OS: Arch Linux x86_64 
>                 `+oooo:                  Kernel: 6.9.2-arch1-1 
>                `+oooooo:                 Uptime: 5 hours, 59 mins 
>                -+oooooo+:                Packages: 952 (pacman) 
>              `/:-:++oooo+:               Shell: bash 5.2.26 
>             `/++++/+++++++:              Resolution: 2560x1440, 1920x1080 
>            `/++++++++++++++:             DE: GNOME 46.2 
>           `/+++ooooooooooooo/`           WM: Mutter 
>          ./ooosssso++osssssso+`          WM Theme: Adwaita 
>         .oossssso-````/ossssss+`         Theme: Adwaita [GTK2/3] 
>        -osssssso.      :ssssssso.        Icons: Adwaita [GTK2/3] 
>       :osssssss/        osssso+++.       Terminal: kgx 
>      /ossssssss/        +ssssooo/-       CPU: AMD Ryzen 9 5950X (32) @ 3.400GHz 
>    `/ossssso+/:-        -:/+osssso+-     GPU: NVIDIA GeForce RTX 3090 
>   `+sso+:-`                 `.-/+oso:    Memory: 9825MiB / 15894MiB 
>  `++:.                           `-/+/
>  .`                                 `/
> ```

Hi @Zaggy1024 
Acknowledged the issue, we have filed a bug 4682501 internally for tracking purpose.

amrit1711, 2024-06-04:
Hi @urbenlegend 
Acknowledged the crash on suspend issue, we have filed a bug 4683310 internally for tracking purpose.

amrit1711, 2024-06-05:
Hi @Zaggy1024 @urbenlegend 
Request you to share please share bug report from repro state.

urbenlegend, 2024-06-06:
> Hi @Zaggy1024 @urbenlegend Request you to share please share bug report from repro state.

The system is unresponsive when it happens. Is it okay to run it after a reboot?

amrit1711, 2024-06-06:
You can configure your system to preserve system logs after a reboot by editing systemd-journald configuration.
And then once system is rebooted, please capture bug report and share with us.

urbenlegend, 2024-06-06:
I think my logs are preserved after reboot already by default. I just wasn't sure if nvidia-bug-report.sh would grab logs from previous boot or not. I'll run it when I run into the issue again. It is not reproducible every time.

Zaggy1024, 2024-06-06:
> Hi @Zaggy1024 @urbenlegend Request you to share please share bug report from repro state.

Since I posted the comment, I switched back to the stable drivers and also changed to SDDM and KDE Plasma with Wayland.

In order to get the info you need, I tried switching back to beta drivers, GDM and GNOME, but I was unable to reproduce the issue again. I think it's possible that I had mismatched package versions on my first install of the beta drivers, which might explain the artifacts?

navr32, 2024-06-07:
Hello all.
So i have the same issue.
Manjaro linux gnome 
linux 6.8.11-1-MANJARO #1 SMP PREEMPT_DYNAMIC Mon May 27 03:48:16 UTC 2024 x86_64 GNU/Linux

The sleep is ok several time and sometime wake-up give black screen ..the usb is blocked too. So i can log by ssh to the computer.
If i try nvidia-smi 
```

Fri Jun  7 11:05:53 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.78                 Driver Version: 550.78         CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3090        On  |   00000000:0F:00.0 Off |                  N/A |
|ERR!   41C    P8             N/A /  N/A  |   20593MiB /  24576MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   1  NVIDIA GeForce RTX 3090        On  |   00000000:42:00.0 Off |                  N/A |
|ERR!   37C    P8             N/A /  N/A  |   20450MiB /  24576MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      1509      G   /usr/lib/Xorg                                 110MiB |
|    0   N/A  N/A      1595      G   /usr/bin/gnome-shell                           11MiB |
|    0   N/A  N/A      2121      G   /usr/lib/firefox/firefox                       11MiB |
|    0   N/A  N/A      7235      G   /usr/bin/gnome-control-center                  13MiB |
|    0   N/A  N/A      7393      C   ...unners/cuda_v12/ollama_llama_server      20424MiB |
|    1   N/A  N/A      1509      G   /usr/lib/Xorg                                   4MiB |
|    1   N/A  N/A      7393      C   ...unners/cuda_v12/ollama_llama_server      20434MiB |
+-----------------------------------------------------------------------------------------+
WARNING: infoROM is corrupted at gpu 0000:0F:00.0
WARNING: infoROM is corrupted at gpu 0000:42:00.0

```

On the dmesg when suspend i have 

```

random: crng reseeded on system resumption
[17429.825720] PM: suspend exit
[17430.276604] nvidia-modeset: ERROR: GPU:0: Unsupported base display class
[17430.276612] nvidia-modeset: ERROR: GPU:0: Failed to allocate display engine base channels
[17430.280264] nvidia-modeset: ERROR: GPU:0: Unsupported base display class
[17430.280269] nvidia-modeset: ERROR: GPU:0: Failed to allocate display engine base channels
[17430.284096] nvidia-modeset: ERROR: GPU:1: Unsupported base display class
[17430.284101] nvidia-modeset: ERROR: GPU:1: Failed to allocate display engine base channels
[17430.287402] nvidia-modeset: ERROR: GPU:1: Unsupported base display class
[17430.287408] nvidia-modeset: ERROR: GPU:1: Failed to allocate display engine base channels
[17430.307649] rfkill: input handler disabled

```

When trying nvidia-bug-report.sh it start but never ended to give the bug report. the computer always run in this bad state. Because if i want force the computer to go to sleep or shutdown this impossible. The computer says it have always a suspend or shutdown that have been started but any success to go to.
So i have to force the shutdown manually and restart.
The problem occurs more often if the memorys of the gpu have many gb use...

the attempt to nvidia-bug-report 
```

nvidia-bug-report.sh will now collect information about your
system and create the file 'nvidia-bug-report.log.gz' in the current
directory.  It may take several seconds to run.  In some
cases, it may hang trying to capture data generated dynamically
by the Linux kernel and/or the NVIDIA kernel module.  While
the bug report log file will be incomplete if this happens, it
may still contain enough data to diagnose your problem.

If nvidia-bug-report.sh hangs, consider running with the --safe-mode
and --extra-system-data command line arguments.

Please include the 'nvidia-bug-report.log.gz' log file when reporting
your bug via the NVIDIA Linux forum (see forums.developer.nvidia.com)
or by sending email to 'linux-bugs@nvidia.com'.

By delivering 'nvidia-bug-report.log.gz' to NVIDIA, you acknowledge
and agree that personal information may inadvertently be included in
the output.  Notwithstanding the foregoing, NVIDIA will use the
output only for the purpose of investigating your reported issue.

Running nvidia-bug-report.sh...

```

navr32, 2024-06-07:
Ok finally i have found if aborted nvidia-bug-report.sh with ctrl-c
it output something very long ..i want to post here but the file of 2mo is always rejected ! where i can post the file to you ?

urbenlegend, 2024-06-08:
> The sleep is ok several time and sometime wake-up give black screen

I sometimes have a black screen issue on wake up, but switching to a different TTY and then switching back to Ctrl-Alt-F1 seems to bring it back.

navr32, 2024-06-08:
Today i have test just load one gpu with 15gb on the vram..
So when wake-up the gpu with memory load is in ERR! state the other one is ok..says nvidia-smi.
```

nvidia-smi                                                                                                                                     ✔  nico@nico-leothespi 
Sat Jun  8 12:57:23 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.78                 Driver Version: 550.78         CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3090        On  |   00000000:0F:00.0  On |                  N/A |
| 30%   42C    P8             14W /  350W |     344MiB /  24576MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   1  NVIDIA GeForce RTX 3090        On  |   00000000:42:00.0 Off |                  N/A |
|ERR!   42C    P8             N/A /  N/A  |   15120MiB /  24576MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      1717      G   /usr/bin/gnome-shell                          125MiB |
|    0   N/A  N/A      2300      G   /usr/lib/firefox/firefox                      126MiB |
|    1   N/A  N/A      3884      C   ...unners/cuda_v12/ollama_llama_server      15104MiB |
+-----------------------------------------------------------------------------------------+
WARNING: infoROM is corrupted at gpu 0000:42:00.0
```

In this case i am abble to connect to by ssh and complete a full nvidia-bug-report.sh  report .

navr32, 2024-06-08:
If i disabled PreserveVideoMemoryAllocations with set =0 The wake up work even if Vram have been many data load ..but the software which was using this memory  crashed when I try to use again after the wake-up. The soft look like to be up but if i ask some works again crash.
And i see too that Cuda become unavailable.
And Cuda stay unavailable  Even if i go again to sleep and wake up..Cuda is crashed and unusable. Nvidia-smi report any error in that case but Cuda lost. I have to full restart to have Cuda come on to life.

But If i stop the software using Cuda with for example the 15GB of Vram before go to sleep. Then the wake up is ok and then Cuda is ok too..So just to think to stop the soft before sleep and restart the Soft after wake-up and Cuda stay available and ok for compute.
But this is very annoying. And i want to have the soft come on to life in any case of use. 

Thanks for investigation about this. Have a nice days.

tob2, 2024-06-10:
**Cross ref:** A **gdm** (Gnome) commit **disables** **Wayland** under some conditions, including when **NVIDIA_PRESERVE_VIDEO_MEMORY_ALLOCATIONS is 0** (i.e. '!= 1'

→ _**"data: Disable wayland on nvidia if suspend is broken"**_ (committed March 4, 2022),
https://gitlab.gnome.org/GNOME/gdm/-/commit/51181871e9db716546e9593216220389de0d8b03

amrit1711, 2024-06-10:
Hi @Zaggy1024 
It doesn't seem that package mismatch should lead to artifacts issue.
However, I will keep the bug status open internally and shall wait for bug report from repro state.

amrit1711, 2024-07-02:
Hi @navr32 
I ran few instances of benchmarks which consumes almost complete VRAM but did not observe repro after spending 40 cycles of sleep/resume operation.
Could you please confirm repro frequency and also if issue is only reproducible after running cuda application.

amrit1711, 2024-07-17:
Hi @navr32 
Please confirm if you are still seeing the issue, if yes, please help to share information as requested earlier.

amrit1711, 2024-08-01:
Hi @navr32 
Please confirm if you are still seeing the issue with 560 beta release driver, if yes, please help to share information as requested earlier.

veldenb, 2024-08-01:
I'm running the open drivers (555 and 560) for a while now, and for me these versions fixed the suspend problems on my 3080. Video memory is preserved and restored without corruption.  See also https://github.com/NVIDIA/open-gpu-kernel-modules/issues/472#issuecomment-1850949675.

ElDavoo, 2024-08-02:
I think it's a separate issue, but I have optimus (dynamically shutting down the card) and i've found out that sometimes it stays on after resuming from standby

moritzschaefer, 2024-08-07:
@aritger could you report back how internal nvidia debug tracking is going on this one please?

Fxzzi, 2024-08-07:
The latest open drivers are 560.31.02. Please try those.

aritger, 2024-08-07:
I'm sorry, a fix isn't available, yet, but we are making progress.

moritzschaefer, 2024-08-07:
@aritger thanks for reporting back! Please keep us in the loop here.

Best,
Moritz

navr32, 2024-08-07:
Excuse me to not reply but i am on Manjaro and the 555 and 560 are always in the  unstable channel..so i don't think this is a good idea to test with it. I prefers to wait they validate it to be put in stable channel.
Tell me if you prefers i test even if unstable on Manjaro distrib.

Fxzzi, 2024-08-07:
The purpose of Nvidia even releasing the 555 and 560 betas are for people to test them.

David7ce, 2024-08-08:
I move from the open-source drivers to the proprietary more than one year ago because of the suspend issue https://github.com/NVIDIA/open-gpu-kernel-modules/issues/472#issuecomment-1465059264.
And now after updates does not even work on the proprietary drivers side, I don't know exactly if it is caused by drivers or another thing...

v1993, 2024-09-01:
My 2c on this: hibernation is broken (same error message as in the og post), but suspend (s0ix) works just fine, seemingly without any issues. Tested with `NVreg_PreserveVideoMemoryAllocations=1` and `NVreg_PreserveVideoMemoryAllocations=1 NVreg_EnableS0ixPowerManagement=1`.

damhiya, 2024-10-19:
I use 3050 Ti Mobile with 560.35.03 driver. Suspend became more stable than before, but it still doesn't work when video cable (specifically, I use DP over USB-C) is attached. Suspend works fine if I disconnect the cable before closing the lid.

amrit1711, 2024-10-21:
> Excuse me to not reply but i am on Manjaro and the 555 and 560 are always in the unstable channel..so i don't think this is a good idea to test with it. I prefers to wait they validate it to be put in stable channel. Tell me if you prefers i test even if unstable on Manjaro distrib.

Please help to confirm if you are still observing this issue with our latest release driver.

[94 more replies]