NVIDIA/open-gpu-kernel-modules #480: Hibernate/resume doesn't work.

shuox, 2023-03-28:
### NVIDIA Open GPU Kernel Modules Version

530.41.03

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Ubuntu 18.04.5 LTS

### Kernel Release

4.15.0-147-generic #151-Ubuntu SMP Fri Jun 18 19:21:19 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

### Hardware: GPU

NVIDIA A10

### Describe the bug

Try to use hibernate and resume but failed.

### To Reproduce

The steps to reproduce the issue:
ENV: 
/etc/modprobe.d/nvidia.conf:
options nvidia NVreg_PreserveVideoMemoryAllocations=1
options nvidia NVreg_EnableGpuFirmware=0

1. run a tensor2tensor train workload in a docker.
2. echo hibernate > /proc/driver/nvidia/suspend
3. echo 1 > /sys/devices/pci0000\:fe/0000\:fe\:00.0/0000\:ff\:00.0/reset
4. echo resume > /proc/driver/nvidia/suspend

Where 0000:ff:00.0 is the GPU pci device. 

The result: The tensor2tensor process stucks.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

no need bug report. Always reproduce.

dylanmtaylor, 2023-05-03:
I think this is a duplicate of #472.

amrit1711, 2023-08-29:
This issue seems to be duplicate of #472 and it would be good to update further progress in a single thread for better tracking purpose.
I will go ahead and close this thread now.

closed 2023-08-29 by amrit1711