NVIDIA/open-gpu-kernel-modules #483: Unable to change power limit with nvidia-smi

machinedgod, 2023-04-01:
### NVIDIA Open GPU Kernel Modules Version

530.41.03-1

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

Linux 6.2.8-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 22 Mar 2023 22:52:35 +0000 x86_64 GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 2060 (UUID: GPU-2f685ce6-33f4-db75-ce05-81d1723a6ddb)

### Describe the bug

Before recent update, I was able to execute:
```~$ sudo nvidia-smi --power-limit 60```

and have it work as expected. 

After update, this is the output:
```
Changing power management limit is not supported for GPU: 00000000:01:00.0.
Treating as warning and moving on.
All done.
```

Changing power limits caused observable changes both in temperature and in performance, so I am pretty sure my GPU supports it.

For context:
I found that default power limit of 80W tends to heat up the GPU enough that it starts throttling itself and cause stuttering - 60W seemed to work perfectly and keep it under 63C during everything, without boosting my fan. The computer itself is a laptop (which explains issues with heat dissipation), a Lenovo Legion Y740, and I have a pretty good cooling pad to help out.

### To Reproduce

```~# nvidia-smi --power-limit 60```

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11129980/nvidia-bug-report.log.gz)

### More Info

Just a bit more context:
you may notice that in my Xorg config, I use option to skip EDID check for HDMI-0 output - the reason why, is because either driver doesn't recognize my (about 7y old) monitor, or the checksum that monitor outputs is invalid, and then it wouldn't let me use FullHD resolution on that monitor.

No idea if this is in any way related (I presume its) not, but this setup worked for over 5-6 months, so I doubt its related.

ValZapod, 2023-04-02:
>the checksum that monitor outputs is invalid,

That is linux kernel problem, it checks the checksum and do not care.

machinedgod, 2023-04-02:
Thanks, I didn't know that, but... luckily its fixed with this Xorg option.

kleidiss, 2023-04-02:
I also cannot change the power limit with nvidia-smi on this driver while i can on 525 series (I have a RTX 3050 laptop)

On 530 it says "Changing power limit isnot supported for this GPU)

So which is intended behavior

dylif, 2023-04-05:
Relevant discussion [here](https://forums.developer.nvidia.com/t/has-anyone-been-able-to-run-an-rtx-3060-laptop-gpu-at-more-than-80w-on-linux/192959/99)

machinedgod, 2023-04-06:
Oh so its a wide-reaching problem, not just for 2060 and 3050.
I suppose visibility on it is nice enough, so should probably be fixed soonish.

msmafra, 2023-04-10:
With 530.30.02 (beta) the power limits were working fine for the first time on my laptop with a RTX 2060 Mobile 6GB. It broke after 530.42.03 update.

xiguayuyichao, 2023-04-13:
Using the 525 version of the driver, my RTX3060 is able to limit power for the first time, but since updating the 530 version of the driver, it has been impossible to limit power

kleidiss, 2023-04-15:
If anybody here is using a lenovo gaming laptop with the 530 drivers
Can anyone confirm this other power limit issue https://github.com/NVIDIA/open-gpu-kernel-modules/issues/492

barokvanzieks, 2023-04-26:
got same issue

barokvanzieks, 2023-04-26:
on a 3060 AMD legion 5

kraskden, 2023-06-03:
Legion S7 15ACH6  / AMD Ryzen 7 5800H / NVIDIA GeForce RTX 3060

Same issue

barokvanzieks, 2023-06-03:
driver version?

> Legion S7 15ACH6 / AMD Ryzen 7 5800H / NVIDIA GeForce RTX 3060
> 
> Same issue

kraskden, 2023-06-03:
> driver version?

530.41.03-15

barokvanzieks, 2023-06-03:
> > driver version?
> 
> 530.41.03-15

try 535 beta

barokvanzieks, 2023-06-03:
unless it doesn't exist for okm then test proprietary

kleidiss, 2023-06-03:
Still have the issue on 535 BETA

barokvanzieks, 2023-06-03:
hmmm

barokvanzieks, 2023-06-03:
test with 525 branch (prop, or okm if exists)

arutar, 2023-06-04:
Ryzen 5 5600H / NVIDIA GeForce RTX 3060
Same issue

barokvanzieks, 2023-06-04:
> Ryzen 5 5600H / NVIDIA GeForce RTX 3060 Laptop Same issue on version 525.78.01 everything works fine

did you test 535 (prop)

arutar, 2023-06-06:
@barokvanzieks Yesterday I checked the work of the power limit on many versions of the drivers. 
(command: `nvidia-smi -pl 50`)

- 525.78.01 power limit works fine
- 525.85.05 power limit works fine
- 530.41.03 power limit is NOT WORKING

barokvanzieks, 2023-06-06:
hmmmmm ok i havent tested 535 myself, guess ill stay on 525 for now

Diafwl, 2023-06-15:
i checked latest version of nvidia driver "535.54.03" with dkms and still NOT WORKING 

(i5-12500H / NVIDIA GeForce RTX 3060 Laptop / EndeavourOS)

kleidiss, 2023-06-15:
I think this is intended at this point

They also unlocked it on windows around driver 528 and locked it back shortly after so maybe this issue should be closed?

arutar, 2023-06-16:
@kleidiss This is a very important opportunity. Laptops get very hot and drain the battery quickly. This feature is required!

machinedgod, 2023-06-16:
I agree. It would be worthwhile checking if anyone diffed the sources and figured out how to make a patch to enable it back. If not, I will attempt that this weekend, because right now - my laptop goes to 80-81C (with a REALLY good cooling pad)... previously I kept the framerate stable at 55C

notnotme, 2023-06-17:
I just noticed it did not work anymore on my system, reverting back to 525.105.17 and everythings good. Thanks you

youmukonpaku1337, 2023-06-18:
> I agree. It would be worthwhile checking if anyone diffed the sources and figured out how to make a patch to enable it back. If not, I will attempt that this weekend, because right now - my laptop goes to 80-81C (with a REALLY good cooling pad)... previously I kept the framerate stable at 55C

thatd be super cool of you

youmukonpaku1337, 2023-06-18:
but yes 535 isnt workinf for me either and its a pain in the arse since now i have to fuck with drivers to get OpenCL working

youmukonpaku1337, 2023-06-18:
this feature is essential for me since i can up wattage by like 50 watts from stock 80

youmukonpaku1337, 2023-06-20:
> I agree. It would be worthwhile checking if anyone diffed the sources and figured out how to make a patch to enable it back. If not, I will attempt that this weekend, because right now - my laptop goes to 80-81C (with a REALLY good cooling pad)... previously I kept the framerate stable at 55C

question is which file would it be in...

darxkies, 2023-06-26:
Any updates? When is it going to work again?

youmukonpaku1337, 2023-06-26:
@machinedgod you were going to make a patch, no?

youmukonpaku1337, 2023-06-26:
> Any updates? When is it going to work again?

535 was supposed to fix it but it didnt so nobody knows

arutar, 2023-06-26:
I have bad news, I hope this bug will be fixed in the future.
[image]

youmukonpaku1337, 2023-06-26:
fucking hell, guess we'll need a patch of our own

machinedgod, 2023-06-26:
> @machinedgod you were going to make a patch, no?

I was, but weekend had its own challenges, unfortunately this - as much as its a priority for me - is not high enough priority to override everything else.

I'll have to cram it into my schedule when I get some breathing space, sorry gents.

darxkies, 2023-06-27:
It's summer and laptops get hotter too. Being able to tweak the temperature of the GPUs is a necessity.

kuriot, 2023-07-01:
And what about gaming laptops which change power limits on Windows in gaming mode and now stuck with low power limit on Linux? Ah, I would like to give one famous quote about NVIDIA from one Finnish guy called Linus, but I won't.

arutar, 2023-07-01:
@kuriot On windows power limit change is also not supported... All version 53x.xx can not change power limit on laptops

kuriot, 2023-07-01:
@arutar I hope it's just a bug and that answer that it's by design on Nvidia forum was just a mistake. But I think they made it on purpose.

kraskden, 2023-07-02:
> And what about gaming laptops which change power limits on Windows in gaming mode and now stuck with low power limit on Linux?

Try to enable nvidia-powerd service, that's [enable dynamic boost](https://download.nvidia.com/XFree86/Linux-x86_64/535.54.03/README/dynamicboost.html)

So power limit is automatically changed on high load. I tested it with Superposition benchmark: on system startup nvidia-smi show 60W power limit, in the benchmark nvidia-smi show ~90W usage (maybe will be 100W, I just tested ~10 seconds in the bench)

kuriot, 2023-07-03:
> > And what about gaming laptops which change power limits on Windows in gaming mode and now stuck with low power limit on Linux?
> 
> Try to enable nvidia-powerd service, that's [enable dynamic boost](https://download.nvidia.com/XFree86/Linux-x86_64/535.54.03/README/dynamicboost.html)
> 
> So power limit is automatically changed on high load. I tested it with Superposition benchmark: on system startup nvidia-smi show 60W power limit, in the benchmark nvidia-smi show ~90W usage (maybe will be 100W, I just tested ~10 seconds in the bench)
> 
> P.S. My config - 3060 RTX (laptop, 100W) + nvidia 535.54.03

No, it doesn't matter if service is running or no.

arutar, 2023-07-03:
@kraskden The question is to set the upper limit of power, and not rely on automatic mode. You need the ability to specify the limits of power change that will be less than the default ones (from the firmware) (Default 0 - 115) (user 0 - 50) `nvidia-smi pl 50`

kraskden, 2023-07-03:
> @kraskden The question is to set the upper limit of power

OMG, I replied to specific question and suggest an option to fix problem with low power limit on gaming laptops.

kraskden, 2023-07-03:
> No, it doesn't matter if the service is running or not.

Really? I made tests, without powerd service my GPU stuck at 60W power.

kuriot, 2023-07-03:
> > No, it doesn't matter if the service is running or not.
> 
> Really? I made tests, without powerd service my GPU stuck at 60W power.

My laptop by default has limit of 115W. Before driver update I could change it up to 140W (it's the true limit) or lower it at night so that laptop worked quieter. Then they added nvidia-powerd.service and limit setting worked only when this service is turned off. Now it doesn't work at all. If the service is turned on, no matter what it doesn't go higher than 115W. And these external 15-25 watts gave me a good FPS boost in heavy games.

kuriot, 2023-07-08:
> Try to enable nvidia-powerd service, that's [enable dynamic boost](https://download.nvidia.com/XFree86/Linux-x86_64/535.54.03/README/dynamicboost.html)
>
> So power limit is automatically changed on [...]

You were right. On my Lenovo Legion 5 performance modes are switched by Fn+Q and I never bothered to switch them on Linux. But yes, now to use higher power I need to switch the laptop to performance mode with these keys (and it's not connected to system, I guess it's some Lenovo thing hardcoded into BIOS) and with running `nvidia-powerd.service` I get 140W limit now.

EKami, 2023-07-24:
I’m in the same boat, I can’t set my Zephyrus duo GX650PY with an RTX 4090 to anything above 115W either. In the past I could set it to 175W. Now the same has happened on Windows with GPU drivers above 536.23 locking the card to 115W. I ran Timespy on 536.23 and got a GPU score of 21 437. The same benchmark on driver 536.67 gives a score of 17 693… Did anyone find a solution other than rolling back the drivers?

jcbritobr, 2023-08-07:
nvidia gtx 1050 ti this dont work in proprietary also. If anyone knows how to save power in gtx 1050 ti please write here.

Slava0135, 2023-08-17:
same with GeForce MX250, wanted to make it cooler but I guess no such option now

Osc100, 2023-08-22:
Same issue here with propietary drivers 535.98 on RTX 4060 Max-Q

Panzerwaffe, 2023-09-26:
The same problem. 
Power limit is stuck at 35W in Linux, but in Windows works perfectly with 50W+15W turbo (65W as designed). 

Ubuntu 23.04, kernel 6.4.11, driver 535.86.05.
Asus ROG Flow Z13 2023 (RTX 4050 mobile)

xAlpharax, 2023-09-26:
> The same problem. 
> Power limit is stuck at 35W in Linux, but in Windows works perfectly with 50W+15W turbo (65W as designed). 
> 
> Ubuntu 23.04, kernel 6.4.11, driver 535.86.05.
> Asus ROG Flow Z13 2023 (RTX 4050 mobile)

I've managed to solve this problem by simply enabling the system service/daemon at startup named `nvpower`

Depending on your distro your mileage may vary.

jcbritobr, 2023-09-26:
> I've managed to solve this problem by simply enabling the system service/daemon at startup named `nvpower`
>
> Depending on your distro your mileage may vary.
I needed to disable this service because in fedora it didn't worked.

yochananmarqos, 2023-09-26:
> I've managed to solve this problem by simply enabling the system service/daemon at startup named nvpower

@xAlpharax What package is this a part of on what distro?

xAlpharax, 2023-09-26:
> The same problem.
> Power limit is stuck at 35W in Linux, but in Windows works perfectly with 50W+15W turbo (65W as designed).
> Ubuntu 23.04, kernel 6.4.11, driver 535.86.05.
> Asus ROG Flow Z13 2023 [...]

Specifically Fedora with Wayland?

xAlpharax, 2023-09-26:
> @xAlpharax What package is this a part of on what distro?

Void Linux shipped this service daemon with its package of the nvidia drivers.

yochananmarqos, 2023-09-27:
@xAlpharax They do, eh? :thinking: It appears they [package the driver](https://github.com/void-linux/void-packages/blob/master/srcpkgs/nvidia/template) similar to just about any other distro. They must have a separate package, but I can't find it.

xAlpharax, 2023-09-27:
@yochananmarqos Looking at the building process it so seems. I will try finding that out as well.

[101 more replies]