NVIDIA/open-gpu-kernel-modules #484: 8K display crashing nvidia GPU, blank screen

dllu, 2023-04-01:
### NVIDIA Open GPU Kernel Modules Version

530.41.03

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

6.2.8-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 22 Mar 2023 22:52:35 +0000 x86_64 GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3090 (UUID: GPU-80b59546-cdef-7dda-b99f-e4e532d3726d)

### Describe the bug

Upon attempting to start an x server with `startx`, the display is blank (it says "Check Device Power")

When I try to SSH into the machine, the following lines are in the `dmesg`.

```
[   23.138228] nvidia-modeset: WARNING: GPU:0: Lost display notification (0:0x00000000); continuing.
[   39.156353] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD data byte 12
[  158.666001] nvidia-modeset: ERROR: GPU:0: Idling display engine timed out: 0x0000c67d:0:0:1051
[  166.672768] nvidia-modeset: ERROR: GPU:0: Idling display engine timed out: 0x0000c67e:0:0:1060
[  174.679521] nvidia-modeset: ERROR: GPU:0: Idling display engine timed out: 0x0000c67e:1:0:1060
[  198.699702] nvidia-modeset: ERROR: GPU:0: Idling display engine timed out: 0x0000c67e:6:0:1060
[  206.706414] nvidia-modeset: ERROR: GPU:0: Idling display engine timed out: 0x0000c67e:7:0:1060
[  214.713127] nvidia-modeset: ERROR: GPU:0: Idling display engine timed out: 0x0000c67d:0:0:1051
[  222.719824] nvidia-modeset: ERROR: GPU:0: Idling display engine timed out: 0x0000c67e:0:0:1060
[  230.726524] nvidia-modeset: ERROR: GPU:0: Idling display engine timed out: 0x0000c67e:1:0:1060
[  254.746614] nvidia-modeset: ERROR: GPU:0: Idling display engine timed out: 0x0000c67e:6:0:1060
[  262.753310] nvidia-modeset: ERROR: GPU:0: Idling display engine timed out: 0x0000c67e:7:0:1060
```

When running `nvidia-smi`, it hangs for a long time, during which the following lines are printed in `dmesg`:
```
[  479.022059] nvidia-modeset: ERROR: GPU:0: Display engine push buffer channel allocation failed: 0x65 (Call timed out [NV_ERR_TIMEOUT])
[  479.022626] nvidia-modeset: ERROR: GPU:0: Failed to allocate display engine core DMA push buffer
[  491.488862] INFO: task nvidia-modeset/:285 blocked for more than 122 seconds.
[  491.488866]       Tainted: P           OE      6.2.8-arch1-1 #1
[  491.488868] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  491.488869] task:nvidia-modeset/ state:D stack:0     pid:285   ppid:2      flags:0x00004000
[  491.488874] Call Trace:
[  491.488875]  <TASK>
[  491.488877]  __schedule+0x3c8/0x12e0
[  491.488883]  ? __schedule+0x3d0/0x12e0
[  491.488888]  schedule+0x5e/0xd0
[  491.488890]  schedule_timeout+0x151/0x160
[  491.488895]  __down_common+0x11a/0x230
[  491.488899]  down+0x47/0x60
[  491.488902]  nvkms_kthread_q_callback+0x9b/0x130 [nvidia_modeset 9bc24184cede2bad48d296aa366c1be1adaa2d72]
[  491.488943]  _main_loop+0x93/0x150 [nvidia_modeset 9bc24184cede2bad48d296aa366c1be1adaa2d72]
[  491.488982]  ? __pfx__main_loop+0x10/0x10 [nvidia_modeset 9bc24184cede2bad48d296aa366c1be1adaa2d72]
[  491.489023]  kthread+0xde/0x110
[  491.489026]  ? __pfx_kthread+0x10/0x10
[  491.489029]  ret_from_fork+0x2c/0x50
[  491.489035]  </TASK>
```
However, after a few minutes, `nvidia-smi` successfully runs.

Running `nvidia-bug-report.sh` also hangs, during which the following are printed out in `dmesg`:

```
[... 96 lines cut ...]
Validated By    : Signature
```

Incidentally, 8K 60 Hz and 4K 120 Hz are both working fine in Windows (it is a dual boot machine).

Prior to upgrading nvidia drivers, 8K 30 Hz and 4K 120 Hz were working fine in Linux with the previous version (525.89.02).

### To Reproduce

1. Connect a PC with a NVIDIA GeForce RTX 3090 to a Samsung QN800A display using a high quality certified HDMI 2.1 cable.
2. Install Arch Linux with nvidia driver version `530.41.03`
3. Set the Samsung QN800A to "Gaming" mode and verify that 8K 60 Hz and 4K 120 Hz are both working in Windows or another machine.
4. place `20-nvidia-settings.conf` in `/usr/share/X11/xorg.conf.d`
[20-nvidia-settings.txt](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11130346/20-nvidia-settings.txt)
5. Upon starting X server with `startx`, a blank screen happens.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

Since `nvidia-bug-report.sh` hangs, I had to run `nvidia-bug-report.sh --safe-mode`

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11130384/nvidia-bug-report.log.gz)

dfadev, 2023-06-09:
QN800C and QN700B also have this issue.  Turning off "game mode" and making sure "Input Signal Plus" is should allow 8k@60 and 4k@119.88 to sync.   Although then they'll fail to recover from a screen saver blanking..

dllu, 2023-06-11:
That is a good hint. After turning off Game Mode, it worked, but after rebooting my computer, it stopped working again. I will debug a little further...

dfadev, 2023-07-15:
535.54.03 seems to be better at syncing 8k@60 and 4k@120 on Samsung TVs but still has issues with DPMS.

This repro's every time for me on a dual TV setup:

1. startx
2. TVs sync (8K@60)
3. hold power button on samsung remote to reboot TVs
4. TVs reboot, fail to sync (No device connected)
5. ALT-CTRL-F2 to switch to text console
6. TVs sync (4k@60)
7. ALT-F1 to switch back to X11
8. screen blanks, kernel hard locks, hard reset cycle required (No device connected displayed on TV)

Also forcing dpms to off `xset dpms force off` and waiting some time before waking will sometimes lock the system requiring a hard reset.

mcsygit, 2023-09-19:
Might have a +1, might be different. RTX 4090, LG C2 TV connected via HDMI 2.1. 4k120hz signal shows up only randomly. Problem's only on Linux but on both Wayland and X11. I have tried unplugging eARC soundbar, turning off VRR, no difference. 4k60Hz works always, but nothing above it consistenly. I did also try changing HDMI 2.1 cable.

Samsung LS28AG700N as secondary monitor works regardless using Displayport (4k144hz IPS), but I did try using just C2. C2 would lose VRR support if active DP adapter were to be used. 

I would blame TV itself if it didn't work just fine on Windows.

Tried drivers from 525 to latest 535. No difference.

I know this is open source driver repo but something like this goes hand in hand with proprietary driver, right? I did only try proprietary driver.

ZombieLurker, 2024-06-26:
This issue still happens with driver 550.90.07. All monitors and Samsung 4K 120Hz tv cuts signal after logging into Plasma 6.1 Wayland and 6.9.6-arch1-1. Unplugging the hdmi to the TV all of a sudden brings signal back to the other two monitors. What is weird, I didn't have this issue for a couple weeks but then all of a sudden after updating to KDE Plasma 6.1 and rebooting, this happened again.