NVIDIA/open-gpu-kernel-modules #485: G-SYNC NOT WORKING PROPERLY ON WAYLAND

Racsuu, 2023-04-03:
### NVIDIA Open GPU Kernel Modules Version

530.41.03

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Fedora 37 - Gnome 43/KDE

### Kernel Release

6.2.8-300.fc38.x86_64

### Hardware: GPU

NVIDIA GeForce GTX 1650 SUPER (UUID: GPU-1b82f730-b99a-4e6a-32b8-a1f703cb5034)

### Describe the bug

Nvidia on wayland is working really fine right now, but I insist, we really need g-sync (vrr) to work on wayland, mostly on low end cards that can't keep high frame rates always, the experience in games is horrible when you get lower fps that your monitor hz.

(option to enable vrr is showing on gnome or kde control center but is not working properly, the monitor just lower randomly and inacurrate the hz on some moments, nothing that help, atleast we are somehow close)

### To Reproduce

-Have a g-sync (or g-sync compatible) monitor, with a nvidia card, use the drivers for linux
-Install any GNU/LINUX distro, with a desktop that have wayland working (example:gnome, kde)
-Patch gnome with vrr if is necessary, on kde is enabled by default, go to monitor settings and enable vrr.
-Test vrr on a game or on a special test for it

Results: vrr (g-sync) is not working yet

### Bug Incidence

Always

### nvidia-bug-report.log.gz

..

gardotd426, 2023-04-05:
I don't believe this is a bug, I believe the option to enable it in KDE/patched Mutter is the bug. I've yet to see any release notes/official Nvidia statements saying that they've enabled Wayland VRR. I think they enabled some of the initial steps and that is why Plasma thinks that it's supported, but it's still not actually supported yet. 

But whether that's true or not, I'm not sure if the open GPU modules even support VRR at all, and they definitely don't support your GPU, therefore this bug report is 100% unequivocally in the wrong place. This GitHub repo issue report thread is for issues that are specific to the Open kernel modules, and yours is not, therefore you need to post this on the Nvidia developer forum

eszlari, 2023-04-10:
https://forums.developer.nvidia.com/t/nvidia-bug-kde-wayland-games-are-force-vsynced/237880/8

> G-Sync should be working with native Wayland applications. With Xwayland applications I suspect the main issue is the lack of direct scan-out support.

closed 2023-06-26 by Racsuu

Mithras, 2023-06-27:
I'm yet to see any proof g-sync is working on Wayland. I would test it myself but I don't know/have any Wayland native game or vrr test.

gardotd426, 2023-06-27:
I was told that it is in fact working by kodatarule over on the developer forum after I explicitly asked him to check for me, since I don't use Wayland.

Racsuu, 2023-06-27:
I test it only on xwayland apps, gnome 44 with the patch for vrr.

The option is there, but the monitor seems to try to adapt the hz to the fps in a random way, is not working properly.

gardotd426, 2023-06-27:
It doesn't work for XWayland, and isn't advertised to work in XWayland. It's only supposed to work in native Wayland apps right now.

Racsuu, 2023-06-27:
https://forums.developer.nvidia.com/t/feature-g-sync-freesync-under-wayland-session/220822/18

> VRR doesn’t seem to work on both native wayland and xwayland applications, I did report such behavior + tried to debug, but didn’t seem to get anything useful.
With 530 driver I can no longer even boot into any wayland session(sway,hyprland, kde, gnome) and I’m not the only one affected.

is just not working, not a surprise anyways..

gardotd426, 2023-06-27:
You just didn't continue reading, did you?

Same person right after that comment:

>With single monitor under KDE Wayland, G-Sync works for any fullscreen app → regardless if it is native wayland or xwayland, it just works!
>Now when my 2nd monitor comes back I’ll be able to test it out with multiple and update if it still kicks in.

Racsuu, 2023-06-27:
You just didn't continue reading, did you?

Same person right after that comment:

> Just a quick update here: not really sure if it works correctly, sometimes it seems to smooth out other times not so much and the monitor frame counter remains at the refresh rate, it doesn’t change so I don’t think it is entirely fixed.

> I did some tests and g-sync doesn’t appear to be working just yet on multi monitors on wayland, not sure what is causing this.

More comments down below:

> It doesn’t work for a single monitor either (at least not for XWayland):
[https://www.reddit.com/r/linux_gaming/comments/1498gtl/comment/joaxbze 26](https://www.reddit.com/r/linux_gaming/comments/1498gtl/comment/joaxbze)

Don't be a jerk.

Been testing this for MONTHS on every driver update, with every gnome or kde update in wayland.

VRR is not working properly for absolute anything, is choppy, doesn't feel okay, and the monitor hz just don't match the fps like it does on both xorg and windows, on one, two or more monitors is the exact same situation, going to stop answering about this here since is kinda pointless too, hopefully nvidia will do a decent update for wayland support someday and vrr will be a thing on it.

GrabbenD, 2023-07-01:
Solved this issue and many more by switching to a AMD GPU. I wish I did it sooner.. Don't waste your time with a company that doesn't properly support your platform @Racsuu

gardotd426, 2023-07-01:
> Solved this issue and many more by switching to a AMD GPU. I wish I did it sooner.. Don't waste your time with a company that doesn't properly support your platform @Racsuu

You mean like how AMD refuses to support HDMI 2.1 on Linux (but do on Windows) and also don't support overclocking or fan control on RDNA 3 on Linux (but do on Windows)? This is an issue thread for actual issues. Take your gripes to reddit, please.

GrabbenD, 2023-07-01:
Sure @gardotd426, AMD would have to distribute support for that through closed binary blobs like NVIDIA does due to licensing of HDMI.