NVIDIA/open-gpu-kernel-modules #488: Completely random locked 60hz refresh rate on boot

Fxzzi, 2023-04-11:
### NVIDIA Open GPU Kernel Modules Version

525.89.02

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

Linux Kappa 6.2.10-zen1-1-zen

### Hardware: GPU

NVIDIA GeForce RTX 3070

### Describe the bug

On some random boots, my main monitor with a refresh rate of 170Hz is instead limited to 60Hz only. This occurs in both X11 and Wayland as I experience this issue completely randomly on both bpswm and hyprland.

I am using the 525.89.02 release instead of the latest 530 as on the 530 drivers I experience [this issue](https://forums.developer.nvidia.com/t/cant-boot-into-kde-wayland-session-with-530-30-02/244559?u=faaris.ansari), stopping me from using the driver. When I was using the 530 driver on a lower refresh rate (120Hz to be exact), I still experienced the same issue.

I have a feeling this is related to this error in `journalctl -b`:
```
Apr 11 12:18:06 Kappa kernel: nvidia-modeset: ERROR: Invalid request parameters, planePitch or rmObjectSizeInBytes, passed during surface registration
Apr 11 12:18:06 Kappa kernel: nvidia-modeset: ERROR: Invalid request parameters, planePitch or rmObjectSizeInBytes, passed during surface registration
```

Here is an output of `journalctl -b | grep -i nvidia` as it may be useful for the report:
```
Apr 11 12:18:03 Kappa kernel: Command line: initrd=\amd-ucode.img initrd=\booster-linux-zen.img root=UUID=0e488fe7-cc5a-44c3-8289-588a02ff9dcc rootflags=subvol=@ rw quiet loglevel=3 nvidia-drm.modeset=1 nvidia.NVreg_PreserveVideoMemoryAllocations=1 zswap.enabled=0 mitigations=off amd_pstate=passive
Apr 11 12:18:03 Kappa kernel: Kernel command line: initrd=\amd-ucode.img initrd=\booster-linux-zen.img root=UUID=0e488fe7-cc5a-44c3-8289-588a02ff9dcc rootflags=subvol=@ rw quiet loglevel=3 nvidia-drm.modeset=1 nvidia.NVreg_PreserveVideoMemoryAllocations=1 zswap.enabled=0 mitigations=off amd_pstate=passive
Apr 11 12:18:03 Kappa kernel: nvidia: loading out-of-tree module taints kernel.
Apr 11 12:18:03 Kappa kernel: nvidia: module license 'NVIDIA' taints kernel.
Apr 11 12:18:03 Kappa kernel: nvidia: module verification failed: signature and/or required key missing - tainting kernel
Apr 11 12:18:03 Kappa kernel: nvidia-nvlink: Nvlink Core is being initialized, major device number 235
Apr 11 12:18:03 Kappa kernel: nvidia 0000:07:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
Apr 11 12:18:03 Kappa kernel: NVRM: loading NVIDIA UNIX x86_64 Kernel Module  525.89.02  Wed Feb  1 23:23:25 UTC 2023
Apr 11 12:18:03 Kappa kernel: nvidia_uvm: module uses symbols nvUvmInterfaceDisableAccessCntr from proprietary module nvidia, inheriting taint.
Apr 11 12:18:03 Kappa kernel: nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  525.89.02  Wed Feb  1 23:09:40 UTC 2023
Apr 11 12:18:03 Kappa kernel: [drm] [nvidia-drm] [GPU ID 0x00000700] Loading driver
Apr 11 12:18:03 Kappa kernel: nvidia-uvm: Loaded the UVM driver, major device number 511.
Apr 11 12:18:03 Kappa kernel: [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:07:00.0 on minor 0
Apr 11 12:18:03 Kappa (udev-worker)[330]: nvidia: Process '/usr/bin/bash -c '/usr/bin/mknod -Z -m 666 /dev/nvidiactl c $(grep nvidia-frontend /proc/devices | cut -d \  -f 1) 255'' failed with exit code 1.
Apr 11 12:18:03 Kappa (udev-worker)[330]: nvidia: Process '/usr/bin/bash -c 'for i in $(cat /proc/driver/nvidia/gpus/*/information | grep Minor | cut -d \  -f 4); do /usr/bin/mknod -Z -m 666 /dev/nvidia${i} c $(grep nvidia-frontend /proc/devices | cut -d \  -f 1) ${i}; done'' failed with exit code 1.
Apr 11 12:18:03 Kappa kernel: input: HDA NVidia HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:03.1/0000:07:00.1/sound/card1/input6
Apr 11 12:18:03 Kappa kernel: input: HDA NVidia HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:03.1/0000:07:00.1/sound/card1/input7
Apr 11 12:18:03 Kappa kernel: input: HDA NVidia HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:03.1/0000:07:00.1/sound/card1/input8
Apr 11 12:18:03 Kappa kernel: input: HDA NVidia HDMI/DP,pcm=9 as /devices/pci0000:00/0000:00:03.1/0000:07:00.1/sound/card1/input9
Apr 11 12:18:04 Kappa systemd[1]: Starting NVidia GPU temperature...
Apr 11 12:18:04 Kappa echo[467]: 37000 > /tmp/nvidia-temp
Apr 11 12:18:04 Kappa systemd[1]: Started NVidia GPU temperature.
Apr 11 12:18:04 Kappa fancontrol[510]:   Depends on /tmp/nvidia-temp
Apr 11 12:18:06 Kappa kernel: nvidia-modeset: ERROR: Invalid request parameters, planePitch or rmObjectSizeInBytes, passed during surface registration
Apr 11 12:18:06 Kappa kernel: nvidia-modeset: ERROR: Invalid request parameters, planePitch or rmObjectSizeInBytes, passed during surface registration
```

### To Reproduce

Have a monitor with a high refresh rate.
Reboot.
On some boots, refresh rate will be limited to 60hz whether it be X11 or wayland.

### Bug Incidence

Sometimes

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11200550/nvidia-bug-report.log.gz)

thesword53, 2023-04-12:
> I am using the 525.89.02 release instead of the latest 530 as on the 530 drivers I experience [this [...]

Did you try the 530.41.03 driver instead of the 530.30.02 beta driver?

Fxzzi, 2023-04-12:
> Did you try the 530.41.03 driver instead of the 530.30.02 beta driver?

Tried both and same issue. nixOS and arch

alekulyn, 2023-06-07:
bump. I'm also getting this error on ArchLinux version 530.41.03-16 with an RTX 2070 Super during suspend/resume. It may be a contributing factor to my Wayland compositor (hyprland) crashing during the suspend/resume process, as I see a few drm calls in its stacktrace after the nvidia-modeset error.

abdulAttya, 2024-08-21:
I get this error in fedora linux 555.58.02 with  NVIDIA GeForce GTX 1650
i use wayland too but this happened to me with kde too
sometimes it happens also i boot as i need to replug the cable to make it work :P

fxzzi, 2025-01-20:
This is still happening to me after over a year and a half. Sometimes my display is stuck at 120Hz with some weird colour fringing, and sometimes at 60Hz but the display itself looks OK. It is usually solved by unplugging the power from my monitor and plugging it back in.