NVIDIA/open-gpu-kernel-modules #491: not being able to access powersource correctly

beh-10257, 2023-04-14:
### NVIDIA Open GPU Kernel Modules Version

530.41.03

### Does this happen with the proprietary driver (of the same version) as well?

No

### Operating System and Version

Arch Linux

### Kernel Release

6.2.10-arch1-1

### Hardware: GPU

NVIDIA GeForce GTX 1650 Ti

### Describe the bug

the consequences of the driver not being able to know the power source basically the balanced powermode is broken
a video explaining the problem:
the command in the bottom is the actual battery state and above what the driver sees I guess

https://user-images.githubusercontent.com/77834479/232152297-98864c15-7a63-4067-9df6-6185146736d3.mp4

### To Reproduce

install nvidia-open-dkms 530.41.03-3
and yeah just watch the command in the video and plug and unplug your laptop

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11237054/nvidia-bug-report.log.gz)

### More Info

this bug will result in you playing a game starting it in battery the fps being 100 or something and then driver realizing its on battery fps plumit into 20 nad then you plug in to the wall thinking you'll get those 100 fps again but no the only way to actually get back playing is rebooting your system

+ I guess cuda cores aren't affected since blender increases in speed even after the initial unplug

kleidiss, 2023-04-15:
The power management on these drivers is the worst shit i have ever seen 
Since i game plugged in i didn't notice this one tho but lol

beh-10257, 2023-04-15:
@kleidiss having poor power management is not really a problem to me but the performance staying poor even after plugging the laptop to the wall is a big deal hopefully it will be fixed soon
+ I installed the propriatary driver just to check if this problem exists there and I realized there is not really a big performance hit or anything so yeah I guess when this problem is fixed
I can continue daily driving the open driver

beh-10257, 2023-06-16:
update: same exact problem in 535.54.03

this bug will take ages to be fixed I guess

beh-10257, 2023-06-16:
@amrit1711 since you have already saw #492 is this problem relates will this issue be fixed as well in future releases

beh-10257, 2023-09-26:
well I guess at least in 535.113.01 this issue is fixed

closed 2023-09-26 by beh-10257