NVIDIA/open-gpu-kernel-modules #492: nvidia-powerd.service needs a reboot to change the power limit state  on AMD lenovo gaming laptops

kleidiss, 2023-04-15:
### NVIDIA Open GPU Kernel Modules Version

530.41.03

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Fedora release 37 (Thirty Seven)

### Kernel Release

Linux DESKTOP-MPKTHBJ 6.2.10-200.fc37.x86_64 #1 SMP PREEMPT_DYNAMIC Thu Apr  6 23:30:41 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3050 Laptop GPU (UUID: GPU-b2f720a1-2b4e-9b17-4383-76f9361248a2)

### Describe the bug

nvidia-powerd.service needs a full system reboot to change the power limit of the GPU from 64-66W in the laptop's balanced mode to 85W that the laptops' performance mode should use

If i press fn+Q and change to performance mode without restarting the laptop will still use 64W 
After a system restart the power limit changes to 85W but if i change back to balenced/power saver mode the power limit stays at 85W which is very bad for the  GPU temp and may cause hardware damage 

The problem here is have to reboot each time you need to change the power limit while on Windows it changes dynamically just by pressing FN + Q

### To Reproduce

Change your lenovo legion/ideapad laptop to performance mode from balenced without a system retsrat and check the GPU power draw 

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11239898/nvidia-bug-report.log.gz)

### More Info

Laptop specs:

Lenovo Ideapad Gaming 3 15ACH6
AMD Ryzen  5 5600H  
RTX 3050 Mobile 85W
16 GB RAM & 1 TB SSD M.2

beh-10257, 2023-04-15:
I would guess your problem is the same or similar to my problem
https://github.com/NVIDIA/open-gpu-kernel-modules/issues/491
can you check that like if the driver even realize the powersource change
most likely internally it's the same bug lool

kleidiss, 2023-04-15:
> I would guess your problem is the same or similar to my problem #491 can you check that like if the driver even realize the powersource change most likely internally it's the same bug lool

it says 0 even when the laptop draws 85W

beh-10257, 2023-04-15:
@kleidiss 
btw 0 is plugged 1 is unplugged
just `watch "nvidia-settings -q GPUPowerSource"`
and plug your laptop I guess give it 10 seconds to make sure
unplug give it another 10 seconds if there is no switch in the open driver you basically have the same problem as me

kleidiss, 2023-04-15:
> @kleidiss btw 0 is plugged 1 is unplugged just `watch "nvidia-settings -q GPUPowerSource"` and plug your laptop I guess give it 10 seconds to make sure unplug give it another 10 seconds if there is no switch in the open driver you basically have the same problem as me

Ok so i unplugged it went to 1 and plugged back in it went to 0 again

beh-10257, 2023-04-15:
@kleidiss you used xorg for that test right
+ yeah there is a difference cause the test I told you to do actually works under wayland for some reason
+ if you are using gnome or kde I guess you can switch in the login manager

beh-10257, 2023-04-15:
[update]
since you game me the idea to just the power usage without unplugging anything fps in minecraft will be 120 with shaders and power draw 50 and the mhz 2000
after unplugging fps to 10 and power draw to 15w ( why is that the case for gods sake just give to the gpu 30w at least )
and the mhz to 300 and sometimes 340 
and the mhz and power draw will stay exactly the same even after replugging the laptop

kleidiss, 2023-04-15:
>     * yeah there is a difference cause the test I told you to do actually works under wayland for some reason

Oh i use wayland on fedora
Yeah switching to Xorg i see now but i dont think is related to my issue because i also tested the proprietary drivers and there was no such issue there even on Xorg but my specific issue also happens on proprietary drivers

kleidiss, 2023-04-15:
> [update] since you game me the idea to just the power usage without unplugging anything fps in minecraft will be 120 with shaders and power draw 50 and the mhz 2000 after unplugging fps to 10 and [...]

15 wats lmao , might as well turn off the gpu at that point

beh-10257, 2023-04-15:
- 15w lmao , might as well turn off the gpu at that point
exactly

> Yeah switching to Xorg i see now

at least the problem is not just for me
so yeah I guess your response confirms https://github.com/NVIDIA/open-gpu-kernel-modules/issues/491

> my specific issue also happens on proprietary drivers

two issue related to power in two days that's hilarious

kleidiss, 2023-04-15:
> 
> two issue related to power in two days that's hilarious

Not to mention that on the Unigine Heaven OGL benchmark i get 80-100 points less on Linux while using the same power draw and clocks as Windows (I checked with mangohud and geforce overlay thing on windows) These crappy drivers are also burning power for no reason 

I notice a fps loss on DXVK as well while using these drivers on Linux but thats more acceptable than a native benchmark

kleidiss, 2023-04-17:
Correction: You just need to reload the service for it to update the power and clock speed limit

beh-10257, 2023-04-17:
@kleidiss 
Wait there is a service for that 
What is it
Probably that would solve my problem as well

kleidiss, 2023-04-17:
> @kleidiss Wait there is a service for that What is it Probably that would solve my problem as well

It manages dynamic boost meaning it takes power from the CPU when the CPU isn't using it and gives it to the GPU
I don't think that is related to battery tho

kleidiss, 2023-04-17:
Just reload the service when unplugged and see if the power draw changes
If it does then it's confirmed we have basically the same problem

FiestaLake, 2023-04-20:
Restarting nvidia-powerd service without rebooting causes dgpu unable to go to d3cold state, making power drain even when it's not being used. So it's still not a ideal solution. My Lenovo laptop is 15ARH7, Ryzen 6600H with NVIDIA RTX3050.

NiiSV811, 2023-04-20:
Hi there. I have the same Laptop and I have problems with suspend. 
Steps to reproduce the problem:
- Plug the laptop
- suspend
- Unplug laptop while it's suspend
- Resume the laptop, now turn off, and it won't turn off.
Also if you plug HDMI and suspend laptop. It won't suspend. It's weird.

P.D: Sorry for my English. I'm learning. And Thank you if you could help me, I am a beginner in Linux world

kleidiss, 2023-04-20:
> P.D: Sorry for my English. I'm learning. And Thank you if you could help me, I am a beginner in Linux world

I usually have my laptop plugged in so idk

beh-10257, 2023-05-02:
have anything changed in your part for the past 2 weeks
for me even though I've switched to open beta drivers no difference

kleidiss, 2023-05-30:
@beh-10257 Finally something might have changed 
Check this line from the new proprietary beta driver 

Extended Dynamic Boost support on notebooks to include older Renoir and Cezanne chipsets, in addition to Rembrandt and newer AMD chipsets

kleidiss, 2023-05-30:
New drivers same bug

beh-10257, 2023-05-30:
lol I will wait for these drivers to drop in the aur in my side and update my problem as well
this repo is pathetic
at least tell us that you have putted the bug in your internal bug tracker

onenowy, 2023-06-04:
I have a similar problem. On hp vicuts 16-d1xxx laptop with patch (https://lore.kernel.org/platform-driver-x86/3b5feff0-1d59-37cb-9a5d-22186271a6a4@linux.intel.com/T/#t), It needs restarting nvidia-powerd.service after changing platform profile to change tgp.

kleidiss, 2023-06-04:
@onenowy 
Seems like this issue thread isn't being seen by the NV team
Better go explain this bug on this issue instead

https://github.com/NVIDIA/open-gpu-kernel-modules/issues/392

And yes we have the same bug
Changing platform profile doesn't do shit unless a restart

amrit1711, 2023-06-05:
Dear All,
We have already filed a bug 4142071 for this issue and it has been also root caused.
We will integrate the fix in future release drivers and shall update accordingly.

amrit1711, 2023-10-10:
Dear All,
Fix is available in latest release driver, please verify and share test results.

onenowy, 2023-10-10:
It's working with 3060 mobile on hp victus 16 laptop.

xAlpharax, 2023-10-10:
> Dear All,
> Fix is available in latest release driver, please verify and share test results.

Great! Works for me (Acer AN515-45) with a 3070.

kleidiss, 2023-10-10:
Still not fixed on 535.113.01 for me
Switching to performance power profile and GPU is still stuck at 64 watts and not going up to 85 watts like its supposed to

amrit1711, 2023-10-11:
@kleidiss 
Could you please help to share nvidia bug report from repro state.
Thanks in advance.

kleidiss, 2023-12-28:
@amrit1711 Sorry for the late reply

using nvidia-open-dmks from arch repos 545.29.06 bug isnt fixed
[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/13788275/nvidia-bug-report.log.gz)

kleidiss, 2023-12-28:
nvidia-smi -i 0 -q -d POWER

==============NVSMI LOG==============

Timestamp                                 : Thu Dec 28 18:52:49 2023
Driver Version                            : 545.29.06
CUDA Version                              : 12.3

Attached GPUs                             : 1
GPU 00000000:01:00.0
    GPU Power Readings
        Power Draw                        : 64.93 W
        Current Power Limit               : 65.00 W
        Requested Power Limit             : N/A
        Default Power Limit               : 60.00 W
        Min Power Limit                   : 1.00 W
        Max Power Limit                   : 85.00 W
    Power Samples
        Duration                          : Not Found
        Number of Samples                 : Not Found
        Max                               : Not Found
        Min                               : Not Found
        Avg                               : Not Found
    GPU Memory Power Readings 
        Power Draw                        : N/A
    Module Power Readings
        Power Draw                        : N/A
        Current Power Limit               : N/A
        Requested Power Limit             : N/A
        Default Power Limit               : N/A
        Min Power Limit                   : N/A
        Max Power Limit                   : N/A

mguinhos, 2024-04-14:
I think i'm with a similar problem on my Dell G15 (5530 RTX 3050 6GB)

luckyluca, 2024-04-16:
I'm having a similar problem with the Nvidia driver on Windows 11 on my Lenovo Legion 7 Slim AMD Nvidia 4060 laptop.
Is this related?
using nvidia-smi does return the following error:
Changing power management limit is not supported in current scope for GPU: 00000000:01:00.0.

https://www.nvidia.com/en-us/geforce/forums/geforce-laptops/6/541317/changing-power-management-limit-is-not-working-on-/

amrit1711, 2024-08-02:
Hi All,
We would like to know if there are any other Legion 5 models apart from Lenovo Ideapad Gaming 3 15ACH6 which still exhibits this issue. If yes, please share model information.

Gert-dev, 2024-08-02:
In case it helps: I have a Lenovo Legion Pro 7 16ARX8H since fairly recently and I can confirm this does not happen for me; I can see the power limit changing from 150 to 175 and back dynamically when demanding tasks run. This also still works when switching between different modes.