NVIDIA/open-gpu-kernel-modules #496: External display doesn't refresh unless cursor is moving

omametech, 2023-05-07:
### NVIDIA Open GPU Kernel Modules Version

530.41.03

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

Linux werewolf-cave 6.3.1-arch1-1 #1 SMP PREEMPT_DYNAMIC Mon, 01 May 2023 17:42:39 +0000 x86_64 GNU/Linux

### Hardware: GPU

NVIDIA GeForce GTX 1650 (TU117M)

### Describe the bug

On both the proprietary and open-source drivers, the external display doesn't refresh unless the cursor is on-screen and moving. When the cursor is not moving, the image rendered on display remains static. This bug doesn't occur on 525.89.02 or older.

### To Reproduce

- Plug in an external display, or boot the system with an external display connected.
- Try to move a window onto the display, without the cursor entering the screen space of that display.
- The display will not refresh until you move the cursor into the screen space of that display.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11414948/nvidia-bug-report.log.gz)

### More Info

I have two displays: one internal and one external over HDMI. The internal display is being driven by an Intel UHD Graphics 630, whereas the external display is driven by the NVIDIA GeForce GTX 1650.
My laptop is a Lenovo IdeaPad Gaming 3 15IMH05.

amrit1711, 2023-05-12:
@omametech 
Thanks for writing to us, this is a known issue and has been discussed on Linux Dev Talk forum as below - 
https://forums.developer.nvidia.com/t/530-41-03-external-monitor-stays-frozen-until-i-move-my-mouse/247244
We have also a bug 4074722  filed internally for tracking purpose.
Since this thread is not limited to open kernel modules, I will close it.
Please continue to follow-up or post queries on DevTalk Linux forum for generic issues.

closed 2023-05-12 by amrit1711

hiaselhans, 2023-07-12:
@amrit1711 in the thread it says that a fix has been found around a month ago.
With release intervals around 20 days up until now, when do you think we can have the fix released?

Thanks!

amrit1711, 2023-07-12:
It should be pretty soon but I can not confirm the exact timeline.
I will update on forum thread thread once it is publicly available.