NVIDIA/open-gpu-kernel-modules #497: Error! Bad return status for module build on kernel 6.3.1

RomanAverin, 2023-05-09:
### NVIDIA Open GPU Kernel Modules Version

530.30.02

### Operating System and Version

Fedora 38

### Kernel Release

6.3.1

### Build Command

sudo dkms --force install nvidia/530.30.02

### Terminal output/Build Log

DKMS make.log for nvidia-530.30.02 for kernel 6.3.1-200.fc38.x86_64 (x86_64)
Пн 08 мая 2023 14:01:35 MSK
make[1]: Entering directory '/usr/src/kernels/6.3.1-200.fc38.x86_64'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 13.1.1 20230426 (Red Hat 13.1.1-1)
  You are using:           cc (GCC) 13.1.1 20230426 (Red Hat 13.1.1-1)
  SYMLINK /var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-kernel.o
  SYMLINK /var/lib/dkms/nvidia/530.30.02/build/nvidia-modeset/nv-modeset-kernel.o
 CONFTEST: hash__remap_4k_pfn
 CONFTEST: set_pages_uc
 CONFTEST: list_is_first
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: set_pages_array_uc
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: get_num_physpages
 CONFTEST: pde_data
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: dma_attr_macros
 CONFTEST: dma_map_page_attrs
 CONFTEST: write_cr4
 CONFTEST: of_find_node_by_phandle
 CONFTEST: of_node_to_nid
 CONFTEST: pnv_pci_get_npu_dev
 CONFTEST: of_get_ibm_chip_id
 CONFTEST: pci_bus_address
 CONFTEST: pci_stop_and_remove_bus_device
 CONFTEST: pci_rebar_get_possible_sizes
 CONFTEST: register_cpu_notifier
 CONFTEST: cpuhp_setup_state
 CONFTEST: dma_map_resource
 CONFTEST: get_backlight_device_by_name
 CONFTEST: timer_setup
 CONFTEST: pci_enable_msix_range
 CONFTEST: kernel_read_has_pointer_pos_arg
 CONFTEST: kernel_write_has_pointer_pos_arg
 CONFTEST: dma_direct_map_resource
 CONFTEST: tegra_get_platform
 CONFTEST: tegra_bpmp_send_receive
 CONFTEST: flush_cache_all
 CONFTEST: vmf_insert_pfn
 CONFTEST: jiffies_to_timespec
 CONFTEST: ktime_get_raw_ts64
 CONFTEST: ktime_get_real_ts64
 CONFTEST: full_name_hash
 CONFTEST: pci_enable_atomic_ops_to_root
 CONFTEST: vga_tryget
 CONFTEST: pgprot_decrypted
[... 202 lines cut ...]
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c: In function ‘nvidia_mmap_numa’:
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c:461:19: error: assignment of read-only member ‘vm_flags’
  461 |     vma->vm_flags |= VM_MIXEDMAP;
      |                   ^~
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c: In function ‘nvidia_mmap_helper’:
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c:606:23: error: assignment of read-only member ‘vm_flags’
  606 |         vma->vm_flags |= VM_IO | VM_PFNMAP | VM_DONTEXPAND;
      |                       ^~
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c:673:23: error: assignment of read-only member ‘vm_flags’
  673 |         vma->vm_flags |= (VM_IO | VM_LOCKED | VM_RESERVED);
      |                       ^~
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c:674:23: error: assignment of read-only member ‘vm_flags’
  674 |         vma->vm_flags |= (VM_DONTEXPAND | VM_DONTDUMP);
      |                       ^~
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c:680:23: error: assignment of read-only member ‘vm_flags’
  680 |         vma->vm_flags &= ~VM_WRITE;
      |                       ^~
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c:681:23: error: assignment of read-only member ‘vm_flags’
  681 |         vma->vm_flags &= ~VM_MAYWRITE;
      |                       ^~
make[2]: *** [scripts/Makefile.build:252: /var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:2037: /var/lib/dkms/nvidia/530.30.02/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.3.1-200.fc38.x86_64'
make: *** [Makefile:82: modules] Error 2

aritger, 2023-05-09:
Hi @RomanAverin.  Yes, the vm_area_struct flags changed in the upstream kernel recently.  Recent NVIDIA driver releases should have a conftest for this and handle it correctly.  Can you please test something more recent than 530.30.02?  It looks like the most recent 530.xx release is 530.41.03.  That looks like it has the fix.

RomanAverin, 2023-05-10:
Yes, build 530.41.03 modules well with some warnings

closed 2023-05-10 by RomanAverin