NVIDIA/open-gpu-kernel-modules #502: Resident Evil Village CTX SWITCH TIMEOUT

Vallentinus, 2023-05-14:
### NVIDIA Open GPU Kernel Modules Version

530.41.03 

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

EndeavourOS

### Kernel Release

6.3.2-1-cachyos-bore from cachyos repos

### Hardware: GPU

RTX 2060

### Describe the bug

[ 7839.537906] NVRM: GPU at PCI:0000:26:00: GPU-b50e4ccf-b785-06da-31c0-1f77c0c46f4c
[ 7839.537911] NVRM: Xid (PCI:0000:26:00): 109, pid=55914, name=re8.exe, Ch 00000046, errorString CTX SWITCH TIMEOUT, Info 0x2c02c  

are in dmesg and journalctl.

When playing resident evil village, game freezes and crashing.

### To Reproduce

[ 7839.537906] NVRM: GPU at PCI:0000:26:00: GPU-b50e4ccf-b785-06da-31c0-1f77c0c46f4c
[ 7839.537911] NVRM: Xid (PCI:0000:26:00): 109, pid=55914, name=re8.exe, Ch 00000046, errorString CTX SWITCH TIMEOUT, Info 0x2c02c

### Bug Incidence

Always

### nvidia-bug-report.log.gz

I only have the information I wrote.

Vallentinus, 2023-05-14:
[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11471462/nvidia-bug-report.log.gz)
finally found this

FSKiller, 2023-05-19:
This also happens in Resident Evil 4! Always Xid 109, got more than 8 crashes in the span of 2/3 hours of testing to try to play the game.

When my GPU usage goes above 80 to 85% the game just hangs, audio keeps playing but the video completely hangs.

It appears to be a 7 MONTH OLD BUG, and it's still not fixed, last known good driver version was 520.56.06.

For reference this is the post that was opened in the [Nvidia Forums](https://forums.developer.nvidia.com/t/multiple-cuda-rtx-vulkan-application-crashing-with-xid-13-109-errors/235459/94).

FSKiller, 2023-05-23:
> 

Don't have them, so i can't test.

kleidiss, 2023-05-30:
Anybody checked the new beta driver?

amrit1711, 2023-06-05:
There is already a bug 3959156  filed for game crashing with xid 109 error. You can also track the updates in below thread.
https://forums.developer.nvidia.com/t/multiple-cuda-rtx-vulkan-application-crashing-with-xid-13-109-errors/235459/109
Since this issue is not restricted to open gpu kernel modules, I will close this thread.

closed 2023-06-05 by amrit1711

DanielJoyce, 2023-12-09:
Got hit by this as well, chrome hangs, then gnome-shell on linux. Can't open a terminal so have to reboot. :P

WhyAreAllTheseTaken, 2025-05-11:
Has this been fixed as I'm having the same issue with Proton Experimental in driver 575.51.03 and also in the version where this was supposedly fixed [535.216.01](https://docs.nvidia.com/datacenter/tesla/tesla-release-notes-535-216-01/index.html)? I get this issue in multiple games including Satisfactory and Warframe.

muff11n, 2025-11-10:
this bug is back, same error in kingdom come deliverance 2, game crashes, doesn't matter if GPU load 99% or 50%

[ 1338.230840] NVRM: Xid (PCI:0000:01:00): 109, pid=7247, name=Main, channel 0x00000013, errorString CTX SWITCH TIMEOUT, Info 0x5c01a