NVIDIA/open-gpu-kernel-modules #507: GPU has fallen off of the bus

RealAstolfo, 2023-05-29:
### NVIDIA Open GPU Kernel Modules Version

nvidia-drivers-530.41.03

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Description:    Gentoo Linux

### Kernel Release

Linux localhost 6.3.4-gentoo-x86_64 #1 SMP PREEMPT_DYNAMIC Wed May 24 21:15:30 CDT 2023 x86_64 12th Gen Intel(R) Core(TM) i7-12700H GenuineIntel GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3070 Ti

### Describe the bug

gpu randomly falls off the bus when playing games on the nvidia gpu

### To Reproduce

play a game with the nvidia gpu

### Bug Incidence

Once

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11587638/nvidia-bug-report.log.gz)

kuator, 2023-06-01:
Pick it up then

amrit1711, 2023-06-05:
@RealAstolfo 
You’re getting a XID 79, fallen off the bus. Most common reasons for this issue are overheating or lack of power or outdated SBIOS.
Kindly monitor temperatures, reseat power connectors/the card in its slot, check/replace PSU and update to latest SBIOS.
To check for power issues, you can use nvidia-smi -lgc to prevent boost situations, e.g.
nvidia-smi -lgc 300,1500

Since this issue is not restricted to open gpu kernel modules, I will close this thread and request you to post new issues on below forum or email to us at "linux-bugs@nvidia.com" for generic issues.
https://forums.developer.nvidia.com/c/gpu-graphics/145

closed 2023-06-05 by amrit1711