NVIDIA/open-gpu-kernel-modules #510: Strange: NVreg_EnableResizableBar=1 only works for 5 out of 6 gpu

Qubitium, 2023-05-31:
### NVIDIA Open GPU Kernel Modules Version

530.41.03 

### Does this happen with the proprietary driver (of the same version) as well?

No

### Operating System and Version

Ubuntu 22.04

### Kernel Release

6.2.9

### Hardware: GPU

4090

### Describe the bug

Cpu: 13900k
GPU: 6x 4090 24gb connected to both raw pcie slots and nvme ports via pcie4.0 x4 (nvme to pcie x4 converters). All are connected as PCIE 4.0 x4. 

Based on the nvidia-smi query output, ResizeBar is only activated on 5 out of the 6 gpus.  Note the last gpu output of `8192 MiB`. I have no idea what is going on. Not sure if this is a bug or current limitation of the pcie rebarSize spec/code. 

```
#modprobe.d config
options nvidia NVreg_EnableResizableBar=1 NVreg_OpenRmEnableUnsupportedGpus=1
```

```
localhost:~$ nvidia-smi -q | grep -i bar -A3
    BAR1 Memory Usage
        Total                             : 32768 MiB
        Used                              : 7 MiB
        Free                              : 32761 MiB
--
    BAR1 Memory Usage
        Total                             : 32768 MiB
        Used                              : 35 MiB
        Free                              : 32733 MiB
--
    BAR1 Memory Usage
        Total                             : 32768 MiB
        Used                              : 7 MiB
        Free                              : 32761 MiB
--
    BAR1 Memory Usage
        Total                             : 32768 MiB
        Used                              : 7 MiB
        Free                              : 32761 MiB
--
    BAR1 Memory Usage
        Total                             : 32768 MiB
        Used                              : 7 MiB
        Free                              : 32761 MiB
--
    BAR1 Memory Usage
        Total                             : 8192 MiB
        Used                              : 7 MiB
        Free                              : 8185 MiB
```

```
localhost:~$ sudo dmesg | grep nvidia
[    0.926711] nvidia: loading out-of-tree module taints kernel.
[    0.927709] nvidia: module verification failed: signature and/or required key missing - tainting kernel
[    0.929768] nvidia-nvlink: Nvlink Core is being initialized, major device number 237
[    0.930170] nvidia 0000:01:00.0: BAR 1: releasing [mem 0x40a0000000-0x40afffffff 64bit pref]
[    0.930172] nvidia 0000:01:00.0: BAR 3: releasing [mem 0x40b0000000-0x40b1ffffff 64bit pref]
[    0.930191] nvidia 0000:01:00.0: BAR 1: assigned [mem 0x4800000000-0x4fffffffff 64bit pref]
[    0.930195] nvidia 0000:01:00.0: BAR 3: assigned [mem 0x4400000000-0x4401ffffff 64bit pref]
[    0.930206] nvidia 0000:01:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[    0.976917] nvidia 0000:02:00.0: BAR 1: releasing [mem 0x4080000000-0x408fffffff 64bit pref]
[    0.976919] nvidia 0000:02:00.0: BAR 3: releasing [mem 0x4090000000-0x4091ffffff 64bit pref]
[    0.976937] nvidia 0000:02:00.0: BAR 1: assigned [mem 0x5000000000-0x57ffffffff 64bit pref]
[    0.976941] nvidia 0000:02:00.0: BAR 3: assigned [mem 0x5800000000-0x5801ffffff 64bit pref]
[    0.976953] nvidia 0000:02:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[    1.024821] nvidia 0000:03:00.0: BAR 1: releasing [mem 0x4060000000-0x406fffffff 64bit pref]
[    1.024821] nvidia 0000:03:00.0: BAR 3: releasing [mem 0x4070000000-0x4071ffffff 64bit pref]
[    1.024857] nvidia 0000:03:00.0: BAR 1: assigned [mem 0x6000000000-0x67ffffffff 64bit pref]
[    1.024865] nvidia 0000:03:00.0: BAR 3: assigned [mem 0x5c00000000-0x5c01ffffff 64bit pref]
[    1.024888] nvidia 0000:03:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[    1.065997] nvidia 0000:05:00.0: BAR 1: releasing [mem 0x4040000000-0x404fffffff 64bit pref]
[    1.065998] nvidia 0000:05:00.0: BAR 3: releasing [mem 0x4050000000-0x4051ffffff 64bit pref]
[    1.066033] nvidia 0000:05:00.0: BAR 1: assigned [mem 0x6800000000-0x6fffffffff 64bit pref]
[    1.066041] nvidia 0000:05:00.0: BAR 3: assigned [mem 0x7000000000-0x7001ffffff 64bit pref]
[    1.066064] nvidia 0000:05:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[    1.115389] nvidia 0000:08:00.0: BAR 1: releasing [mem 0x4020000000-0x402fffffff 64bit pref]
[    1.115391] nvidia 0000:08:00.0: BAR 3: releasing [mem 0x4030000000-0x4031ffffff 64bit pref]
[    1.115427] nvidia 0000:08:00.0: BAR 1: assigned [mem 0x7800000000-0x7fffffffff 64bit pref]
[    1.115435] nvidia 0000:08:00.0: BAR 3: assigned [mem 0x7400000000-0x7401ffffff 64bit pref]
[    1.115458] nvidia 0000:08:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[    1.165926] nvidia 0000:0a:00.0: BAR 1: releasing [mem 0x4000000000-0x400fffffff 64bit pref]
[    1.165927] nvidia 0000:0a:00.0: BAR 3: releasing [mem 0x4010000000-0x4011ffffff 64bit pref]
[    1.165964] nvidia 0000:0a:00.0: BAR 1: no space for [mem size 0x800000000 64bit pref]
[    1.165965] nvidia 0000:0a:00.0: BAR 1: failed to assign [mem size 0x800000000 64bit pref]
[    1.165965] nvidia 0000:0a:00.0: BAR 3: no space for [mem size 0x02000000 64bit pref]
[    1.165966] nvidia 0000:0a:00.0: BAR 3: failed to assign [mem size 0x02000000 64bit pref]
[    1.166034] nvidia 0000:0a:00.0: BAR 1: no space for [mem size 0x400000000 64bit pref]
[    1.166034] nvidia 0000:0a:00.0: BAR 1: failed to assign [mem size 0x400000000 64bit pref]
[    1.166035] nvidia 0000:0a:00.0: BAR 3: no space for [mem size 0x02000000 64bit pref]
[    1.166036] nvidia 0000:0a:00.0: BAR 3: failed to assign [mem size 0x02000000 64bit pref]
[    1.166102] nvidia 0000:0a:00.0: BAR 1: assigned [mem 0x4200000000-0x43ffffffff 64bit pref]
[    1.166110] nvidia 0000:0a:00.0: BAR 3: assigned [mem 0x4100000000-0x4101ffffff 64bit pref]
[    1.166131] nvidia 0000:0a:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[    1.223840] nvidia-modeset: Loading NVIDIA UNIX Open Kernel Mode Setting Driver for x86_64  530.41.03  Release Build  (dvs-builder@U16-T02-35-3)  Thu Mar 16 19:23:50 UTC 2023
[    1.224367] [drm] [nvidia-drm] [GPU ID 0x00000100] Loading driver
[    2.361137] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:01:00.0 on minor 0
[    2.361167] [drm] [nvidia-drm] [GPU ID 0x00000200] Loading driver
[    3.784696] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:02:00.0 on minor 1
[    3.784724] [drm] [nvidia-drm] [GPU ID 0x00000300] Loading driver
[    5.034745] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:03:00.0 on minor 2
[    5.034769] [drm] [nvidia-drm] [GPU ID 0x00000500] Loading driver
[    6.267148] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:05:00.0 on minor 3
[    6.267177] [drm] [nvidia-drm] [GPU ID 0x00000800] Loading driver
[    7.503734] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:08:00.0 on minor 4
[    7.503762] [drm] [nvidia-drm] [GPU ID 0x00000a00] Loading driver
[    8.727897] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:0a:00.0 on minor 5
[    9.573879] nvidia-uvm: Loaded the UVM driver, major device number 234.
```

### To Reproduce

...

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11609996/nvidia-bug-report.log.gz)

aaronp24, 2023-05-31:
I don't think this is a driver bug, and arguably isn't really a bug but a platform limitation. I think your system just doesn't leave enough room in the upstream PCIe topology for the full 32 GiB BAR size for that GPU. Take a look at `/proc/iomem` from your bug report log:
```
  4100000000-43ffffffff : PCI Bus 0000:0a
  4400000000-4fffffffff : PCI Bus 0000:01
  5000000000-5bffffffff : PCI Bus 0000:02
  5c00000000-67ffffffff : PCI Bus 0000:03
  6800000000-73ffffffff : PCI Bus 0000:05
  7400000000-7fffffffff : PCI Bus 0000:08
```
This is 48 GiB of address space for every bus except for bus 0a, which only got 12 GiB. Looking at the kernel log and the output of `lspci -nntv`, bus 0a is served by bridge device 00:1d.4:
```
           +-1d.4-[0a]--+-00.0  NVIDIA Corporation Device [10de:2684]
           |            \-00.1  NVIDIA Corporation Device [10de:22ba]
```
You can see where the kernel tried to reassign its window to resize it to 48 GiB and ran out of space:
```
[    1.165960] pcieport 0000:00:1d.4: BAR 15: releasing [mem 0x4000000000-0x4011ffffff 64bit pref]
[    1.165962] pcieport 0000:00:1d.4: BAR 15: no space for [mem size 0xc00000000 64bit pref]
[    1.165963] pcieport 0000:00:1d.4: BAR 15: failed to assign [mem size 0xc00000000 64bit pref]
````
Stepping back a few more steps, it looks like the PCIe root port just plain doesn't have enough space for six 48 GiB address windows:
```
[    0.491797] pci_bus 0000:00: root bus resource [mem 0x000a0000-0x000bffff window]
[    0.491797] pci_bus 0000:00: root bus resource [mem 0x000e0000-0x000effff window]
[    0.491798] pci_bus 0000:00: root bus resource [mem 0x52000000-0xbfffffff window]
[    0.491798] pci_bus 0000:00: root bus resource [mem 0x4000000000-0x7fffffffff window]
[    0.491799] pci_bus 0000:00: root bus resource [bus 00-e0]
```
That last memory window is 256 GiB and you need 288 GiB for all six GPUs at their maximum sizes.

Qubitium, 2023-06-01:
@aaronp24  Thank you for the dissection of the issue. Do you or have an idea which specific chipset/motherboard paper-spec do I, or others, need to specifically look for that holds this limit? Not sure right now how I can avoid this right now (not knowing what to look for) in the future in purchasing decision for a consumer board from intel/amd that supports more than 5x 4090 with rebarsize support.

aaronp24, 2023-06-01:
> Do you know or have an idea which specific chipset/motherboard paper-spec do I, or others, need to specifically look for that holds this limit?

I don't offhand, sorry. This is a pretty specialized configuration; I'd recommend contacting platform vendors directly to find something that's certified to handle that many high-end GPUs.

aritger, 2023-06-01:
Yes, thanks for the analysis, @aaronp24 .  I'm going to go ahead and close this issue.  @Qubitium let me know if you disagree.

closed 2023-06-01 by aritger