NVIDIA/open-gpu-kernel-modules #511: [REGRESSION] [535.43.02 beta] The entire screen is constantly flickering

birdie-github, 2023-05-31:
### NVIDIA Open GPU Kernel Modules Version

535.43.02

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Fedora 38

### Kernel Release

6.3.5

### Hardware: GPU

NVIDIA GeForce GTX 1660 Ti

### Describe the bug

The screen is constantly flickering, no matter what applications are running.

In Firefox it's happening every few seconds. In other "simple" applications it's less frequent.

### To Reproduce

Install.

### Bug Incidence

Once

### nvidia-bug-report.log.gz

https://forums.developer.nvidia.com/t/regression-535-43-02-beta-the-entire-screen-is-constantly-flickering/255054

### More Info

This is a regression.

I've reverted to 530.41.03 and it's all good.

Windows users seem to be [affected](https://www.nvidia.com/en-us/geforce/forums/game-ready-drivers/13/519777/geforce-grd-53598-feedback-thread-released-53023/) as well. Must be a change which affect both code bases.

thesword53, 2023-06-01:
Do you have 2+ monitors?

I think it's a problem with VRR and dual monitor setup. I also have the issue on Windows with the 530 and 535 drivers and Linux with the 535 drivers. The 530 Linux drivers are not affected because of another bug sticking GPU at higher power state on multi-monitor and >60Hz setup.
I also noticed flickering is caused by VRR/G-Sync screen frequency stuttering and happens when GPU is switching power state.

Littlejth, 2023-06-02:
I also have this issue on NixOS with kernel 6.3.5 on plasma 5.27.5 running on my RTX 3070. I tried to log into wayland initially and my screen perpetually flickered without ever making it into the session and I had to reboot my machine. I restarted and then tried to go into my X session which worked fine. I was able to go into the nvidia-settings panel and set both my monitors (144hz 1440p secondary over HDMI and 144hz 4K primary over DVI) down to 60hz. Then when I logged out and switched into wayland, no issue. So at least in my experience here and now, it's not necessarily whether or not it's VRR but more to do with high refresh rate in general. Both my monitors are freesync and work when forcing them to work with G-Sync but I didn't have that enabled at all.

birdie-github, 2023-06-05:
530.41.03 drivers are **also** affected only it takes quite a lot of work to trigger the bug.

Here's roughly what I did:

1. I changed resolution to 1920x1080@60Hz
2. I changed resolution back to 2560x1440@144Hz
3. I switch to Linux console and back

Now the entire screen is flashing and flickering like crazy. Only a complete reboot fixed the issue. Restarting X.org didn't help.

Something is deeply broken for the last two versions of drivers.

birdie-github, 2023-06-15:
It's still reproducible in 535.54.03 though to a much lesser extent.

For instance when I open Firefox the screen is briefly flashing. Sometimes opening new websites in Firefox also produces flashing.

I can live with that but not a single driver prior had this bug.

birdie-github, 2023-06-15:
This is not limited to Firefox. Sometimes the screen briefly flashes even when I'm on the XFCE desktop with all the applications minimized.

I'm not using compositing.

This is reproducible with or without GSync enabled.

birdie-github, 2023-06-15:
I've reverted back to 530.41.03 which are free from this bug.

The screen flashing with the new drivers is driving me insane. I don't want my monitor to die because someone in NVIDIA fucked up and I've heard that it's possible to kill a monitor using software.

TBO I'm mad. This bug report has seen zero replies from NVIDIA developers, the release notes don't confirm it either. At least there's a line in your Windows drivers: _"When using multiple monitors which support adaptive sync, users may see random flicker on certain displays when G-SYNC is enabled after updating to driver 535.98 [4138119]"_

Too bad I've got a single 144Hz HRR/HDR10 GSync 2560x1440 compatible monitor.

birdie-github, 2023-06-15:
Pinging @aritger @aaronp24 @amrit1711

sentakuhm, 2023-06-15:
> Please attach your `sudo nvidia-bug-report`.

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11760971/nvidia-bug-report.log.gz)

Zenzi0, 2023-06-15:
Yep, happens to me too. As long as this happens this driver version really isn't usable.
Nvidia driver version: 535.54.03
Arcolinux (Arch based, uses same repositories)
Kernel: 6.3.8-zen1-1-zen
Wayland, Sway

amrit1711, 2023-06-16:
Hi All,
We see similar issue reported on forum with earlier released driver 525.89.02, will you be able to test if you also see same issue with 525.89.02.
If that's the case, then we are already working on it.
Otherwise, we will need to consider it as a different bug and shall try to reproduce issue locally.

Zenzi0, 2023-06-16:
525.89.02 is not in my repos, so I hope somebody else can provide that information. 
This is probably not very helpful but in my case with Sway I don't experience the flickering issue only with one very specific version of sway-git (r7096.f21090f9-1) and wlroots-git (0.17.0.r6176.12e28c34-1) which an additional non-Arch repo from Arcolinux provides. Any newer version of sway-git or the non-git sway (1:1.8.1-1) have the flickering issue.

birdie-github, 2023-06-17:
> Hi All, We see similar issue reported on forum with earlier released driver 525.89.02, will you be able to test if you also see same issue with 525.89.02. If that's the case, then we are already [...]

I'v had this issue **only** with 535.43.02 and 535.54.03 drivers.

I've **never** had it with older drivers.

dbrhks490, 2023-06-17:
Hi, 

I don't have this issue with the 525.xx.xx. Problem appeared since the 530 beta. I experience a short black flickering at the top of the screen. 

Here what i tried : 

- With the 525 driver, i've no flicker,
- With the 530 driver, my monitor flicker **but** if i add the following command to the kernel parameter `nvidia_drm.modeset=1` the screen doesn't flicker anymore,
- With the new 535 driver, monitor flicker, even if `nvidia_drm.modeset=1` is set. Tried also without this command and my monitor still flicker.

If i put my screen to 144hz and if i enable compositor, my computer is unusable because the flickering is omnipresent. Just scrolling an html web page make the monitor flicker. Watching a video is extremely unpleasant since the screen can flicker more than 10 time in a minute.

The only way to reduce this weird thing is to loverate the framerate (100hz is a good compromise for my 2560x1440 monitor) and the most important is to disable the compositor if possible. By doing this, the GPU does not switch to much between p-states, so the screen continue to flicker but much less frequently. The problem dost not occur in games, when the GPU is at his maximum performances.

I suspect too that only GPU based on the Turing architecture are affected. The people talking about this thing on the Nvidia devlopper forum are all equipped with RTX 2000 or GTX 1660.

Monitor : real g-sync 2560x1440 connected throught displayport
GPU : RTX 2080
OS : Debian or Ubuntu

LoipesMas, 2023-06-17:
My experience has been similar to @dbrhks490 
Arch updated me from 530 to 535 and the flickering appeared.
If I lower my framerate to 120Hz (or lower), the flickering is gone.
If I disable my second display, the flickering is also gone, even at 165Hz.
Happens both on Wayland (Hyprland) and X11 (dwm).

My setup:
Main monitor: 3440x1440@165Hz
Second monitor: 1920x1440@60Hz
GPU: RTX2060
OS: Arch

Let me know if you need any more info or testing

notfood, 2023-06-17:
Chiming in to share my experience.
I'm experiencing partial flickering on my main monitor in a dual monitor setup using `535.54.03` at 60Hz.

Environment:
Main monitor: 1920x1080@60Hz
Second monitor: 1024x768@60Hz
GPU1: GeForce GTX 1650
GPU2: Tesla M40 (not in use)
OS: ArchLinux

It's occasional, I suspect GPU is switching power states, the flickering only happens on the top of the screen. It doesn't matter what I'm running, I can be on X11 or Wayland, it happens on KDE Plasma and I3w, it happens with no applications running, it happens during the beginning of heavy GPU usage. Downgrading it to `530.41.03` shows none of these issues.

birdie-github, 2023-06-17:
Would be great if all the affected people
1. attached their `sudo nvidia-bug-report` and 
2. specified the exact monitor vendor and model

Thanks

CrossroadInTheVoid, 2023-06-17:
530.41.03 flickers even with nvidia-drm.modeset=1 in 3d or playing accelerated video. Bug was published at nvidia.com. So 535.54.03 is not the only bugged version.

birdie-github, 2023-06-18:
> 530.41.03 flickers even with nvidia-drm.modeset=1 in 3d or playing accelerated video. Bug was published at nvidia.com. So 535.54.03 is not the only bugged version.

I intended this bug report to be relevant only for the use case of 535.43.02 and 535.54.03 drivers regressing and 530.41.03 being bug free. And I've been using kms since forever `options nvidia-drm modeset=1`

You really could file a new bug report if you were affected earlier.

CrossroadInTheVoid, 2023-06-18:
```
I intended this bug report to be relevant only for the use case of 535.43.02 and 535.54.03 drivers regressing and 530.41.03 being bug free. And I've been using kms since forever options nvidia-drm modeset=1

You really could file a new bug report if you were affected earlier.
```

530.41.03 is a new feature branch version so I do not think that bugfixing is actual for it. The only thing I want to say — 530.41.03 IS NOT bug free version. It has the same black-flickering bug (or this bug only looks like the same, idk), several users have reported this. So, I think this regression has appeared not in 535.* but in 530.*.

z1atk0, 2023-06-18:
Same here, occasional odd flicker across the top of both monitors, ever since I upgraded to 535.54.03. I always stick to the latest "Production Branch Version", and I never had this problem before.

System: GeForce GTX 1660 Ti displaying on two AOC 24G2SPU connected via HDMI, running Slackware64-15.0 on XOrg-X11 (not Wayland).

```
[root@disclosure:~]# inxi -G
Graphics:
  Device-1: NVIDIA TU116 [GeForce GTX 1660 Ti] driver: nvidia v: 535.54.03
  Device-2: Sunplus Innovation Full HD webcam driver: snd-usb-audio,uvcvideo
    type: USB
  Display: x11 server: X.Org v: 1.20.14 with: Xwayland v: 21.1.4 driver: X:
    loaded: nvidia unloaded: nouveau gpu: nvidia resolution: 1: 1920x1080~60Hz
    2: 1920x1080~60Hz
  API: OpenGL v: 4.6.0 NVIDIA 535.54.03 renderer: NVIDIA GeForce GTX 1660
    Ti/PCIe/SSE2
```

My nvidia-bug-report.log.gz is attached as well.

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11781221/nvidia-bug-report.log.gz)

jarrard, 2023-06-18:
Same issue on NixOS with 4090 and 535 54 03 driver.  X11 has this flicker, often when I'm using a app/browser, doesn't happen on Wayland but then I must set primary display to 60hz or I can't login. (plasma)

Haven't tested Gnome X11 yet.  Kind of warn out with all these nvidia bugs lately...

birdie-github, 2023-06-18:
First things first: the display flickering bug is not the Linux NVIDIA developers screw up, it's the part that comes from the Windows 535.98 drivers which were reported to break multi-monitor systems left and right. NVIDIA has ostensibly fixed this in the Windows 536.23 drivers:

> When using multiple monitors which support adaptive sync, users may see random flicker on certain displays when G-SYNC is enabled after updating to driver 535.98 [4138119]

The bigger issue is that multiple users complained that the Windows 536.23 drivers broke **single** monitor systems as well but it looks like NVIDIA hasn't paid enough if any attention to the issue.

Now considering all the reports above it looks like the Windows multi-monitor bug fix hasn't been ported to Linux and the single monitor flickering issue continues to plague both branches.

Worst of all, it looks like the NVIDIA Linux engineers have been left in the dark or simply don't care about the fact that the last released Linux driver is simply broken and unusable and I reported the bug **three weeks** ago when drivers were still in beta, so they had ample time to investigate and fix it or at least ping their Windows colleagues and ask for help.

I'm mad, I'm simply mad. Luckily 530.41.03 drivers still work flawlessly here, and they support Linux 6.3, so NVIDIA has got another chance of fixing everything.

sachnr, 2023-06-18:
just updated today can confirm new drivers cause flickering in both x11 and wayland.

GrzegorzKozub, 2023-06-19:
For me **only the primary monitor flickers**. I've got two almost identical displays. Also, the flickering **only happens on Wayland**. When select *GNOME on Xorg* as a session to use in GDM the flicker is gone.

I'm on Arch Linux using GDM and GNOME. Potentially relevant packages are in these versions on my system:

```
linux 6.3.8.arch1-1
linux-firmware 20230404.2e92a49f-1

nvidia 535.54.03-2
nvidia-lts 1:535.54.03-2
nvidia-utils 535.54.03-1

gdm 44.1-1
gnome-shell 1:44.2-1
gnome-session 44.0-1
mutter 44.2-1
```

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11784843/nvidia-bug-report.log.gz)

The flickering is so intense that it happens every few seconds. So this is not something one can live with 😃

jarrard, 2023-06-19:
You might be able to stop the flicker if you open nvidia settings and turn OFF No Flipping in the options there. Worked for me.

CrossroadInTheVoid, 2023-06-19:
> You might be able to stop the flicker if you open nvidia settings and turn OFF No Flipping in the options there. Worked for me.

No Flipping? Is it new option?
There is Allow Flipping checkbox in OpenGL Settings only in earlier versions.

birdie-github, 2023-06-19:
> You might be able to stop the flicker if you open nvidia settings and turn OFF No Flipping in the options there. Worked for me.
> 
> However I think there is a performance hit if you do so. Could be related to plasma not having triple buffer set which you can do with the xorg config. No idea about Wayland since I can barely get that working under PLASMA.

Could you show how it's done?

This is what I have here and I have flickering:

[image]

jarrard, 2023-06-19:
Yeah I think it is Allow Flipping. I turned that off and issue went away but perhaps if you try with it on?

It's also possible my issue is different because I managed to install the older 530 drivers and it still did it.

birdie-github, 2023-06-19:
> Yeah I think it is Allow Flipping. I turned that off and issue went away but perhaps if you try with it on?
> 
> It's also possible my issue is different because I managed to install the older 530 drivers and it still did it.

This option is applicable only to OpenGL applications or when you're running compositing.

I've got neither. Plain X11 XFCE session.

jarrard, 2023-06-19:
Ok well it helped under Plasma, that is my test case since I want that to work in X11 and WL. 

Also I thought XFCE did have a compositor.

z1atk0, 2023-06-19:
> Also I thought XFCE did have a compositor.

OT: It does, but it can be disabled. Which I also did on my old laptop running Slack64-15.0 & XFCE, because it just cost too much performance there (the laptop has an old Celeron CPU and an Intel Mobile 4 GPU).

birdie-github, 2023-06-19:
> Ok well it helped under Plasma, that is my test case since I want that to work in X11 and WL.
> 
> Also I thought XFCE did have a compositor.

I'm not interested in bells and whistles and increased GPU power consumption. Besides this bug is not about OpenGL/Vulkan applications.

GrzegorzKozub, 2023-06-19:
For my scenario https://github.com/NVIDIA/open-gpu-kernel-modules/issues/511#issuecomment-1596567416 disabling `Allow Flipping` under `OpenGL Settings` did not fix the issue. This is probably because my issue is on Wayland only.

Also note that disabling this toggle did not persist a reboot.

sentakuhm, 2023-06-19:
@birdie-github 
i'm about to build my next pc its all AMD, i already decided that, especially when AMD finally move all their drivers and firmware to open-source.

CrouchingTigger9, 2023-06-19:
Hello,

I am affected by the issue too. The flickering didn't start right away after installation/reboot. First time I tested the new driver, noticeable flickering started only after a short Proton gaming session and persisted after a reboot. So, I went back to 530.41.03. On the next try I was unaffected for the better part of the day and it started quite randomly while watching YouTube on Firefox. Unchecking "Allow Flipping" seems to have fixed it for now.
System information and bug report below.

System: Gentoo (17.1 profile with some ~amd64 packages), kernel 6.3.8, OpenRC, X11, KDE Plasma 5.27.5.
Monitor: ASUS ROG Swift PG329Q @165Hz connected via DP.
Video Card: Gigabyte RTX 3700.

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11789538/nvidia-bug-report.log.gz)

valvatorres, 2023-06-20:
I posted elsewhere about this problem and was referred to this thread to report so here it goes:

This latest nvidia driver on linux causes a flickering black jagged line to appear on the screen whenever I use any gpu accelerated program. 
My relevant info:
linux 6.3.8.arch1-1
nvidia 535.54.03-2
nvidia-lts 1:535.54.03-2
nvidia-utils 535.54.03-1
RTX 3080
Monitor: Dell AW3420DW 120hz w/ gsync on
Hyprland (a wayland compositor)

I reverted the nvidia drivers to 530.41.03 and the flickering is gone now.

jarrard, 2023-06-20:
> KDE Plasma 5.27.5.

The flickering is likely also a Plasma update issue that caused the NVIDIA bug to suddenly be noticed. I wonder if rolling back a couple plasma releases would fix the subtle flicker issue.

Anyway I'm still trying to get 535 and Plasma WL working together. There is also a X11 to WL Plasma bug that causes WL logins to kick back to the login manager, THAT isn't the >60hz bug NVIDIA, that bug causes system lock-up more or less needing a hard reset.

aritger, 2023-06-20:
Screen flicker is a fairly generic symptom, and there are no doubt different bugs discussed here, particularly if the affected driver versions are different.  I don't know for certain if the originally reported issue here is the same as what was reported on Windows.  Due to release schedule differences, some times it can take a release or two for some of those hotfixes to propagate from Windows to Linux, unfortunately.

@birdie-github: it is curious that the regression point for you is the same as when we enabled support for driving a single display with multiple heads.  As an experiment, you could try setting:

    Option "ModeValidation" "MaxOneHardwareHead"

though, I'm not optimistic that will impact your case.  It would still be a useful data point.

Another experiment would be, in nvidia-settings, to change the PowerMizer "Preferred Mode" to "Prefer Maximum Performance".

It would also be interesting to disable your CompositionPipeline options in the X configuration file.  That will of course result in tearing, but it would be a helpful data point to know if that alters the current flickering you see.

birdie-github, 2023-06-20:
@aritger 

> it is curious that the regression point for you is the same as when we enabled support for driving a single display with multiple heads. As an experiment, you could try setting:

Not sure if it's relevant in my case considering I've got a single monitor.

> in nvidia-settings, to change the PowerMizer "Preferred Mode" to "Prefer Maximum Performance".

Will try it later.

In the past it was possible to switch NVIDIA drivers on the fly without rebooting, now with KMS it requires reboots which totally sucks. And then it looks like recent NVIDIA drivers have become somewhat persistent? I don't know how to describe it but again in the past restarting X.org resulted in the driver resetting whatever settings it had, nowadays it's not enough. Oh, and there's firmware involved. Darn, Linux has become so much worse in some regards than Windows recently.

There's a way to unload a KMS driver but I'm not sure it will work: https://nouveau.freedesktop.org/KernelModeSetting.html - that's for nouveau but should work for other KMS drivers:
```bash
#!/bin/bash

echo 0 > /sys/class/vtconsole/vtcon1/bind
rmmod nouveau
/etc/init.d/consolefont restart
rmmod ttm
rmmod drm_kms_helper
rmmod drm
```

birdie-github, 2023-06-20:
Again this file report is for the following specific case:

1. A single monitor
2. GTX 1660 Ti
3. 530.41.03 drivers are **totally fine**
4. I'm using KMS (`options nvidia-drm modeset=1`)
5. I'm **not** using compositing or anything fancy - I've got a plain XFCE X.org session.

If people who experience flickering had it with drivers 530.41.03 as well, you definitely want to file a separate bug report.

AlexGoinsNV, 2023-06-20:
Hello @birdie-github ,

> I've had this issue **only** with 535.43.02 and 535.54.03 drivers.
> 
> I've **never** had it with older drivers.

Could you confirm that you tried with 525.89.02 with and without OpenRM? If not, it would be worth trying both of those cases. I have some suspicion that although 530 is "newer", the regression could be present in 525.89.02 when running without OpenRM.

Thanks!

birdie-github, 2023-06-20:
@AlexGoinsNV 

What's "OpenRM"? :)

AlexGoinsNV, 2023-06-20:
@birdie-github 

> What's "OpenRM"? :)

Sorry, I'm referring to the open source version of nvidia.ko, explained here:

https://us.download.nvidia.com/XFree86/Linux-x86_64/525.89.02/README/kernel_open.html

There are some differences in behavior between that and the proprietary version that could affect reproducibility. Your original bug report indicates that you aren't using it, so I mainly just want to confirm that you tried with 525.89.02; the fact that it doesn't reproduce on 530.41.03 unfortunately doesn't rule it out.

birdie-github, 2023-06-20:
This bug report is strictly about the closed/proprietary driver. Sorry if I didn't make it clear earlier.

I don't remember if I ever ran 525.89.02 drivers, should I try them? Again, I have no issues with 530.41.03 drivers which I have installed now.

LoipesMas, 2023-06-20:
@AlexGoinsNV I couldn't downgrade to 525.89.02, but I'm *pretty sure* I was using them when they were the current release (the proprietary version) and I'm *pretty sure* I didn't have any issues.

For me, the issues went from 0 to 100 after first reboot after upgrade to 535.54.03 (proprietary version). Then I switched to open version, but the issue is the same.

AlexGoinsNV, 2023-06-20:
> This bug report is strictly about the closed/proprietary driver.

This GitHub repository tracks the open source driver, but that's fine, I just wanted to confirm since it could have an effect on this bug.

> I don't remember if I ever ran 525.89.02 drivers, should I try them? Again, I have no issues with 530.41.03 drivers which I have installed now.

Yes, please do. Depending on the root cause, I have some reason to believe that 525.89.02 may exhibit the issue despite 530.41.03 being fine.

> It would be great if you released whatever display fixes you've made to your Windows driver ASAP, even as a beta. It's going to be more productive than chasing old bugs which might have already been fixed.

I looked into the Windows bug that you mentioned:

> When using multiple monitors which support adaptive sync, users may see random flicker on certain displays when G-SYNC is enabled after updating to driver 535.98 [4138119]

Although the symptoms sound similar, the original bug and the fix are Windows-specific and do not impact the Linux driver.

LoipesMas, 2023-06-20:
@aritger 
> As an experiment, you could try setting:
> 
> ```
> Option "ModeValidation" "MaxOneHardwareHead"
> ```
I tried that and it didn't fix it. Maybe it made it slightly better, but maybe not, can't really tell. 

> Another experiment would be, in nvidia-settings, to change the PowerMizer "Preferred Mode" to "Prefer Maximum Performance".

It seemed to help for a few minutes, but then it got bad again, so maybe it didn't actually do anything.

The only thing that actually helps is making the gpu do stuff. If I run `glxgears -swapinterval 0` (even in the background), everything is fine: no flickers, no "blackouts", even with both displays at max refresh rate, X11 or Wayland.

dbrhks490, 2023-06-20:
I've already tried past Sunday the following setting : `Option "ModeValidation" "MaxOneHardwareHead"`
It was added under the display section of my xorg file. Sadly, no change and flicker still occur. _Maybe the flickering is a little bit reduced with this option but i can be wrong and it's can be a placebo. Maybe someone can test further longer._

Set "prefer maximum performance" mode help to greatly reduce flickering since the pstates doesn't switch anymore between P0 and P4. But now, the GPU stuck his memory clock at maximum frequency and consume 40w (66w in dual monitor). In adaptive mode, it only consume 12w (and 19w in dual monitor). Moreover, the GPU heat very fast. So no, it's not an acceptable solution for me.

Moreover, I've tried the following settings : 

- Added this lines in my environment file 
`export __GL_MaxFramesAllowed=1`
`export __GL_SYNC_TO_VBLANK=0`
`export __GL_SYNC_DISPLAY_DEVICE=DP-2`
`export __NVD_BACKEND=direct`
`export __GL_b5f2b3=0xFFFFFFFF`

- set force composition pipeline and full composition pipeline
- set disable "allow flipping" and "g sync"
- Tried to lock GPU core clock at their minimum with the following command : `nvidia-smi -lgc 300,300`

Sadly, all failed and flickering still occur. 

I've not found a viable solution for using my computer nowadays. Even at 60hz, flickering occur. Downgrade to the 530 new feature branch or 525 production branch is the solution but some games I'm playing crash with these drivers (ACValhalla and RE4).

You're saying that it could take 2 or 3 major revision before a potential fix come from the Windows to the Linux driver.
Since news drivers are releases approximately every 3 month, this means that a fix could possibly be included in the 550 release ! We will need to wait the end of the first quarter of 2024.

I'm sincerely hope that a fix will be found before the 550 release. I' will continue to try some things when possible and report it here.

dbrhks490, 2023-06-20:
Moreover, I've tried the following settings : 

- Added this lines in my environment file 
`export __GL_MaxFramesAllowed=1`
`export __GL_SYNC_TO_VBLANK=0`
`export __GL_SYNC_DISPLAY_DEVICE=DP-2`
`export __NVD_BACKEND=direct`
`export __GL_b5f2b3=0xFFFFFFFF`

- set force composition pipeline and full composition pipeline
- set disable "allow flipping" and "g sync"
- Tried to lock GPU core clock at their minimum with the following command : `nvidia-smi -lgc 300,300`

Sadly, all failed and flickering still occur. 

I've not found a viable solution for using my computer nowadays. Even at 60hz, flickering occur. Downgrade to the 530 new feature branch or 525 production branch is the solution but some games I'm playing crash with these drivers (ACValhalla and RE4).

You're saying that it could take 2 or 3 major revision before a potential fix come from the Windows to the Linux driver.
Since news drivers are releases approximately every 3 month, this means that a fix could possibly be included in the 550 release ! We will need to wait the end of the first quarter of 2024.

I'm sincerely hope that a fix will be found before the 550 release. I' will continue to try some things when possible and report it here.

aritger, 2023-06-20:
> You're saying that it could take 2 or 3 major revision before a potential fix come from the Windows to the Linux driver.

If you're referring to this from me:

> Due to release schedule differences, some times it can take a release or two for some of those hotfixes to propagate from Windows to Linux, unfortunately.

I meant individual release builds within a release branch, not entirely separate release branches.  I.e., it may take a few weeks or so; not many months.

Anyway, to be clear: none of these suggested experiments were suggested as /solutions/.  They are experiments to help identify the cause, so that we can focus our efforts on identifying a root cause and fix.

From the results so far (thanks) it sounds like the problem is that GPU clocks are not being raised high enough to satisfy the display.

So, I think the next question is:
(a) Did the clock requirements of display increase to cause the regression?
(b) Or, are the clocks not being raised as high as they were before?

It would help to identify:

- For a driver version that shows the problem, what clocks (reported through nvidia-settings PowerMizer page, or nvidia-smi) is the GPU using when flickering occurs?  What clocks are used when flickering does not occur?

- For a driver version that does NOT show the problem, what clocks are used?

birdie-github, 2023-06-21:
> * For a driver version that does NOT show the problem, what clocks are used?

530.41.03, 1660 Ti, DP 1.4, 2560x1440, 144Hz, FreeSync/Gsync compatible, **no** flickering with:
```
$ nvidia-smi dmon
# gpu    pwr  gtemp  mtemp     sm    mem    enc    dec   mclk   pclk 
# Idx      W      C      C      %      %      %      %    MHz    MHz 
    0      8     35      -      1     15      0      0    405    300 
    0      8     35      -      5     16      0      0    405    300 
    0      8     35      -      3     15      0      0    405    300 
    0      8     35      -      3     16      0      0    405    300 

nvidia-smi -a | grep -i mhz 
        Graphics                          : 300 MHz
        SM                                : 300 MHz
        Memory                            : 405 MHz
        Video                             : 540 MHz
        Graphics                          : 2400 MHz
        SM                                : 2400 MHz
        Memory                            : 6001 MHz
        Video                             : 1950 MHz
```

LoipesMas, 2023-06-21:
RTX 2060, 3440x1440@120Hz, VRR disabled
<details><summary>530.41.03 -- no issues </summary>
<p>

```
❯ nvidia-smi dmon
# gpu    pwr  gtemp  mtemp     sm    mem    enc    dec    jpg    ofa   mclk   pclk
# Idx      W      C      C      %      %      %      %      %      %    MHz    MHz
    0     16     41      -     6      8      0      0      0      0    810    435
    0     14     41      -     8     10      0      0      0      0    405    435
    0     36     42      -    24     25      0      0      0      0   7000   1320
    0     34     42      -    13      3      0      0      0      0   7000   1320
    0     19     41      -     2      1      0      0      0      0    810    435
    0     16     41      -     3      7      0      0      0      0    810    435
    0     15     41      -     7     11      0      0      0      0    405    330
    0     27     42      -    28     25      0      0      0      0   7000   1320
    0     35     42      -    22     20      0      0      0      0   7000   1320
    0     34     42      -     3      3      0      0      0      0   5000    465
    0     28     42      -    16      7      0      0      0      0   5000    435
    0     32     42      -    12      5      0      0      0      0   5000    435
    0     34     42      -    10      4      0      0      0      0   7000   1365
    0     22     42      -     5      3      0      0      0      0    810    750
    0     18     42      -     2      5      0      0      0      0    810    435
    0     16     41      -    19     18      0      0      0      0    810    570
    0     20     42      -    13     15      0      0      0      0    810    780
    0     34     42      -    26     26      0      0      0      0   7000   1410
    0     35     42      -    22      6      0      0      0      0   7000   1410
    0     19     42      -     4      2      0      0      0      0    810    435
    0     16     41      -    18     14      0      0      0      0    810    465
    0     14     41      -    29     19      0      0      0      0    405    585
    0     35     42      -    18     18      0      0      0      0   7000   1350
    0     35     42      -    33      6      0      0      0      0   7000   1350
    0     29     42      -    20      5      0      0      0      0   5000    795
    0     16     42      -     5      2      0      0      0      0    810    540
    0     16     41      -    13     10      0      0      0      0    810    435
    0     14     41      -     7      9      0      0      0      0    405    435
    0     14     41      -    10     16      0      0      0      0    405    435
    0     15     41      -    10     18      0      0      0      0    405    435
    0     14     41      -    15     19      0      0      0      0    405    435
    0     14     41      -     9     17      0      0      0      0    405    435
    0     14     41      -    10     18      0      0      0      0    405    435
    0     14     41      -    11     18      0      0      0      0    405    435
    0     39     43      -    34     15      0      0      0      0   7000   1365
    0     40     43      -    11      5      0      0      0      0   7000   1365
    0     33     42      -    10      4      0      0      0      0   5000    600
    0     19     42      -     8      6      0      0      0      0    810    435
    0     16     42      -     5      9      0      0      0      0    810    435
    0     14     42      -     7      9      0      0      0      0    405    435
    0     15     41      -    10     18      0      0      0      0    405    330
    0     14     42      -    16     18      0      0      0      0    405    330
    0     14     41      -    17     18      0      0      0      0    405    330
    0     14     42      -    16     18      0      0      0      0    405    330
# gpu    pwr  gtemp  mtemp     sm    mem    enc    dec    jpg    ofa   mclk   pclk
# Idx      W      C      C      %      %      %      %      %      %    MHz    MHz
    0     15     42      -    16     18      0      0      0      0    405    330
    0     15     41      -    16     18      0      0      0      0    405    330
    0     14     41      -    20     22      0      0      0      0    405    435
    0     15     41      -    10     18      0      0      0      0    405    435
    0     16     42      -    14     21      0      0      0      0    405    435
    0     14     41      -    15     21      0      0      0      0    405    435
    0     20     42      -    11     19      0      0      0      0    405    630
    0     17     41      -    28     28      0      0      0      0    810   1080
    0     22     42      -     7     11      0      0      0      0    810    915
    0     17     42      -    22     22      0      0      0      0    810   1005
    0     16     41      -     7     11      0      0      0      0    405    630
    0     17     42      -    24     23      0      0      0      0    810   1065
    0     25     42      -    14     16      0      0      0      0   7000   1575

    0     36     43      -    21     19      0      0      0      0   7000   1575
    0     19     42      -     2      1      0      0      0      0    810    780
[... 146 lines cut ...]
    0     23     42      -    21     21      0      0      0      0    810   1035
    0     43     43      -    38     32      0      0      0      0   7000   1710
[... cut ...]
    0     16     42      -    31     22      0      0      0      0    810    780
    0     22     42      -    10     12      0      0      0      0    810   1050
    0     37     43      -    35     25      0      0      0      0   7000   1290
    0     34     43      -     6      4      0      0      0      0   5000    450
    0     18     42      -     2      2      0      0      0      0    810    435
    0     17     42      -    22     18      0      0      0      0    810    645

```

</p>
</details>

LoipesMas, 2023-06-21:
Also I feel like there are three types of "symptoms" that (at least for me) are present in 535 but no in 530.

1. "Blackouts" - as if the display was disconnected and connected again. Only happens with both displays on and with high refresh rate, but only the main display turns off. And seems to happen when changing P-states (I think mainly between P0<->P5). (And it seems that on 530 my GPU is always at P0, so that might be a clue).
2. Partial flickers - top ~third of the display goes black for a frame or so. I can't see any obvious correlations with other things, so it's hard to reliably reproduce (using Firefox it may happen every few seconds, but also may not happen at all for a minute or two)
3. Corrupted redraws with VRR - if I turn on VRR, when parts of the screen get redrawn, most of the time about 15% of the redrawn area is black. This one isn't confined to the top of the screen and is much more visible. But this only happens with VRR on.

(I also get some artifacts - parts of screen not being updated or being updated incorrectly - but this seems even more unrelated and it might be a Firefox issue (but it's on 535 driver only)) 

Do you think those three "symptoms" show the same underlying issue? Or are they likely unrelated to each other?

dbrhks490, 2023-06-21:
> I do have local repro now but it is not 100% consistent on my setup.

You can induce flickering more frequently by :

- Opening `nvidia-settings` on the "powermizer" tab.
- Open multiple instances of `vkcube`

With my GPU, an RTX 2080 and a monitor with a refresh rate of 144hz, after 6 or 7 instances open, the pstates start switching between P0 and P1 constantly. It is at this moment that the flickering occur very often.

My clocks at idle are : 

- Graphics 300 mhz
- SM 300 mhz
- Memory 405 mhz
- Video 540 mhz

PowerDraw
- Approximately 12 watt single monitor (144hz)
- Approximately 20 watt dual monitor (60hz - 120hz)

Whether with the 525 or 530 or 535, the clocks are the same, even when their change when pstates start switching.

> Also I feel like there are three types of "symptoms"

Yes, you are right from what I have read here and elsewhere. But do you experience all 3 depending on the situation?
About me, whether with 1 or 2 screens, I am in case number 2, a flicker on the top of the screen. Not a complete black screen. Whatever the situation, whatever the frequency, I am always in case 2.

amrit1711, 2023-06-21:
Thanks @dbrhks490 
I will try the recommended steps for consistent repro.

LoipesMas, 2023-06-21:
> > 
> 
> You can induce flickering more frequently by :
> 
>     * Opening `nvidia-settings` on the "powermizer" tab.
> 
>     * Open multiple instances of `vkcube`

I tried this and in my case it didn't seem any more "effective" than normal usage

> But do you experience all 3 depending on the situation?

Yes, depending on how I set up my displays I can be in either of the 3 cases.

GrzegorzKozub, 2023-06-22:
Reading you guys are talking about VRR I realized one of my monitors, the flickering one, supports FreeSync. I found that disabling FreeSync helps. I updated my original https://github.com/NVIDIA/open-gpu-kernel-modules/issues/511#issuecomment-1596567416 now.

errantmind, 2023-06-23:
I'm also having this issue after updating from 525 to 535 (on Arch Linux). I can confirm it happens when the pstate changes as setting the PowerMizer to 1 with `nvidia-settings -a "[gpu:0]/GPUPowerMizerMode=1"` prevents the flickering. 

My primary monitor is 240hz, my secondary monitor is 60hz. I haven't tested whether or not it stops when only using one monitor. A fix would be appreciated.

birdie-github, 2023-06-23:
Yesterday after connecting and disconnecting a 4K TV via HDMI, the upper one third of my monitor started flickering with 530.41.03 drivers :-( After a few reboots, a complete power off and booting into Windows for an hour, the bug disappeared. It's all quite unnerving. Windows is running 531.68 drivers.

LoipesMas, 2023-06-23:
@birdie-github have you tried 535 with FreeSync/GSync __disabled on the display__?

[169 more replies]

closed 2023-10-17 by aritger