NVIDIA/open-gpu-kernel-modules #521: {Wayland} Screen tearing in Proton/Vulkan Games

VeilSilence, 2023-06-19:
### NVIDIA Open GPU Kernel Modules Version

nvidia-open-dkms 535.54.03-2

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

uname -aLinux Veil 6.3.8-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Wed, 14 Jun 2023 20:10:15 +0000 x86_64 GNU/Linux

### Hardware: GPU

NVIDIA GeForce RTX 3090 (UUID: GPU-0a730a3d-f915-32f2-19e0-bef38f6809b4)

### Describe the bug

Greetings. I observe screen tearing in Proton/Native games ( Xwayland ). No screen tearing in desktop/daily usage.
Dual monitor setup. One is Samsung Odyssey g7 240Hz ( DP ), second old Aoc 60Hz (HDMI ).
Disabling `GL_VRR_ALLOWED` seems fix the issue.

### To Reproduce

1. Launch any game with `__GL_VRR_ALLOWED=1`.  
2. Move mouse quickly.
3. Observe tearing.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11792665/nvidia-bug-report.log.gz)

### More Info

My env variables
```
env = GBM_BACKEND, nvidia-drm 
env = LIBVA_DRIVER_NAME, nvidia
env = __GLX_VENDOR_LIBRARY_NAME, nvidia
env = __GL_GSYNC_ALLOWED, 1
env = __GL_VRR_ALLOWED, 1
```
My modprobe.d/nvidia.conf
```                                     
options nvidia-drm modeset=1
options nvidia NVreg_UsePageAttributeTable=1
options nvidia NVreg_EnableResizableBar=1
```

jkrhu, 2023-06-21:
Hi!

Same thing happening with the regular driver on Fedora 38 using the RTX 3060. Posted my issue over on the Nvidia dev forum [here](https://forums.developer.nvidia.com/t/535-54-03-screen-tearing-while-gaming-on-wayland-session/257065).

SneakyLittleman, 2023-07-22:
Same thing here... Same OS as original poster. Arch 6.4 zen kernel. 535 nvidia drivers on 3080FE.

jkrhu, 2023-08-11:
Issue is still present on 535.98 drivers. There's a clear visible horizontal tearing line on Wayland. Issue is not present on X11.

VeilSilence, 2023-08-24:
Tested 535.104.05. Problem is still there.

jkrhu, 2023-09-24:
Re-tested on 535.113.01, issue still present on Wayland and GNOME 44.5.