NVIDIA/open-gpu-kernel-modules #522: Older Minecraft versions are broken with the 535.54.03 driver

Tarballwalf, 2023-06-23:
### NVIDIA Open GPU Kernel Modules Version

535.54.03

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

EndeavourOS Linux

### Kernel Release

Linux TUF-F15 6.3.9-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Wed, 21 Jun 2023 20:46:07 +0000 x86_64 GNU/Linux

### Hardware: GPU

NVIDIA GeForce RTX 3060 Laptop GPU

### Describe the bug

The issue appeared as soon as I updated to 535.43.03 proprietary driver, later on I tried to see if this happens with the open kernel modules, and it indeed happens. 

**Words cannot describe what I saw**
But I'll try to sum it up. **Really** broken textures, sometimes they start to flicker. Resizing the window makes them go even crazier.

See attached screenshots
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### To Reproduce

Open older Minecraft versions running java 8 (other people pointed out it’s because of LWJGL 2). Modded or not

### Bug Incidence

Always

### nvidia-bug-report.log.gz

Bug report is so large (almost 2 million lines), I have to upload it here as a file
[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/11847316/nvidia-bug-report.log.gz)

### More Info

The issue does not seem to be related to X11 or Wayland. This happens on both.

vova7865, 2023-06-26:
Can confirm, this appears to affect Minecraft versions 13w41a to 1.12.2. Does not seem to affect versions 13w39b and below, or 17w43a and above. Minecraft switched to LWJGL 3.1.2 in 17w43a, however, both 13w39b and 13w41a use LWJGL 2.9.4-nightly-20150209.

ttabi, 2023-06-27:
I don't know anything about LWJGL, but it seems to me that "LWJGL 2.9.4-**nightly**-20150209" is a nightly, unstable build.  It looks like this was a bug in LWJGL that was fixed.  What are you expecting Nvidia to do about it?

Lolothepro, 2023-07-01:
This version of the driver also crashes Minecraft in certain situations.

Lolothepro, 2023-07-03:
workaround: __GL_THREADED_OPTIMIZATIONS=0

Tarballwalf, 2023-07-03:
> workaround: __GL_THREADED_OPTIMIZATIONS=0

It indeed works. 
You're a lifesaver, after kernel 6.4 NVIDIA 530 drivers stopped working.

FelipeFMA, 2023-07-08:
> workaround: __GL_THREADED_OPTIMIZATIONS=0

what exactly is the workaround? where do i need to put this?

ErrorNoInternet, 2023-07-09:
> > workaround: __GL_THREADED_OPTIMIZATIONS=0
> 
> what exactly is the workaround? where do i need to put this?

It's an environment variable, you can put it in /etc/environment to apply it globally or do `__GL_THREADED_OPTIMIZATIONS=0 minecraft-launcher`.

FelipeFMA, 2023-07-09:
> > > workaround: __GL_THREADED_OPTIMIZATIONS=0
> > 
> > 
> > what exactly is the workaround? where do i need to put this?
> 
> It's an environment variable, you can put it in /etc/environment to apply it globally or do `__GL_THREADED_OPTIMIZATIONS=0 minecraft-launcher`.

it works, thank you

Guanran928, 2023-07-09:
Can confirm this issue on driver version 535.xx, the glitch happens in Minecraft version 13w41a to 1.12.2 (including both version), downgrading the driver to 530.41.03 fixed this for me, so I am assuming it was NVIDIA.

Lolothepro, 2023-07-10:
> I don't know anything about LWJGL, but it seems to me that "LWJGL 2.9.4-**nightly**-20150209" is a nightly, unstable build. It looks like this was a bug in LWJGL that was fixed. What are you expecting Nvidia to do about it?

That's not true, this version of LWJGL is stable, it's just that NVIDIA checks that liblwjgl64.so or liblwjgl.so is loaded to apply the optimization that creates the graphics corruptions, but these versions of LWJGL are not detected by the driver.

RockyRoad65, 2023-08-12:
> It's an environment variable, you can put it in /etc/environment to apply it globally or do `__GL_THREADED_OPTIMIZATIONS=0 minecraft-launcher`.

When I try running __GL_THREADED_OPTIMIZATIONS=0 minecraft-launcher when I try to launch older versions it crashes when it tries to launch I get an error "The game crashed whilst initializing game
Error: org.lwjgl.LWJGLException: X Error - disp: 0x7f9de49ac130 serial: 157 error: BadValue (integer parameter out of range for operation) request_code: 152 minor_code: 24" On newer versions that were previously unaffected it now doesn't work either.
[image]

OilFreak, 2023-09-06:
> > > workaround: __GL_THREADED_OPTIMIZATIONS=0
> > 
> > 
> > what exactly is the workaround? where do i need to put this?
> 
> It's an environment variable, you can put it in /etc/environment to apply it globally or do `__GL_THREADED_OPTIMIZATIONS=0 minecraft-launcher`.

any way to do this with multi mc?

Tarballwalf, 2023-09-06:
> > > > workaround: __GL_THREADED_OPTIMIZATIONS=0
> > > 
> > > 
> > > what exactly is the workaround? where do i need to put this?
> > 
> > 
> > It's an environment variable, you can put it in /etc/environment to apply it globally or do `__GL_THREADED_OPTIMIZATIONS=0 minecraft-launcher`.
> 
> any way to do this with multi mc?
`env __GL_THREADED_OPTIMIZATIONS=0` in wrapper command

OilFreak, 2023-09-06:
Ig i'm more asking how to launch Multi Mc from terminal.

Guanran928, 2023-09-06:
> > > > workaround: __GL_THREADED_OPTIMIZATIONS=0
> > > 
> > > 
> > > what exactly is the workaround? where do i need to put this?
> > 
> > 
> > It's an environment variable, you can put it in /etc/environment to apply it globally or do `__GL_THREADED_OPTIMIZATIONS=0 minecraft-launcher`.
> 
> any way to do this with multi mc?

https://wiki.archlinux.org/title/Environment_variables

OilFreak, 2023-09-06:
running pop_os.

OilFreak, 2023-09-08:
> `env __GL_THREADED_OPTIMIZATIONS=0` in wrapper command
this worked yesterday and today it doesn't

Nixuge, 2023-09-12:
> Ig i'm more asking how to launch Multi Mc from terminal.

With prism launcher (multimc fork) you just have to run prismlauncher, i guess it'd be the same for multimc @OilFreak

Tortillas-IT, 2023-11-16:
with the new drivers (545.29.02) the issue is now fixed

Tarballwalf, 2023-11-18:
I guess I can close this issue now

closed 2023-11-18 by Tarballwalf